UniversalRobots/Universal_Robots_ROS2_Driver #737: Latest ros2_control mock hardware setup has deprecated xacro parameters

sea-bass, 2023-07-10:
I just tested out the 2023-06-29 ROS Humble snapshot and (surprise!) `ros2_control` has introduced another incompatibility with mock hardware, it seems.

```
[ros2_control_node-2] [INFO] [1689026791.796165788] [resource_manager]: Loading hardware 'ur5e' 
[ros2_control_node-2] [INFO] [1689026791.797834110] [resource_manager]: Initialize hardware 'ur5e' 
[ros2_control_node-2] [WARN] [1689026791.797878390] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-2] [WARN] [1689026791.797894218] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-2] <state_interface name="velocity"> 
[ros2_control_node-2]   <param name="initial_value">0.0</param> 
[ros2_control_node-2] </state_interface>
```

Tracking down the changelist, I suspect it has to do with https://github.com/ros-controls/ros2_control/pull/1051

sea-bass, 2023-07-10:
Actually, this issue pertains to the description package. Closing for another issue there.

fmauch, 2024-04-10:
This should be fixed in #920

closed 2024-04-10 by fmauch