UniversalRobots/Universal_Robots_ROS2_Driver #740: Ros2

fafux, 2023-07-11:


fmauch, 2023-07-11:
This looks like you are running into #677 and you don't have an up-to-date system. Could you try updating your system and try again?

fafux, 2023-07-11:
I already updated my system but still I get the same error.

fmauch, 2023-07-12:
No need to compile ros2_control from source.

What is the output of `apt list ros-humble-ur-client-library ros-humble-ur-robot-driver ros-humble-controller-manager`?

fafux, 2023-07-12:
Ur robot driver and ur client library are installed from source. 
ur-robot-driver: commit f7b880249161a21cc4906ae6ca56aa6a19fa7c15 (HEAD -> humble, tag: 2.2.8, origin/humble)
ur-client-library: commit 2e69bc165da0ad22fe9a1a3417e7eaa4b082ebac (HEAD -> master, origin/master, origin/HEAD)
ros-humble-controller-manager: /jammy,now 2.28.0-1jammy.20230626.183643 amd64 [installed]

fmauch, 2023-07-12:
Did you try rebuilding from scratch / using the binary installed packages? Because, this really should not be an issue anymore.

fafux, 2023-07-12:
I found out the problem. If the client library is compiled and installed outside of the colcon workspace and in a local directory , I get the error. If I clone it inside the ros2 workspace and compile it with colcon, the problem is solved.

fmauch, 2023-07-12:
Then I'll go ahead and close this. Feel free to comment / reopen.

closed 2023-07-12 by fmauch