UniversalRobots/Universal_Robots_ROS2_Driver #746: Unable to build Universal Robots ROS2 Driver on Humble

adithyakumarcr, 2023-07-17:
Hi community, 

I am unable to build ROS2 Drivers,  I have installed all the dependencies but still I am having issues with building. This wasnt the case couple of months ago. I was able to build the same. Today I cannot. 

```
In file included from /root/colcon_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
/root/colcon_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41:10: fatal error: joint_trajectory_controller/joint_trajectory_controller.hpp: No such file or directory
   41 | #include "joint_trajectory_controller/joint_trajectory_controller.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [0.22s, exited w
```

This is the error I am facing. Can someone help me out. 

Thanks.

fmauch, 2023-07-18:
As the error output suggests, you don't have the `joint_trajectory_controller/joint_trajectory_controller.hpp` header available. Where should it come from? Did you either install `joint_trajectory_controller` from binary packages (e.g. using `rosdep install ...` or build it from source?

Since things seem to be working fine in our CI and our daily workflow, I assume that some step of the installation process wasn't executed correctly. Could you please provide the full list of commands that you performed in order to get to the point of compiling this repo?

PietroDrd, 2023-10-17:
Hi, i faced your same problem and i solved it in the following way:

(My setup: Ubuntu 22.4.3, UR_ROS2 installed from binary, cloned the driver from GitHub in my workspace folder)

First you have to set authorities (ownership) of the files contained in the "opt/ros/humble/**" folder of your linux package:
`sudo chown -R $USER:$USER /opt/ros/humble/ `

Then modify the **joint_trajectory_controller.hpp** file in the opt/ros/humble/inlude/joint_trajectory_controller/joint_trajectory_controller folder, more precisely:

at line 255 insert: 
   `JOINT_TRAJECTORY_CONTROLLER_PUBLIC
    bool has_active_trajectory() const;`

at (current) line 260 sobstitute the entire **publish_state** function as:
`void publish_state(
    const rclcpp::Time & time, const JointTrajectoryPoint & desired_state, 
    const JointTrajectoryPoint & current_state, const JointTrajectoryPoint & state_error);`

save it.

Once done it, i was able to correctly build the whole driver.

fmauch, 2023-10-17:
@PietroDrd That does not seem like a good advice for the following reasons:

* The issue you seem to be facing does not seem to be related to the OP. The OP was missing a header file, while you seem to have an incompatible version installed.
* You seem to be trying to build the main branch against ROS Humble. As the README states. the main branch is targeting Rolling only. If you want to build against Humble, you should use the `humble` branch.
* Is it really required to build from source? Unless you want to join development it should not be necessary to build the driver from source, but rather use the binary packages.
* Modifying files in `/opt/ros/...` never is a pretty good idea. Best case, your changes only get reverted on the next package update. Worst case you introduce some API incompatibilities, that will make other parts of your software stack simply crash.

Since the original author did not report back any problems and I am not aware of any build problems when using the correct branches, I will close this issue.

closed 2023-10-17 by fmauch

PietroDrd, 2023-10-17:
Oh, ok.. sorry for that. 
Thanks for the explanation.
I've hidden my answer to avoid to create problems.