UniversalRobots/Universal_Robots_ROS2_Driver #747: colcon build fails on ur_calibration

francescoo99, 2023-07-18:
Hi everyone, 
in my ROS 2 foxy workspace I cloned the [Universal_Robots_ROS2_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/foxy) (foxy branch) and the [Universal_Robots_Client_Library](https://github.com/UniversalRobots/Universal_Robots_Client_Library) (in my workspace I only have these two repos). 
I followed all the instructions on the  [Universal_Robots_ROS2_Driver README ](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/foxy/README.md)to correctly compile the driver but when I do it I get: **_undefined reference to urcl::UrDriver::UrDriver_** during the ur_calibration package compilation.  What it is due to? 

The complete error is: 

`
undefined reference to `urcl::UrDriver::UrDriver(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (bool)>, bool, std::unique_ptr<urcl::ToolCommSetup, std::default_delete<urcl::ToolCommSetup> >, unsigned int, unsigned int, int, double, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/calibration_correction.dir/build.make:276: calibration_correction] Error 1
make[1]: *** [CMakeFiles/Makefile2:151: CMakeFiles/calibration_correction.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ur_calibration [17.5s, exited with code 2]

`

I'm using Ubuntu 20 with ROS 2 foxy.

francescoo99, 2023-07-18:
I saw the #500 issue but it didn' work for me

fmauch, 2023-07-20:
Hi!

As a disclaimer: ROS 2 Foxy is no longer supported / maintained as it is End of Life.

To answer your question: It could be that you have `ros-foxy-ur-client-library` installed as an apt package. If your linker tries to link the driver against the installed version of `ur_client_library` instead of the one built in your workstation (which can happen, since the installed version comes first in your LD_LIBRARY_PATH), there might be issues regarding ABI compatibility.

Try running `sudo apt remove ros-foxy-ur-client-library` and try again.

closed 2023-07-20 by francescoo99