UniversalRobots/Universal_Robots_ROS2_Driver #748: ur_moveit_config move() failed or timeout bug

TZECHIN6, 2023-07-19:
### Environment
- Ubuntu 22.04.2 LTS
- ROS2 humble (clean build on 2023/07/19)
- UniversalRobots driver humble branch (f7b880249161a21cc4906ae6ca56aa6a19fa7c15)
- Real UR5e arm with urcap (v1.0.5) enable
- rmw cyclone with ros2 cyclonedds tuning (reference to ros2 document)

### Issue
I have tested in ursim which works very well, all the examples can be repeated. Therefore I head to the lab today and try repeating the same example with real ur5e arm. When I try to use the moveit to send goal pose of TCP, and click "plan" .... below errors come out. The planning phase can be done, but the controller fail to move the robot arm due to timeout.

When I click "plan and execute", the robot arm do move a little bit toward the goal pose, but same error will pop up after several seconds.

**Error log**
```
[move_group-1] [WARN] [1689760449.589027588] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[move_group-1] [INFO] [1689760449.591591745] [moveit_move_group_default_capabilities.move_action_capability]: Motion plan was computed successfully.
[rviz2-2] [INFO] [1689760449.592107812] [move_group_interface]: Planning request complete!
[rviz2-2] [INFO] [1689760449.630042890] [move_group_interface]: time taken to generate plan: 0.00608139 seconds
[move_group-1] [INFO] [1689760453.738234622] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-1] [INFO] [1689760453.738326822] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-1] [INFO] [1689760453.738357175] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1689760453.738373989] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[rviz2-2] [INFO] [1689760453.738365562] [move_group_interface]: Execute request accepted
[move_group-1] [INFO] [1689760453.738463640] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-1] [INFO] [1689760453.740708626] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-1] [INFO] [1689760453.740746658] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1689760453.740762447] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1689760453.740890654] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to scaled_joint_trajectory_controller
[move_group-1] [INFO] [1689760453.741390484] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: scaled_joint_trajectory_controller started execution
[move_group-1] [INFO] [1689760453.741406119] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[move_group-1] [WARN] [1689760457.472366589] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: waitForExecution timed out
[move_group-1] [ERROR] [1689760457.472395751] [moveit_ros.trajectory_execution_manager]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 3.730762 seconds). Stopping trajectory.
[move_group-1] [INFO] [1689760457.472443941] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Cancelling execution for scaled_joint_trajectory_controller
[move_group-1] [INFO] [1689760457.472768063] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status TIMED_OUT ...
[move_group-1] [INFO] [1689760457.472840696] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: TIMED_OUT
[rviz2-2] [INFO] [1689760457.473186583] [move_group_interface]: Execute request aborted
[move_group-1] [INFO] [1689760457.491993339] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'scaled_joint_trajectory_controller' successfully finished
[rviz2-2] [ERROR] [1689760457.542131508] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```

Could anyone help me on this? I believe this is a bug issue. This will block my development as I have already written some cpp code to do more complex move in the ursim... which I expect to repeat in the real arm as well. Currently situation blocks my whole development timeline :(

TZECHIN6, 2023-07-20:
For the ros driver, nope actually. It prints out goal request accepted, sth like that.

For the robot log, how can I export a log from the real robot? Sorry I have no experience on this, might need your guide.

TZECHIN6, 2023-07-23:
Is there any thing I need to pay attention to when configurating the real robot in the UR scope? I have followed the guide which telling me to generate a calibration file, how this file being used in Moveit?

So far I am still get stuck at this part... anyway I could further debug this issue or provide hints to the community for help?

fmauch, 2023-07-24:
> https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/395c0541b20d0da2cd480e2ad85b2100410fb043/ur_robot_driver/src/hardware_interface.cpp#L159

Could you please post the full output?

> For the robot log, how can I export a log from the real robot?

- Insert a USB drive into your Teach Pendant
- Clear the log output (see image below)
- Download the support file (see image below)
- In the generated zip you will find a `log_history.txt` file.

[image]

TZECHIN6, 2023-07-24:
I will get back to you as soon as possible as the lab is quite far away from where I live. Thanks for the guide. Btw, do u mean I should enable the non_blocking_mode? Sorry for not so clear about this part and I would like to clarify before going to the lab to ensure I capture most of the log you required.

urfeex, 2025-01-15:
Closing this as no input has been added in >1.5 years.

closed 2025-01-15 by urfeex