UniversalRobots/Universal_Robots_ROS2_Driver #750: URSim Docker Image VNC Random crash

bmdyrdal, 2023-07-20:
Hi,

When using the docker images for URSim, the VNC server often randomly crashes causing the UI to be inaccessible. 
I don't get any log messages except the one in the VNC Client: "Failed to connect to server (code: 1006)".

Both Polyscope and the controller is still working.
I have mostly used version 5.13.1.

Is there a way to force a restart of the VNC server, without having to restart the container?

Best,
Bent Martin

fmauch, 2023-07-20:
Yes, that's a known issue. The URSim docker images are not part of this repo and we are not actively maintaining this, but I know that the docker image maintainers are aware of this and plan to fix this soon.

bmdyrdal, 2023-07-20:
Ok, thanks for the quick reply!

Yeah.. I didn't know where to post this :sweat_smile:

urfeex, 2025-01-15:
This should not happen since quite a while.

closed 2025-01-15 by urfeex