UniversalRobots/Universal_Robots_ROS2_Driver #761: Dashboard client services do not respect receive_timeout parameter

CDietzel, 2023-07-24:
I am using ROS2 Humble, and I have installed the binary driver in ubuntu 22.04 as described in the README. I am connected to a UR5 robot simulated with the CB3 URSim docker image.

After launching the ur_control.launch.py launch file, I then ran the command:

`ros2 param set /dashboard_client receive_timeout 4.0`

Then I tried to load a program on the UR5 dashboard with `ros2 service call /dashboard_client/load_program ur_dashboard_msgs/srv/Load "filename: '/G-253/ros_control_v2.urp'"`

I am currently running ROS2 from inside an agonizingly slow linux VM, so that service call sometimes takes more than 1 second to complete. Thus, I get the following error:
```
waiting for service to become available...
requester: making request: ur_dashboard_msgs.srv.Load_Request(filename='/G-253/ros_control_v2.urp')

response:
ur_dashboard_msgs.srv.Load_Response(answer='Did not receive answer from dashboard server in time. Disconnecting from dashboard server.(Configured timeout: 1 sec)', success=False)
```

As can be seen from the error, the timeout is configured at 1 second despite setting it to 4 seconds previously. I have tried setting the timeout parameter to even higher values and I still sometimes receive the same error. It is clear that the timeout is not actually being set to the desired value. 

Any thoughts?

fmauch, 2023-07-25:
Thanks for reporting this. We'll try to reproduce and investigate this further. If you encounter something in the meantime, please let us know.

bmdyrdal, 2023-07-27:
Experiencing the same. We're setting it in the ur_control.launch.py.
`dashboard_client_node = Node(
        package="ur_robot_driver",
        condition=IfCondition(launch_dashboard_client)
        and UnlessCondition(use_fake_hardware),
        executable="dashboard_client",
        name="dashboard_client",
        output="screen",
        emulate_tty=True,
        parameters=[{"robot_ip": robot_ip, "receive_timeout": 240.0}],
    )
`

bmdyrdal, 2023-07-27:
It looks like its reset to 1 second in the URCL connect function:
```
  // Reset read timeout to "normal" socket timeout
  tv.tv_sec = 1;
  tv.tv_usec = 0;
  TCPSocket::setReceiveTimeout(tv);
```

fmauch, 2023-08-29:
This should get fixed by https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/171

VinDp, 2024-07-10:
Since the PR has been merged and the issue is quite old, I will close this.
Feel free to re-open, if still relevant!

closed 2024-07-10 by VinDp