UniversalRobots/Universal_Robots_ROS2_Driver #763: No control of real robot using WSL2 on Windows

Adrian2702, 2023-07-25:
Hi community,

I am trying to control a UR10 robot (CB3 series; URSoftware 3.12.0.90886) from my Windows 10 PC running WSL2. WSL2 is running Ubuntu 22.04 with ROS2 Humble. I have installed the _Universal_Robots_ROS2_Driver_ (Branch: Humble) and now I want to control the real robot.

## My configuration

When I run `ipconfig` in the Windows command line, I get the following network settings:

IP address of the Windows PC: 192.168.0.211 (Subnet: 255.255.255.0)
IP address of the UR10 robot: 192.168.0.210 (Subnet: 255.255.255.0)
IP address of the vEthernet adapter (WSL): 172.22.144.1 (Subnet: 255.255.240.0)

When I call `ifconfig` or `hostname -I` in the WSL2 prompt, I get the WSL2_IP address: 172.22.148.69

The robot is directly connected to my Windows PC via an Ethernet cable.

As [MiroslavKohut](https://github.com/MiroslavKohut) mentioned in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/486#issuecomment-1096776059, i setup port forwarding in an elevated powershell:

```
netsh interface portproxy add v4tov4 listenport=50001 connectport=50001 connectaddress=172.22.148.69
netsh interface portproxy add v4tov4 listenport=50002 listenaddress=0.0.0.0 connectport=50002 connectaddress=172.22.148.69
netsh interface portproxy add v4tov4 listenport=50003 connectport=50003 connectaddress=172.22.148.69
```

I configured the `reverse_ip` of the `ur_robot_driver` to my Windows-IP address (192.168.0.211). I have configured the inbound and outbound rules for ports 50001 to 50004 in the Windows Firewall rules. Additionally, I added `C:\Windows\System32\bash.exe` and `C:\Windows\System32\wsl.exe` to the firewall whitelist as described in https://stackoverflow.com/questions/63784906/firewall-blocking-wsl-2.

I checked the status of the UFW firewall in WSL2 with `sudo ufw status` and I get the answer: `Status: inactive`.

From the Windows PC, I can ping any IP address (WSL2_IP address and robot IP). When I access the robot controller with `ssh`, I can ping 192.168.0.211 (Win PC). Ping to WSL2 vEthernet adapter (172.22.144.1) or WSL2_IP (172.22.148.69) is not possible. But this should be normal, because they are different networks.

I have disabled PROFINET as well as EtherNet/IP on our robot controller. Also, there is no other URCap installed on the robot (except the externalcontrol-1.0.5.urcap).

I start the robot driver with the following command: 
`ros2 launch ur_robot_driver ur_control.launch.py use_fake_hardware:=false initial_joint_controller:=joint_trajectory_controller`

(Attention! The robot_ip=192.168.0.210 is stored in the script and therefore not visible in the launch command.)

Rviz starts and everything looks pretty good. The robot is displayed in Rviz as the real robot is currently. Then I start the URCap on the UR robot controller and it runs normally. Now I get the INFO in the terminal:

```
[ur_ros2_control_node-1] [INFO] [1690278472.642736258] [UR_Client_Library]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1690278472.642794758] [UR_Client_Library]: Sent program to robot
```
## Issue

I have checked if the joint_trajectory_controller is active. It is. Then I called the joint trajectory controller plugin available in rqt to control the UR robot with the joint_trajectory_controller. When I press the power button in rqt and then move the sliders of each joint, the real robot does not move as expected. The same happens with the scaled_joint_trajectory_controller.

I tried the whole thing under ROS Noetic on a Linux PC, there it worked.

So my question now is if it is possible to control a real robot via WSL2? Could it be that WSL2 blocks outgoing signals via port 50002? Did I miss something in the configuration?

I would be very grateful for any tips. Thank you in advance.

fmauch, 2023-08-02:
While I don't have **any** experience at all using WSL2, the setup you described seems to in line with the linked comment from the ROS1 driver.

However, in your output I am missing `Robot connected to reverse interface. Ready to receive control commands.` After the program has been sent to the robot and it gets executed, it should connect to the driver's reverse interface. At this place the driver should print the above output. If this is not happening, there probably is an issue with your robot reaching the WSL2 port 50001. If it is not able to connect at all, you should see an error popup on the TP:

[image]

If you don't get this, the robot probably is able to make a connection to Windows' port 50001, but this being bound to something else. But then I also expect the robot program to fail (probably with a timeout log and the program stopping / repeating, when configured to loop).

Adrian2702, 2023-08-03:
Thank you @fmauch for your response!

Since you mentioned that port 50001 might be bound to something else, I took a closer look at the network traffic over port 50001. Using `netstat -ano` in an elevated Windows command prompt shows the ports on my Windows machine:

[image]

Entries with a local address of `0.0.0.0` and an associated port should mean that the port is listening on all network interfaces and accepts all incoming connections through this port number. According to the status, there is currently no active connection to a service. (State=Listen)

When I run `netstat -ab` (the -b parameter displays the executable involved in creating each connection or listening port), I get the information that ports 50001 - 50004 are listened by the `iphlpsvc` service. AFAIK the Internet Protocol Helper service is required in Windows to use `netsh interface portproxy`. Without this service, port forwarding to my WSL2 does not work. I also once disabled `iphlpsvc` (see: https://www.minitool.com/news/iphlpsvc.html) and tried to connect to the URCap, but in this case I get the error `connection refused` as shown in your picture. Once I re-enable `iphlpsvc`, I can run the `ur_robot_driver` as before, then start the external URCap and no longer get an error message from URCap. So it looks like the Windows service is important for the communication between host and WLS2. However, I suspect that this does not block the traffic.

The log of the command prompt looks quite good when I run: `ros2 launch ur_robot_driver ur_control.launch.py use_fake_hardware:=false launch_rviz:=false`

```
[INFO] [launch]: All log files can be found below /home/praezzelle/.ros/log/2023-08-03-09-47-10-807438-w10ws0293-2538
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [2543]
[INFO] [controller_stopper_node-3]: process started with pid [2545]
[INFO] [ur_ros2_control_node-1]: process started with pid [2541]
[INFO] [robot_state_publisher-4]: process started with pid [2547]
[INFO] [spawner-5]: process started with pid [2549]
[INFO] [spawner-6]: process started with pid [2551]
[INFO] [spawner-7]: process started with pid [2553]
[INFO] [spawner-8]: process started with pid [2555]
[INFO] [spawner-9]: process started with pid [2557]
[INFO] [spawner-10]: process started with pid [2598]
[dashboard_client-2] [INFO] [1691048831.524258511] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2]
[controller_stopper_node-3] [INFO] [1691048831.529970111] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_publisher-4] [INFO] [1691048831.559324411] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1691048831.559549311] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1691048831.559566711] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1691048831.559576311] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1691048831.559584211] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1691048831.559591711] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1691048831.559599211] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1691048831.559607811] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1691048831.559615411] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1691048831.559623911] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1691048831.559632411] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1691048831.559640011] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1691048831.559648811] [robot_state_publisher]: got segment wrist_3_link
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1691048831.566184110] [resource_manager]: Loading hardware 'ur10'
[ur_ros2_control_node-1] [INFO] [1691048831.570578810] [resource_manager]: Initialize hardware 'ur10'
[ur_ros2_control_node-1] [INFO] [1691048831.571539510] [resource_manager]: Successful initialization of hardware 'ur10'
[ur_ros2_control_node-1] [INFO] [1691048831.573601010] [resource_manager]: 'configure' hardware 'ur10'
[ur_ros2_control_node-1] [INFO] [1691048831.574464410] [resource_manager]: Successful 'configure' of hardware 'ur10'
[ur_ros2_control_node-1] [INFO] [1691048831.575295110] [resource_manager]: 'activate' hardware 'ur10'
[ur_ros2_control_node-1] [INFO] [1691048831.576342310] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1691048831.576392910] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1691048831.579478910] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[ur_ros2_control_node-1] [INFO] [1691048831.965481409] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1691048831.966494709] [UR_Client_Library]: Setting up RTDE communication with frequency 125.000000
[ur_ros2_control_node-1] [INFO] [1691048833.046982804] [URPositionHardwareInterface]: Calibration checksum: 'calib_9576051469539730839'.
[... cut ...]
[ur_ros2_control_node-1] [INFO] [1690278472.642794758] [UR_Client_Library]: Sent program to robot
```
Should the output `Robot connected to reverse interface. Ready to receive control commands.` directly follow after this two lines? Is there no additional error message if the connection to the reverse interface failed? Could one change something in the script to limit the error further?

I am very grateful for any tips or ideas. I think it would be very cool if the UR driver would also work over WSL2. That opens up a lot of great possibilities.

pkoprov, 2024-03-23:
I have a similar issue, but it is from Ubuntu 22. I can connect to my UR10e, and when I move it to reality, it moves to RViz. But when I click to play the external control program, I get this error message:
The connection to the remote PC at 192.168.10.29:5002 could not be established.
Reason: No route to host (Host unreachable)

RobertWilbrandt, 2024-04-10:
@pkoprov this issue so far is concerned with WSL2 specific network problems, your issue seems unrelated. Please open a new issue if this persists for you.

We try to collect some information from previous issues about WSL2 networking in #829. I'll close this issue for now, as there have not been updates for some time, but please feel free to create new issues if you encounter further problems. Any update on successful WSL2 usage would also be highly appreciated.

closed 2024-04-10 by RobertWilbrandt