UniversalRobots/Universal_Robots_ROS2_Driver #769: Consider removing docker dependency from package.xml

firesurfer, 2023-08-01:
I suggest removing 

```
  <test_depend>docker.io</test_depend>
```

As this is a very large dependency where many people prob. want to explicitly decide whether they install it or not. If implicitly installed it can even be a security risk on some systems especially if users are unaware of it.

There is `--skip-keys=SKIP_KEYS` for `rosdep`. But in order to specify this option one has to be aware of what is installed which might not be the case especially for quickly put together testing environments.

RobertWilbrandt, 2023-08-14:
I am not quite sure what to do with this.

We should have probably had this as an `exec_depend` previously, as the `start_ursim` script depended on it. Having it as a `test_depend` does not make sense when we don't call it directly from tests and all tests using ursim are optional through a flag. Seeing that we now have moved the script to the client library i see two options:
- Dropping it as dependency here, but adding it to the client library. This does not really change anything for the user, but would be more consistent.
- Drop the dependency altogether. The script would get logic to determine if docker is missing and warn about that.

I think i'd prefer the second option, as docker is not really needed for the majority of driver usage scenarios. @fmauch what do you think about this?

fmauch, 2023-08-14:
Yes, dropping it from the driver makes sense and making the script generate good output if docker is missing seems like a fair solution. We'll have to remember installing docker in ci manually, then.

fmauch, 2023-08-28:
I'll mark this an enhancement, as this is not necessarily at the top of our priority list.

PR #985 (2024-05-07): Remove dependency to docker.io
The dependency was basically added since this is needed to spawn up ursim. However, the start_ursim script catches the case that no docker executable is available and the tests requiring it are behind a compile definition which is off by default.

So, in order to not install docker for users automatically, the dependency shall be removed.

As a side-effect this has always stopped us from releasing the package for RHEL.

This closes #769
PR #988 (2024-05-07): Remove dependency to docker.io (backport #985)
The dependency was basically added since this is needed to spawn up ursim. However, the start_ursim script catches the case that no docker executable is available and the tests requiring it are behind a compile definition which is off by default.

So, in order to not install docker for users automatically, the dependency shall be removed.

As a side-effect this has always stopped us from releasing the package for RHEL.

This closes #769 
<hr>This is an automatic backport of pull request #985 done by [Mergify](https://mergify.com).
closed 2024-05-07 by fmauch