UniversalRobots/Universal_Robots_ROS2_Driver #776: unsmooth trajectory in when using scaled trajectory controller for hybrid planning

santoshbalaji, 2023-08-15:
The following are the setup I have currently 

ROS2 Humble
Moveit2 Hybrid planner
UR Driver with Scaled joint trajectory controller 
UR16e robot

When I try to use hybrid planner to send real UR robot to a specific point the executed motion seems more jerky and it is not smooth. When I try to do some analysis the following is observed.

> Local planner (Hybrid planner) -> Trajectory (Start point) -> Scaled joint trajectory Controller
 
> Local planner (Hybrid planner) -> Trajectory (Via point) -> Scaled joint trajectory Controller

> Local planner (Hybrid planner) -> Trajectory (Via point) -> Scaled joint trajectory Controller

> Local planner (Hybrid planner) -> Trajectory (Goal point) -> Scaled joint trajectory Controller

Because of the above operations the joint trajectory controller was not able to maintain constant velocity as these controllers looks like designed to get all positions in one single step to maintain constant velocities. Let me know if above assumptions are correct. What can be done to mitigate this issue ?. I was thinking of modifying the controller to queue up the commands and keep planning to maintain constant velocities. Require suggestion on directions. 

Thanks in advance !!!!

RobertWilbrandt, 2023-08-21:
Unfortunately i don't have any experience with MoveIt hybrid planning myself. Could you please provide the steps that you used to arrive at this behavior? And can you provide how exactly the local planner solution gets sent to the JTC?

An easy way to pin down if this issue is related to our scaled JTC would be to just use the `joint_trajectory_controller` controller instead of `scaled_joint_trajectory_controller`. In general these controllers should be able to handle new goals correctly, documentation on how that works can be found [here](https://control.ros.org/master/doc/ros2_controllers/joint_trajectory_controller/doc/userdoc.html) (specifically under "Preemption").

urfeex, 2025-01-15:
Closing this due to inactivity

closed 2025-01-15 by urfeex