UniversalRobots/Universal_Robots_ROS2_Driver #777: publish_state : candidate expects 3 arguments, 4 provided

Tangshp163, 2023-08-18:
/home/tangshp/code/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/tangshp/code/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:274:16: error: no matching function for call to ‘ur_controllers::ScaledJointTrajectoryController::publish_state(const rclcpp::Time&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&)’
  274 |   publish_state(time, state_desired, state_current, state_error);
      |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/tangshp/code/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
                 from /home/tangshp/code/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
/home/tangshp/code/ros_ur_driver/install/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:257:8: note: candidate: ‘void joint_trajectory_controller::JointTrajectoryController::publish_state(const JointTrajectoryPoint&, const JointTrajectoryPoint&, const JointTrajectoryPoint&)’
  257 |   void publish_state(
      |        ^~~~~~~~~~~~~
/home/tangshp/code/ros_ur_driver/install/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:257:8: note:   candidate expects 3 arguments, 4 provided
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76：CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] 错误 1
gmake[1]: *** [CMakeFiles/Makefile2:223：CMakeFiles/ur_controllers.dir/all] 错误 2
gmake: *** [Makefile:146：all] 错误 2
---
Failed   <<< ur_controllers [17.6s, exited with code 2]
Aborted  <<< admittance_controller [59.1s]                                         

Summary: 35 packages finished [7min 24s]
  1 package failed: ur_controllers
  1 package aborted: admittance_controller
  2 packages had stderr output: ur_client_library ur_controllers
  4 packages not processed

Tangshp163, 2023-08-22:
> Can you please provide more information on this error? In particular:
> 
> * Which ros distribution are you using?
> * Were you following our instructions on [building from source](../#build-from-source)?
> * Did you build `ros2_control` and `ros2_controllers` from source?
> * Which branches are you on for all relevant packages?
> 
> This looks vaguely like the problem in #674, but without more information it is impossible to tell.

> Can you please provide more information on this error? In particular:
> 
> * Which ros distribution are you using?
> * Were you following our instructions on [building from source](../#build-from-source)?
> * Did you build `ros2_control` and `ros2_controllers` from source?
> * Which branches are you on for all relevant packages?
> 
> This looks vaguely like the problem in #674, but without more information it is impossible to tell.

Ros distribution is humble.
Follow the Build from source in the readme
Both ros2_control and ros2_controllers are in Universal_Robots_ROS2_Driver
main branch

Tangshp163, 2023-08-23:
> Building for humble with `ros2_control` and `ros2_controllers` on main branch won't work, you need to go to their respective `humble` branches for that (just as you need to use the `humble` branch of this repo). Even more so, you should not need to build them from source at all (step 5 should not be necessary). Be aware that the README has the distribution-aware build steps on the corresponding branches, so [the humble README](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/humble#build-from-source) lists the relevant steps for you.
> 
> Can you check if changing the repo branches fixes the build for you?

Thanks. The humble branch can be compiled successfully, but after using launch (ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3), plan&execute always fails in rivz
file:///home/tangshp/%E5%9B%BE%E7%89%87/2023-08-23_08-42.png

RobertWilbrandt, 2023-08-24:
What exactly is your setup for this:
- Are you using a real robot or ursim? If so, did you setup the robot [as described in our documentation](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/install_urcap_cb3.html)? And is the program running on the robot when you try to move it?
- Which part fails exactly? Does planning work and execution fail? Please provide the console error output for further investigation.
- Are you able to move the robot without MoveIt? Can you e.g. test if [rqt_joint_trajectory_controller](https://index.ros.org/p/rqt_joint_trajectory_controller/github-ros-controls-ros2_controllers/#humble) works for manual motion?

VinDp, 2024-02-12:
Hi @Tangshp163, 
since some time has passed since our reply I will close this issue. If it is still relevant, feel free to reopen it.

closed 2024-02-12 by VinDp