UniversalRobots/Universal_Robots_ROS2_Driver #788: Extracted Calibration Information not Utilized by tf2

cpsiff, 2023-08-30:
I have installed the UR Robot Driver, extracted the calibration information from the robot and set the driver up to use that information by changing the kinematics_params parameter in ur_control_launch.py as described in [this issue](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/711). 

I have written a node that reads the current end effector pose using tf2_ros.Buffer.lookup_transform. A minimal example of my node is below:

```
import rclpy
import tf2_ros
from rclpy.node import Node

class TFTest(Node):
    def __init__(self):
        super().__init__('tf_test')

        self.tf_buffer = tf2_ros.Buffer()
        self.tf_listener = tf2_ros.TransformListener(self.tf_buffer, self)

        self.transform_timer = self.create_timer(0.1, self.transform_callback)

    def transform_callback(self):
        try:
            self.last_tf = self.tf_buffer.lookup_transform('base', 'tool0', rclpy.time.Time())
            self.tf_updated = True
        except tf2_ros.ConnectivityException as e:
            self.get_logger().warn("Failed to get transform from base to end effector")

        if self.last_tf:
            print(self.last_tf.transform.translation)

def main(args=None):
    rclpy.init(args=args)
    tf_test = TFTest()
    rclpy.spin(tf_test)

    tf_test.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

The output of the node is something like:

> geometry_msgs.msg.Vector3(x=0.11440437456172814, y=-0.4682725706540506, z=0.5879769079364487)

whereas the teach pendant displays something like:

> (115.05, -468.18, 584.83)

The coordinates output by the above node do not seem to be affected by what calibration information I use, leading me to believe they're not using it at all. Is this intended behavior? Is there another way I should be getting the transform to the end effector, or is this a bug in the driver / tf2?

Thank you

cpsiff, 2023-08-30:
As a debugging step, I also copied the contents of my custom kinematics.yaml file into the Universal_Robots_ROS2_Description/config/ur5/default_kinematics.yaml file, and I still see no change in the output end effector pose.

cpsiff, 2023-08-30:
After some more debugging, it seems like part of the issue is inconsistent messages being published to the /tf topic.

I noticed that my example node above actually irregularly alternates between two values, with the less common of the two being pretty close to what's on the teach pendant:

> ee pos: 0.11080233668334363 -0.4717967404531766 0.5568064775808639 (less common)
> ee pos: 0.11018918709475961 -0.47206304103057 0.5598878236787377 (more common, incorrect)

This seems to be because the transforms published to /tf are not consistent (even though the robot is not moving). The more common, incorrect end effector position is returned when the most recent message on the /tf topic has had some rotation components rounded to 0.0. For example, here is what the first transform looks like when the incorrect end effector pose is returned:

> frame_id: upper_arm_link
child_frame_id: forearm_link
transform: geometry_msgs.msg.Transform(translation=geometry_msgs.msg.Vector3(x=-0.4251354753856735, y=0.0, z=0.0), rotation=geometry_msgs.msg.Quaternion(**x=0.0, y=0.0**, z=0.7511925329895845, w=0.6600831602008127))

here is what the same transform looks like when the more correct end effector pose is returned:

> frame_id: upper_arm_link
child_frame_id: forearm_link
0 : geometry_msgs.msg.Transform(translation=geometry_msgs.msg.Vector3(x=-0.4251354753856735, y=0.0, z=0.0), rotation=geometry_msgs.msg.Quaternion(**x=0.0006167428612089586, y=-0.0002544616729952269,** z=0.7511917598096455, w=0.6600837029283393))

A similar problem seems to occasionally occur for the fourth transform (forearm link to wrist_1_link). These always seem to occur at the same time (i.e. the rotation components are rounded to 0.0 in both the upper_arm to forearm and the forearm to wrist_1 transformation)

fmauch, 2023-09-08:
Thanks for reporting this, this looks like it would have to be further investigated. I consider the calibration correction one of the core features of this driver and if it isn't working correctly, we'll have to fix it!

fmauch, 2023-09-08:
A quick check with URSim: I get a consistent

```
geometry_msgs.msg.Vector3(x=-0.14396865671352146, y=-0.4356200608031923, z=0.20203002542838502)
(3.1166393514839523, 0.02508103277980295, -3.141121755220579)
```
with a corresponding 
[image]

so not all seems to be wrong. However, I'll try to do a real-robot test next Monday and check again. Note: I've extended your script a little bit in order to also print the orientation:

```python
import rclpy
import tf2_ros
from rclpy.node import Node
from tf_transformations import euler_from_quaternion

class TFTest(Node):
    def __init__(self):
        super().__init__('tf_test')

        self.tf_buffer = tf2_ros.Buffer()
        self.tf_listener = tf2_ros.TransformListener(self.tf_buffer, self)

        self.transform_timer = self.create_timer(1.0, self.transform_callback)

    def transform_callback(self):
        try:
            self.last_tf = self.tf_buffer.lookup_transform('base', 'tool0', rclpy.time.Time())
            self.tf_updated = True
        except tf2_ros.ConnectivityException as e:
            self.get_logger().warn("Failed to get transform from base to end effector")

        if self.last_tf:
            print(self.last_tf.transform.translation)
            orientation_q = self.last_tf.transform.rotation
            orientation_list = [orientation_q.x, orientation_q.y, orientation_q.z, orientation_q.w]
            print(euler_from_quaternion(orientation_list))

def main(args=None):
    rclpy.init(args=args)
    tf_test = TFTest()
    rclpy.spin(tf_test)

    tf_test.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

@cpsiff the output in your second post do not really seem to surprise, though. Fluctuations in the sub-mm range are from my perspective unavoidable, since the reported joint values are also a bit noisy. However, 0.7mm deviation maybe seems a bit high for this explanation, joint values shouldn't be noisy in that precision range from my expecations.

However, the discrepancy between the reported tf and the values on the TP seem not right.... As said, this needs a deeper look with a real robot next week.

cpsiff, 2023-09-10:
@fmauch Thank you for looking into this!

Note that in my [second post](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/788#issuecomment-1699386314), the fluctuation is 3mm in the z direction, even though the robot is not receiving any movement commands, and these fluctuations are happening something like a few times a second (say one in 20 messages at 75Hz). I agree that sub-mm fluctuations would not be concerning (you can see these on the teach pendant itself, for example), but these are larger fluctuations and the nature of it indicates to me that it's not coming from hardware. It's not random fluctuation around a mean, it's alternating between two distinct points.

The most concerning thing to me is that what calibration information (or lack of it) I use seems to have no effect on the reported end effector pose. I'm curious if you'll find the same thing. There's a chance that I'm doing something wrong with loading in the calibration information - I don't know how to check that it's working other than checking if the reported end effector poses change.

I have opened the calibration information file and confirmed that the information is in there (it's the same as it was in ROS 1).

fmauch, 2023-09-11:
@cpsiff I tested things with one of our robots today and I cannot reproduce the behavior that you are describing.

The tcp pose reported by the driver matches the one from the TP exactly when I launch it with the extracted calibration given. Since you don't experience any changes from using the calibration information: Can you confirm that you don't see the error

```
 [ERROR] [1694437624.484381456] [URPositionHardwareInterface]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract th
e correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_calibration/README.md] for details.
```

If everything is correct, you should see something like
```
[INFO] [1694437690.406932381] [URPositionHardwareInterface]: Calibration checksum: 'calib_1588877523983968359'.
``` 
in the logs.

Did you extract the calibration (and start the driver) while the robot was running (Motors switched on, brakes not necessarily released). Apparently that makes a difference, as calibration information is (internally) read by the controller while turning on as far as I have understood. We'll update the calibration_extraction instructions and code regarding this, but in the meantime it would be interesting whether that makes a difference for you.

If you keep experiencing problems, could you please post the complete output of your calibration extraction and your driver run?

cpsiff, 2023-09-11:
@fmauch I have solved the problem - it was on my end. As far as I can tell there is nothing wrong with the driver.

My problem was that the launch file I was using (which launches ur_control.launch in addition to other things) included a robot state publisher, i.e.:
```python
Node(
    package='robot_state_publisher',
    name='robot_state_publisher',
    executable='robot_state_publisher',
    output='screen',
    arguments=[urdf_path],
    parameters=[{'publish_frequency': 50.0, 
                'robot_description': urdf_string}]
),
```
this was a remnant of some copy-pasted code that I didn't check carefully. The state publisher was publishing uncalibrated robot poses on the same topic. Once I removed it, my issue was resolved.

I also tested what happens if I try to read the calibration information at a different time. If the robot is off, the calibration script just hangs up on "[UR_Client_Library]: Failed to read from stream, reconnecting in 1 seconds...". If the robot is on at all (even before pressing "go to initialization screen" on the teach pendant), the calibration is extracted successfully. I don't think there's any way for a user to think they have good calibration info when they don't.

I do think there should be better documentation around how to use the extracted calibration information (as explained in #711) and confirm that it's working. If I had known to look in the console log for the calibration checksum, that could have saved me a lot of debugging time.

It might be worth reopening #791 as I doubt their underlying cause was the same as mine.

Thank you for your help.

closed 2023-09-11 by cpsiff