UniversalRobots/Universal_Robots_ROS2_Driver #789: joint_trajectory_controller inactive with ursim

linmeon, 2023-08-31:
Hi:
ROS2 Humble Ubuntu22.04 ur3
with doc https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/usage.html

i am testing ursim and moveit

when i run 
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=<UR_TYPE> robot_ip:=<IP_OF_THE_ROBOT> launch_rviz:=true
`
and I control the robot using VNC, the robot in RViz also moves. However, it doesn't work when I use MoveIt to plan and execute.

so  i  run
`ros2 control list_controllers`
shows 
`speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active
**joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] inactive**
forward_position_controller[position_controllers/JointGroupPositionController] inactive
io_and_status_controller[ur_controllers/GPIOController] active`
then i close all 
run 
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true
`
The control through VNC cannot synchronize with RViz, but the planning in MoveIt is consistent with RViz.
the  run
`ros2 control list_controllers`
shows
`speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active
**joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] active**
forward_position_controller[position_controllers/JointGroupPositionController] inactive
io_and_status_controller[ur_controllers/GPIOController] active`

I also tried the example 
`ros2 launch ur_robot_driver test_joint_trajectory_controller.launch.py
`
same result 

i also tried 
`Fake hardware on ROS2 Galactic[](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/usage.html#fake-hardware-on-ros2-galactic)
Currently, the scaled_joint_trajectory_controller does not work on ROS2 Galactic. There is an [upstream Merge-Request](https://github.com/ros-controls/ros2_control/pull/635) pending to fix that. Until this is merged and released, please change the default controller in the [controllers.yaml](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/ur_moveit_config/config/controllers.yaml) file. Make sure that the default field is assigned true for the joint_trajectory_controller and false for the scaled_joint_trajectory_controller.`
also not work 
what should i do?

thank you!

fmauch, 2023-08-31:
You'll have to start the external_control program to make the controller running. Skimming over the documentation I think this isn't clearly stated after the last iteration. In short, install the [URCap and create a program on the robot](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/installation/install_urcap_e_series.html) and then start this program after the driver has been started.

fmauch, 2023-09-01:
@linmeon would the changes in #790 have been of any help for you?

PR #1110 (2024-10-01): Improve usage documentation
closes #934, closes #789 and closes #1044.

This should hopefully significantly improve the usage documentation for end users and clarify a couple of things.

I'm not done, yet, hence the draft status.
closed 2024-10-01 by fmauch