UniversalRobots/Universal_Robots_ROS2_Driver #791: Wrong pose got when using TF

WundtAsim, 2023-09-05:
Hi, I use:
ROS2 version: humble (I also use the "humble" branch)
ROBOT: UR10e

First, I run `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.1.24 launch_rviz:=false`

then, I want to echo the pose information about the flange, so I run `ros2 run tf2_ros tf2_echo base flange`

then I got this:
```
At time 1693910819.12247712
- Translation: [0.246, -0.249, 0.619]
- Rotation: in Quaternion [0.170, 0.472, -0.074, 0.862]
- Rotation: in RPY (radian) [0.421, 0.994, 0.059]
- Rotation: in RPY (degree) [24.108, 56.975, 3.383]
- Matrix:
  0.544  0.288  0.788  0.246
  0.032  0.931 -0.363 -0.249
 -0.838  0.223  0.497  0.619
  0.000  0.000  0.000  1.000
```

while, the pose information on the teaching pendant was like: 
```
TCP
x: 243.69mm
y: -249.04mm
z: 618.32mm

Rx: 149.49 degree
Ry: -12.78 degree
Rz: 72.64 degree
```
Apparently, the position coordinate was almost right, but the rotation in RPY I got from ros2 was wrong.

what should I do to get the right pose from tf?
thank you for your answer~

fmauch, 2023-09-08:
@WundtAsim I'll close this as duplicate of #788, but thanks for bringing my attention to this once more.

fmauch, 2023-09-12:
Since #788 was resolved to be not an issue with the driver, I'll reopen this.

@WundtAsim In order to get the same values, please make sure you

- Use a [calibrated URDF](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/installation/robot_setup.html#extract-calibration-information). 
- Make sure that on the TP you select "Base" as reference and not "View" or "Tool". 
- Make sure to check `tool0`, not `flange`. Flange and tool0 use different orientation standards.
- Make sure to look at the same rotation convention. The TP by default shows the Rotation vector notation. For example, on the shell I get 
  ```
  - Translation: [-0.144, -0.436, 0.202]
  - Rotation: in Quaternion [-0.000, 1.000, 0.012, 0.013]
  - Rotation: in RPY (radian) [3.117, 0.025, -3.141]
  - Rotation: in RPY (degree) [178.570, 1.437, -179.973]
  - Matrix:
   -1.000 -0.001  0.025 -0.144
   -0.000  1.000  0.025 -0.436
   -0.025  0.025 -0.999  0.202
    0.000  0.000  0.000  1.000
  ```
  In the default view I see this:
  [image]
  However, when you click on the rotation fields, you'll get to another view where you can select "RPY (rad)" This will show the same values as ROS:
  [image]

I hope this explanation helps.

WundtAsim, 2023-09-12:
Thanks！ 
- I just tried 'tool0' and realized my error, Now I can get the right RPY degree which carries an error of 0.003 [rad] level. 
- I have extracted the calibration information file to my home path, but the position in ros2 still has an error of 2mm. I have tried the way #711 showed, but the error still be there. how can I confirm the calibration file is working?@fmauch 
(Btw, I didn't modify any code in this repository)

fmauch, 2023-09-13:
I've explained [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/788#issuecomment-1713854051) how to verify that the correct calibration is actually being used.

WundtAsim, 2023-09-13:
Thank you all for your support in addressing this issue. It has been determined that the problem I encountered was related to the calibration process on my end.

Appreciate to @fmauch and @cpsiff  for providing assistance in troubleshooting the issue.

closed 2023-09-13 by WundtAsim

fmauch, 2023-09-18:
@WundtAsim Any hints what we could improve in the documentation to avoid things like this in future?