UniversalRobots/Universal_Robots_ROS2_Driver #793: Not able to build UR driver

Maxeva, 2023-09-08:
Hey,

I’m trying to build Universal Robots ROS 2 Driver on Ubuntu 22.04 Humble to control a UR10e with MoveIt2.
Unfortunately I'm facing issues: 

```
Starting >>> ur_dashboard_msgs
Starting >>> ur_moveit_config
Finished <<< ur_moveit_config [2.09s]                                 
Finished <<< ur_dashboard_msgs [21.4s]                       
Starting >>> ur_controllers
--- stderr: ur_controllers                               
/home/max/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/max/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:160:7: error: ‘has_active_trajectory’ was not declared in this scope
  160 |   if (has_active_trajectory()) {
      |       ^~~~~~~~~~~~~~~~~~~~~
/home/max/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:272:16: error: no matching function for call to ‘ur_controllers::ScaledJointTrajectoryController::publish_state(const rclcpp::Time&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&, joint_trajectory_controller::JointTrajectoryController::JointTrajectoryPoint&)’
  272 |   publish_state(time, state_desired, state_current, state_error);
      |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/max/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
                 from /home/max/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
/opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:257:8: note: candidate: ‘void joint_trajectory_controller::JointTrajectoryController::publish_state(const JointTrajectoryPoint&, const JointTrajectoryPoint&, const JointTrajectoryPoint&)’
  257 |   void publish_state(
      |        ^~~~~~~~~~~~~
/opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:257:8: note:   candidate expects 3 arguments, 4 provided
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [15.9s, exited with code 2]
                                 
Summary: 2 packages finished [37.6s]
  1 package failed: ur_controllers
  1 package had stderr output: ur_controllers
  3 packages not processed
```
Any input would be appreciated. Thanks!

fmauch, 2023-09-08:
This looks like you are trying to build the main branch on Humble. If you use humble, please use the humble branch.

However, just in case you haven't noticed: It's not necessary to build from source, you can install the binary package, see https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#install-from-binary-packages.

VinDp, 2024-01-30:
Hi @MadMaxF, 
do you have any updates? Otherwise I will close this issue.

closed 2024-02-12 by VinDp