UniversalRobots/Universal_Robots_ROS2_Driver #794: How to get joint torque of robot?

sm3304love, 2023-09-09:
When I subscribe to the /joint_states topic, I see the effort data. However, I'm not sure whether this data represents motor current or torque. Can you Please let me know what this data means?  Additionally, if this data represents current, how can I determine the torque constant K?

torque = K * current

TZECHIN6, 2023-10-08:
according to the message description in [here](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/JointState.html). The unit should be in Nm or N, I don't know if that already means torque.

I am also working on calculating the energy that has been used for that planning trajectory. Thus this information is also essential for my study.

VinDp, 2024-08-07:
Since the issue is quite old and an answer was given, I will close it. 
Feel free to re-open if still relevant.

closed 2024-08-07 by VinDp

marip8, 2024-11-14:
Can we re-open this issue (or transfer to the UR clients library repo and re-open)? Per the discussion, the question "how do I get joint torques?" was not actually answered; only an explanation of what the values in the effort field represent (i.e., motor current) was provided. I also have an application where I need joint torque feedback, and it took me a while to find this post and understand that the reported effort values are not joint torques.

A quick search shows a few others have attempted to answer this question (see [here](https://forum.universal-robots.com/t/correct-method-to-get-the-joints-torques/34685) and [here](https://www.researchgate.net/figure/Torque-constant-for-each-joint-of-the-UR3-and-the-UR5-used-in-the-experimental-campaign_tbl1_352860483)), but it doesn't seem like these were official answers from UR and it's hard to know whether those answers are accurate.

Is it possible to update this driver to report actual joint torques? It appears that the client library may only report actual motor current, but it seems like the torque constants of the motors should be available somewhere on the controller