UniversalRobots/Universal_Robots_ROS2_Driver #796: initial_positions seems to be having no effect

ravijo, 2023-09-10:
I am using `ur_robot_driver` on ROS Humble (Ubuntu 22.04.3 LTS). Based on [the documentation](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#using-moveit), the `initial_positions` can be used to set the initial joint values for the UR arm. But it seems to be having no effect.

#### Objective
I am trying to generate arm trajectory in  _simulation (i.e., no real robot)_ using MoveIt APIs with the UR arm.

#### Steps
1. I tried following the [Mock hardware](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#mock-hardware) section but it was reporting following errors:
  ```
  $ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_mock_hardware:=true \
                launch_rviz:=false initial_joint_controller:=joint_trajectory_controller
  [ur_ros2_control_node-1] [ERROR] [1694335857.294092578] [UR_Client_Library]: Failed to get address for yyy.yyy.yyy.yyy:30002
  [ur_ros2_control_node-1] [INFO] [1694335857.833236460] [io_and_status_controller]: Waiting for system interface to initialize...
  ```
2. Later, by following [this post](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/301#issuecomment-1151054415) that suggest using `use_fake_hardware`, I was able to get rid of the above errors.
3. Finally, based on [the documentation](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#using-moveit), I tried setting the `initial_positions` but no sucess. Below is my launch file:

  ````py
  from launch import LaunchDescription
  from launch_ros.actions import Node
  from launch.substitutions import PathJoinSubstitution
  from launch_ros.substitutions import FindPackageShare
  from launch.actions import IncludeLaunchDescription
  from launch.launch_description_sources import PythonLaunchDescriptionSource
  from ur_moveit_config.launch_common import load_yaml
  
  
  def generate_launch_description():
      ur_driver_launch = IncludeLaunchDescription(
          PythonLaunchDescriptionSource(
              PathJoinSubstitution([FindPackageShare("ur_robot_driver"), "launch", "ur_control.launch.py"])
          ),
          launch_arguments={
              "ur_type": "ur3e",
              "robot_ip": "yyy.yyy.yyy.yyy",
              "use_fake_hardware": "True",
              "launch_rviz": "False",
              "initial_joint_controller": "scaled_joint_trajectory_controller",
              # loading from YAML does not work
              # "initial_positions": load_yaml("my_pkg", "config/initial_positions.yaml"),
              # directly assigning a dict also does not work
              "initial_positions": {
                  "shoulder_pan_joint": 1.0,
                  "shoulder_lift_joint": -0.5,
                  "elbow_joint": 1.0,
                  "wrist_1_joint": -2.0,
                  "wrist_2_joint": -1.5,
                  "wrist_3_joint": -0.3,
              },
          }.items(),
      )
  
      ur_moveit_launch = IncludeLaunchDescription(
          PythonLaunchDescriptionSource(
              PathJoinSubstitution([FindPackageShare("ur_moveit_config"), "launch", "ur_moveit.launch.py"]),
          ),
          launch_arguments={
              "ur_type": "ur3e",
              "launch_rviz": "True",
          }.items(),
      )
  
      return LaunchDescription([ur_driver_launch, ur_moveit_launch])
  ````

#### Issues
1. The [Mock hardware](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#mock-hardware) section needs more testing or explanation. At present, it seems not working as reported above.
2. The `initial_positions` seems to be having no effect.

ShahhhVihaan, 2024-02-27:
@ravijo  I'm dealing with the same problem. If you were able to find a solution I'd appreciate some help. Also @fmauch do you have any ideas as to why the flag isn't working?

Spartan-Velanjeri, 2024-03-02:
+1, facing the same issue

jozefkosecky, 2024-03-02:
the same issue

VinDp, 2024-07-10:
Hey @ravijo, 
sorry for the late answer, but contrary to what the documentation states, it is not possible to specify the `initial_positions` as launch argument for `ur_control.launch.py`. It apparently was a legacy missing feature, so thanks for spotting it, we will soon update the documentation.
The way in which you can change them is to go in the [config folder](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/rolling/config/initial_positions.yaml) of the description package and change the values in `initial_positions.yaml`.

Instead, regarding mock_hardware: for Humble the argument is `use_fake_hardware` instead of `use_mock_hardware`. I tested it and it seems to work.

VinDp, 2024-08-07:
Since there are no follow ups and the issue is solved, I will close it.
Feel free to re-open if still relevant.

closed 2024-08-07 by VinDp