UniversalRobots/Universal_Robots_ROS2_Driver #799: Improve tutorial on calibration extraction

fmauch, 2023-09-13:
There are a couple of things that can be improved:

- During the extraction make sure that the robot is actually switched on (at least in IDLE)
- Explain how to verify that the correct calibration is actually being used
- Explain how to properly use it (create an own description / launch package)

Related issues:
- https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/711
- https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/788
- https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/791

cpsiff, 2024-12-17:
Hello, I'd like to help with this but I'm not sure where I can make the changes to make a pull request. When I click "edit on GitHub" on [this link](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/robot_setup.html#extract-calibration-information) I get a 404 error (directed [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Documentation/blob/main//doc/ur_robot_driver/ur_robot_driver/doc/installation/robot_setup.rst)). Can you direct me to where I can make edits to that doc?

I was thinking at the very least it would be helpful to add documentation to the bottom of [this page](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/robot_setup.html#extract-calibration-information) describing how to pass the extracted calibration as a parameter to ur_robot_driver (as explained in #918) and how to verify the checksum by searching the command line output. AFAIK the only way to know to do these things currently is by reading the API or looking through Github issues.

urfeex, 2024-12-18:
Hi @cpsiff thank you for reaching out, any help would be very appreciated! Sources for the page you referred to should be that: 
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/doc/installation/robot_setup.rst

While being in there, we could add a `:github_url: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/doc/installation/robot_setup.rst` at the top in order to fix that edit link. That should, however probably be done for every page, so that might be a good task for a separate PR.

To build the complete documentation, please see https://github.com/UniversalRobots/Universal_Robots_ROS_Documentation

closed 2024-12-20 by urfeex