UniversalRobots/Universal_Robots_ROS2_Driver #800: UR5e moves to (0,0,0,0,0,0) during normal execution

iydv, 2023-09-13:
### Affected ROS2 Driver version(s)

latest humble commit 1828fe02e7a6b15c3f16097975ed1a10e40015c5

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel, Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot, Real robot

### Robot SW / URSim version(s)

5.13

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
We are using UR5e for a small picking demonstration. We are sending basic trajectories to move the robot inside a box, pick part, move outside and place in another box. Trajectories are tested and verified on ros2 control simulator. 

## Issue details
Once a new trajectory goal is published by our node to the driver, instead of moving along desired trajectory, robot suddenly moves to initial 0,0,0,0,0,0 position with very high speed. This behavior occurs randomly and is completely unpredictable. Here is a detailed description of our setup:

1) We use Ubuntu 22.04 with PREEMPT_RT kernel 6.1.46. ROS2 humble is obtained from latest available binary packages. The limits in /etc/security/limits.conf also correctly configured. We have also tested on standard 6.2 kernel without realtime and got same result.
2) We have built both ROS2 driver and Client Library from source. We used latest commit in humble branch for the driver and latest commit for Client Library.
3) We are using scaled_joint_trajectory_controller . We are publishing simple trajectories using node with rclcpp_action client. Each trajectory is either 2 points (start joint state, goal joint state and duration) or 6 points with spline time parameterization.
4) We have installed latest update SW 5.13 on the PolyScope.
5) Robot is running in remote mode. We have installed External Control URcap v1.0.5 on the PolyScope.
6) We start robot (power, break release etc) using dashboard ROS2 services. We load and play the URCap program also using dashboard ROS2 services.
7) Robot controller is connected to the PC using single dedicated network cable.

The error occured in following cases:
1) Normal operation. We are sending a trajectory of 6 points and instead of executing it, the robot shows error attached in the log and moves to 0,0,0,0,0,0 with high speed.
2) We stopped the robot using emergency button on teach panel. Then we resume the robot power. Stop URCap program and play it again. When the new trajectory is published the robot moves to to 0,0,0,0,0,0 with high speed. This occurs randomly and typically the robot continues operate normally after stop.
3) We stop and start the driver multiple times during the day. If the driver did not started correctly (no indication in the log) same error occurs. 
4) We have switched robot to the local control, moved robot using teach panel, switched back to remote mode, started URCap program using dashboard play service. 
## Steps to Reproduce
The error occurs randomly, you can try to replicate cases above and send single trajectory of >6 points. 

## Expected Behavior
According to the error log the driver lost connection or not getting correct feedback from the robot. If the connection to the robot is lost/interrupted the driver should cancel motion and stop. The robot should never randomly move to 0,0,0,0,0,0 position in any circumstance. 

## Actual Behavior
The robot starts moving to 0,0,0,0,0,0 position at very high speed. 

# Workaround Suggestion
We have not tested to run in headless mode. But our current configuration is not safe to use.

### Relevant log output

```shell
Sep 12 13:21:23  [ur_ros2_control_node-1] [INFO] [1694517683.420553366] [scaled_joint_trajectory_controller]: Received new action goal
Sep 12 13:21:23  [ur_ros2_control_node-1] [INFO] [1694517683.420580925] [scaled_joint_trajectory_controller]: Accepted new action goal
Sep 12 13:21:27  [ur_ros2_control_node-1] [INFO] [1694517687.421991020] [scaled_joint_trajectory_controller]: Goal reached, success!
Sep 12 13:21:27  [ur_ros2_control_node-1] [INFO] [1694517687.615490226] [io_and_status_controller]: Setting digital output '4' to state: '1'.
Sep 12 13:21:27  [ur_ros2_control_node-1] [INFO] [1694517687.715651625] [io_and_status_controller]: Setting digital output '5' to state: '1'.
Sep 12 13:21:27  [ur_ros2_control_node-1] [INFO] [1694517687.826102946] [scaled_joint_trajectory_controller]: Received new action goal
[... 267 lines cut ...]
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.803944621] [tolerances]: Path state tolerances failed:
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.803948859] [tolerances]: Position Error: -0.552150, Position Tolerance: 0.200000
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.803975183] [tolerances]: Path state tolerances failed:
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.803981340] [tolerances]: Position Error: 0.221268, Position Tolerance: 0.200000
Sep 12 13:21:30  [ur_ros2_control_node-1] [INFO] [1694517690.804131093] [io_and_status_controller]: Configure UR gpio controller with tf_prefix:
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805794439] [tolerances]: Path state tolerances failed:
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805806854] [tolerances]: Position Error: 0.290558, Position Tolerance: 0.200000
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805814448] [tolerances]: Path state tolerances failed:
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805818513] [tolerances]: Position Error: -0.669092, Position Tolerance: 0.200000
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805824779] [tolerances]: Path state tolerances failed:
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805828827] [tolerances]: Position Error: 0.643898, Position Tolerance: 0.200000
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805834427] [tolerances]: Path state tolerances failed:
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805838407] [tolerances]: Position Error: -0.270121, Position Tolerance: 0.200000
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805843697] [tolerances]: Path state tolerances failed:
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805847569] [tolerances]: Position Error: -0.552150, Position Tolerance: 0.200000
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805852823] [tolerances]: Path state tolerances failed:
Sep 12 13:21:30  [ur_ros2_control_node-1] [ERROR] [1694517690.805856786] [tolerances]: Position Error: 0.221267, Position Tolerance: 0.200000
Sep 12 13:21:30  [ur_ros2_control_node-1] [INFO] [1694517690.805965141] [io_and_status_controller]: Configure UR gpio controller with tf_prefix:
Sep 12 13:21:35  [ur_ros2_control_node-1] [INFO] [1694517695.439057342] [UR_Client_Library:]: Robot requested program
Sep 12 13:21:35  [ur_ros2_control_node-1] [INFO] [1694517695.439086989] [UR_Client_Library:]: Sent program to robot
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2023-09-15:
Hi there, thanks for reporting this. The output regarding failed tolerances indicates that there is velocity scaling active. Could you provide a complete sample trajectory with which this is reproducible?

Since you said, you confirmed this is working with the simulator: this does not happen when using ursim?

urmahp, 2023-09-15:
For know I haven't been able to reproduce this issue with a ursim. As @fmauch suggested is it possible that you could provide a complete sample trajectory where this issue occurs? Or even better can you provide the code you use to generate and publish the trajectory?

iydv, 2023-09-15:
> this does not happen when using ursim?

I have tested with ursim today extensively and the problem did not happen. I even simulated various network failures via tc-netem tool and everything behaved as expected (robot stopped when the connection to the interface was lost)
But the behavior of the simulator is ideal. With real hardware the behavior was different. 

Here is an example of short trajectory cycle we are sending to the robot. 
The problem occurred when we sent `MOVE_LINEAR` trajectory.

Unfortunately, I cannot provide an example which can be reproduced on ursim, since the issue happens only on real robot hardware. 

```shell
[test_node-1] [INFO] [1694786693.356491537] [test_node]: received 'MOVE' request
[test_node-1] Trajectory point N 0
[test_node-1] time from start: 0
[test_node-1] joint positions: 0.043128, -1.28824, 1.37179, -1.82208, -1.63632, -2.13071, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] Trajectory point N 1
[test_node-1] time from start: 4
[test_node-1] joint positions: -0.195016, -1.70093, 0.902027, -0.944217, -1.52982, -0.195171, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] [INFO] [1694786697.370322140] [test_node]: Trajectory was executed in 4.00366 seconds.

[test_node-1] [INFO] [1694786699.063698768] [test_node]: received 'MOVE' request
[test_node-1] Trajectory point N 0
[test_node-1] time from start: 0
[test_node-1] joint positions: -0.195016, -1.70094, 0.902027, -0.944217, -1.52982, -0.195171, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] Trajectory point N 1
[test_node-1] time from start: 8
[test_node-1] joint positions: 0.30493, -0.982258, 0.955637, -1.48215, -1.72737, 0.204445, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] [INFO] [1694786707.180767196] [test_node]: Trajectory was executed in 8.10695 seconds.

[test_node-1] [INFO] [1694786707.273103585] [test_node]: received 'MOVE' request
[test_node-1] Trajectory point N 0
[test_node-1] time from start: 0
[test_node-1] joint positions: 0.30493, -0.982258, 0.955637, -1.48215, -1.72737, 0.204445, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] Trajectory point N 1
[test_node-1] time from start: 4
[test_node-1] joint positions: 0.298479, -0.988688, 1.04161, -1.56271, -1.72777, 0.197927, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] [INFO] [1694786711.286847325] [test_node]: Trajectory was executed in 4.00365 seconds.

[test_node-1] [INFO] [1694786711.576626438] [test_node]: received 'MOVE' request
[test_node-1] Trajectory point N 0
[test_node-1] time from start: 0
[test_node-1] joint positions: 0.298479, -0.988688, 1.04161, -1.56271, -1.72777, 0.197927, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] Trajectory point N 1
[test_node-1] time from start: 2
[test_node-1] joint positions: 0.30493, -0.982258, 0.955637, -1.48215, -1.72737, 0.204445, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] [INFO] [1694786713.688662269] [test_node]: Trajectory was executed in 2.10191 seconds.

[test_node-1] [INFO] [1694786713.778641016] [test_node]: received 'MOVE_LINEAR' request
[test_node-1] Trajectory point N 0
[test_node-1] time from start: 0
[test_node-1] joint positions: 0.30493, -0.982258, 0.955637, -1.48215, -1.72737, 0.204445, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] Trajectory point N 1
[test_node-1] time from start: 0.257005
[test_node-1] joint positions: 0.304943, -0.978438, 0.891114, -1.42144, -1.72737, 0.204458, 
[test_node-1] joint velocities: -5.57244e-05, 0.0461017, -0.498813, 0.452714, 1.13123e-05, -5.70261e-05, 
[test_node-1] Trajectory point N 2
[test_node-1] time from start: 0.382078
[test_node-1] joint positions: 0.30493, -0.969934, 0.818666, -1.3575, -1.72737, 0.204445, 
[test_node-1] joint velocities: -6.99323e-05, 0.0885967, -0.621143, 0.53255, 1.41965e-05, -7.15658e-05, 
[test_node-1] Trajectory point N 3
[test_node-1] time from start: 0.514344
[... 46 lines cut ...]
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] [INFO] [1694786726.700448778] [test_node]: Trajectory was executed in 4.10383 seconds.

[test_node-1] [INFO] [1694786726.790036854] [test_node]: received 'MOVE' request
[test_node-1] Trajectory point N 0
[test_node-1] time from start: 0
[test_node-1] joint positions: -1.73736, -2.26675, 1.99658, -1.296, -1.57509, -2.32499, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] Trajectory point N 1
[test_node-1] time from start: 6
[test_node-1] joint positions: 1.16829, -1.86254, 1.60861, -1.31265, -1.57557, -0.440643, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] [INFO] [1694786732.905519651] [test_node]: Trajectory was executed in 6.10539 seconds.

[test_node-1] [INFO] [1694786733.195629718] [test_node]: received 'MOVE' request
[test_node-1] Trajectory point N 0
[test_node-1] time from start: 0
[test_node-1] joint positions: 1.16829, -1.86254, 1.60861, -1.31265, -1.57557, -0.440643, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] Trajectory point N 1
[test_node-1] time from start: 4
[test_node-1] joint positions: -0.195016, -1.70093, 0.902023, -0.944215, -1.52982, -0.19517, 
[test_node-1] joint velocities: 0, 0, 0, 0, 0, 0, 
[test_node-1] [INFO] [1694786737.309067384] [test_node]: Trajectory was executed in 4.10333 seconds.
```

fmauch, 2023-09-18:
I'll try to put together a script testing this on one of our robots.

> Here is an example of short trajectory cycle we are sending to the robot.
> The problem occurred when we sent MOVE_LINEAR trajectory.

What's the difference between your MOVE and MOVE_LINEAR? Are both simply goals sent to the scaled JTC?

iydv, 2023-09-18:
> What's the difference between your MOVE and MOVE_LINEAR? Are both simply goals sent to the scaled JTC?

Yes, there is no difference. Both are joint trajectories sent to the JTC. The naming is only for internal usage

urmahp, 2023-09-19:
I have investigated the issue further and managed to reproduce an error similar to what you are experiencing. Is the first point in each trajectory defined with a time of zero? If yes then this could be causing your issue.

If the first point is defined with a time of zero and differs from the current robot position, it would result in the robot moving very quickly towards the target. This is because the commands sent to the robot won't be achievable within a single control cycle, which is expected. I can see that the first point in each trajectory is equal to the last point in the previous trajectory, ensuring that the robot should always start in the correct position. This could explain why you are only experiencing it intermittently, as the robot might have been moved to a different position than expected before you started the driver, or for some reason the target from the previous trajectory hasn't been reached. 

This however does not answer why the robot is moving towards joint positions [0,0,0,0,0,0]. Where do you see that the robot is moving towards joint position [0,0,0,0,0,0]? I haven't been able to find it in the logs you have provided. 

When you have a chance to test it again you can either try to remove the first point with time zero or set the time to be larger than zero to give the robot time to reach the point, to see if that fixes the issue. 

When the current robot position deviates from the path and prints the error you see in the log the robot should definitely stop moving, which is currently not the case, so I will have a look at implementing a fix for that.

fmauch, 2023-09-19:
@urmahp note that the upstream JTC recently got updates regarding trajectory_start time. They are also planning on integrating the scaling functionality upstream soon.

iydv, 2023-09-19:
> Is the first point in each trajectory defined with a time of zero?

Yes, The first point in each trajectory is current joint states, from which robot should start the motion. And time from the start is also zero according to [this comment in joint trajectory controller](https://github.com/ros-controls/ros2_controllers/blob/255b2f9290b7c3c83984ac285fe3572de275c21d/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L1400). Unfortunately, I do not know if scaled joint trajectory controller behaves same way. Is there any example trajectory for scaled JTC ?

> Where do you see that the robot is moving towards joint position [0,0,0,0,0,0]? I haven't been able to find it in the logs you have provided.

We saw it visually, when the real robot moved, this is not indicated in the driver logs and I cannot repeat the experiment since I don't have access to the robot anymore. We will get another UR10 in October and I will check if the issue still persists.

urrsk, 2023-09-20:
@iydv we have not succeed to reproduce and understand where the [0,0,0,0,0,0] are coming from as @fmauch and @urmahp mentioned.
But we have identified that scale JTC does not cancel the motion when the tolerance is too high, which we will fix asap.
In addition, we are also planning to put some checks in the Client library to avid this in the future.

fmauch, 2023-09-20:
I've created a possible fix in #810. This is targeting Rolling, but I'll prepare something similar for Humble, as well.

fmauch, 2023-09-21:
I'll go ahead and close this, since I think this should be fixed by the changes in #810 / #811. @iydv feel free to reopen if you still experience the problem, though.

closed 2023-09-21 by fmauch