UniversalRobots/Universal_Robots_ROS2_Driver #801: Missing command interface when launching the ROS2 driver

urmahp, 2023-09-15:
### Affected ROS2 Driver version(s)

Humble master

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

5.14.1

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
When building the ROS2 Driver from the humble branch and launching the ROS2 driver, the ur_ros2_control_node dies as the command interface force_mode is missing. Se log output below. 

## Issue details
The force_mode command interface is part of the Universal_Robots_ROS2_Description on the ros2 branch. The ros2 branch is checked out when cloning the Universal_Robots_ROS2_Description using VCS import on the required [repos](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/Universal_Robots_ROS2_Driver.humble.repos#L9) or the not released [repos](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/Universal_Robots_ROS2_Driver-not-released.humble.repos#L10). However, the force_mode command interface is not yet supported in the driver and therefore the error occurs. 

## Steps to Reproduce
Build the ROS2 driver from the humble branch and try to launch the driver. 

# Workaround Suggestion
I will add a pull request with a fix later today, where the humble branch is checked out instead of the ros2 branch. 

### Relevant log output

```shell
root@ur-li-95qgtn2:/workspace/ros_ws_ur_driver# ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=127.0.0.1
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-09-15-11-12-28-492646-ur-li-95qgtn2.unirobotts.local-193
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [198]
[INFO] [controller_stopper_node-3]: process started with pid [200]
[INFO] [ur_ros2_control_node-1]: process started with pid [196]
[INFO] [urscript_interface-4]: process started with pid [202]
[INFO] [robot_state_publisher-5]: process started with pid [204]
[INFO] [rviz2-6]: process started with pid [206]
[INFO] [spawner-7]: process started with pid [208]
[INFO] [spawner-8]: process started with pid [210]
[INFO] [spawner-9]: process started with pid [221]
[INFO] [spawner-10]: process started with pid [253]
[INFO] [spawner-11]: process started with pid [255]
[INFO] [spawner-12]: process started with pid [262]
[rviz2-6] qt.qpa.xcb: could not connect to display 
[rviz2-6] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-6] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-6] 
[rviz2-6] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-6] 
[dashboard_client-2] [INFO] [1694776349.855177240] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1694776349.935935560] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1694776349.942745800] [resource_manager]: Loading hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1694776349.946935475] [resource_manager]: Initialize hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1694776349.947084685] [resource_manager]: Successful initialization of hardware 'ur5e'
[ur_ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ur_ros2_control_node-1]   what():  Wrong state or command interface configuration.
[ur_ros2_control_node-1] missing state interfaces:
[ur_ros2_control_node-1] 
[ur_ros2_control_node-1] missing command interfaces:
[ur_ros2_control_node-1] ' force_mode/task_frame_x '	' force_mode/task_frame_y '	' force_mode/task_frame_z '	' force_mode/task_frame_rx '	' force_mode/task_frame_ry '	' force_mode/task_frame_rz '	' force_mode/selection_vector_x '	' force_mode/selection_vector_y '	' force_mode/selection_vector_z '	' force_mode/selection_vector_rx '	' force_mode/selection_vector_ry '	' force_mode/selection_vector_rz '	' force_mode/wrench_x '	' force_mode/wrench_y '	' force_mode/wrench_z '	' force_mode/wrench_rx '	' force_mode/wrench_ry '	' force_mode/wrench_rz '	' force_mode/limits_x '	' force_mode/limits_y '	' force_mode/limits_z '	' force_mode/limits_rx '	' force_mode/limits_ry '	' force_mode/limits_rz '	' force_mode/type '	' force_mode/disable_cmd '	' force_mode/force_mode_async_success '	' force_mode/damping '	' force_mode/gain_scaling '	
[robot_state_publisher-5] [INFO] [1694776349.969780941] [robot_state_publisher]: got segment base
[robot_state_publisher-5] [INFO] [1694776349.972263852] [robot_state_publisher]: got segment base_link
[robot_state_publisher-5] [INFO] [1694776349.973768817] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-5] [INFO] [1694776349.975891055] [robot_state_publisher]: got segment flange
[robot_state_publisher-5] [INFO] [1694776349.975927767] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-5] [INFO] [1694776349.975951133] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-5] [INFO] [1694776349.975968136] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-5] [INFO] [1694776349.975980562] [robot_state_publisher]: got segment tool0
[robot_state_publisher-5] [INFO] [1694776349.975991146] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-5] [INFO] [1694776349.976002979] [robot_state_publisher]: got segment world
[robot_state_publisher-5] [INFO] [1694776349.976014429] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-5] [INFO] [1694776349.976025884] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-5] [INFO] [1694776349.976044511] [robot_state_publisher]: got segment wrist_3_link
[ERROR] [rviz2-6]: process has died [pid 206, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /workspace/ros_ws_ur_driver/install/ur_description/share/ur_description/rviz/view_robot.rviz --ros-args -r __node:=rviz2'].
[ERROR] [ur_ros2_control_node-1]: process has died [pid 196, exit code -6, cmd '/workspace/ros_ws_ur_driver/install/ur_robot_driver/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_mjn77s9q --params-file /workspace/ros_ws_ur_driver/install/ur_robot_driver/share/ur_robot_driver/config/ur5e_update_rate.yaml --params-file /tmp/launch_params_1hzy1xfp'].
[spawner-8] [INFO] [1694776352.580198848] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-9] [INFO] [1694776352.702013811] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1694776352.804029969] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1694776352.832444689] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-12] [INFO] [1694776352.839343608] [spawner_forward_position_controller]: Waiting for '/controller_manager' node to exist
[spawner-11] [INFO] [1694776352.854625189] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2023-09-18:
yes, thanks for notifying!

Apart from this being a problem, obviously, the repos files need updates, since we use the `humble` branch of the description for humble.

fmauch, 2023-09-21:
This has been fixed in #802 -> closing.

PR #802 (2023-09-18): Use humble branch from Universal_Robots_ROS2_Description
This will fix #801
closed 2023-09-21 by fmauch