UniversalRobots/Universal_Robots_ROS2_Driver #814: Request to release new binaries for Iron

Yadunund, 2023-09-21:
### Feature summary

I was planning to do an Iron sync by the end of this week and the process is blocked by regressions in UR packages due to API changes in ros2_controls again. See [sync announcement](https://discourse.ros.org/t/preparing-for-iron-sync-2023-09-22/33660)

I see that the relevant [fixes](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/commit/e5e49ed4bb8cd457ad88dfb815cfdad7104685f7) have already been merged into `iron` so I'd like to request for a new release to be bloomed to help unblock the sync!

Thanks in advance!

fmauch, 2023-09-21:
On my way.

fmauch, 2023-09-21:
done https://github.com/ros/rosdistro/pull/38610

closed 2023-09-21 by fmauch