UniversalRobots/Universal_Robots_ROS2_Driver #817: Moveit error kinematics parameter double humble ursim

gautz, 2023-09-21:
seems related to #265 and #278 but it is still not solved for me too

### Discussed in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/discussions/592

<div type='discussions-op-text'>

<sup>Originally posted by **NickBak08** February  7, 2023</sup>
Hello,

ROS2 version: humble

First I run this line of code: `ros2 run ur_robot_driver start_ursim.sh ur_type:=ur5e`
Then I start the drivers: `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.56.101 launch_rviz:=true`

The next step is too start the moveit example with the ursim but then I get this error. I says that I didn't define the kinematics.yaml. But I didn't change anything to the moveit launch file. What is going wrong?

```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true description_pacakge:=ur_description moveit_config_package:=ur_moveit_config
[INFO] [launch]: All log files can be found below /home/nick/.ros/log/2023-02-07-16-01-45-674683-nick-OMEN-by-HP-Laptop-15-dc1xxx-15394
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [15402]
[INFO] [rviz2-2]: process started with pid [15404]
[INFO] [servo_node_main-3]: process started with pid [15406]
[servo_node_main-3] [WARN] [1675782106.823961856] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node_main-3] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node_main-3] to the Servo composable node in the launch file
[move_group-1] [WARN] [1675782106.833800588] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1675782106.836516169] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00251814 seconds
[move_group-1] [INFO] [1675782106.836550493] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[move_group-1] [INFO] [1675782106.836562431] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[servo_node_main-3] [INFO] [1675782106.836756183] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00496205 seconds
[servo_node_main-3] [INFO] [1675782106.836784917] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[servo_node_main-3] [INFO] [1675782106.836796940] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[servo_node_main-3] [WARN] [1675782106.850179389] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[servo_node_main-3] [INFO] [1675782106.913356333] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/joint_states'
[servo_node_main-3] [INFO] [1675782106.930023590] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[servo_node_main-3] [INFO] [1675782106.930054889] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1675782106.930262992] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1675782106.930444260] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-1] [INFO] [1675782106.932948826] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[servo_node_main-3] [INFO] [1675782106.933465861] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1675782106.933647565] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1675782106.933668302] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1675782106.934278935] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1675782106.934298780] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[servo_node_main-3] [INFO] [1675782106.934373708] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on '/servo_node/publish_planning_scene'
[move_group-1] [INFO] [1675782106.934943376] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1675782106.935472007] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1675782106.935696615] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1675782106.935714430] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[move_group-1] [INFO] [1675782106.942403682] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[servo_node_main-3] [WARN] [1675782106.948057681] [moveit_servo.servo_calcs]: No kinematics solver instantiated for group 'ur_manipulator'. Will use inverse Jacobian for servo calculations instead.
[servo_node_main-3] [WARN] [1675782106.948092604] [moveit_servo.collision_check]: Collision check rate is low, increase it in yaml file if CPU allows
[move_group-1] [INFO] [1675782106.959439692] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1675782106.962029381] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1675782106.962043832] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1675782106.962048986] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1675782106.962061919] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1675782106.962075849] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[move_group-1] [INFO] [1675782106.962081211] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1675782106.962092104] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1675782106.962097350] [moveit_ros.fix_start_state_collision]: Param 'move_group.jiggle_fraction' was not set. Using default value: 0.020000
[move_group-1] [INFO] [1675782106.962102523] [moveit_ros.fix_start_state_collision]: Param 'move_group.max_sampling_attempts' was not set. Using default value: 100
[move_group-1] [INFO] [1675782106.962113469] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-1] [INFO] [1675782106.962118669] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[move_group-1] [INFO] [1675782106.962122708] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[move_group-1] [INFO] [1675782106.962126401] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[move_group-1] [INFO] [1675782106.962150331] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[move_group-1] [INFO] [1675782107.002051818] [moveit.plugins.moveit_simple_controller_manager]: Added FollowJointTrajectory controller for scaled_joint_trajectory_controller
[move_group-1] [INFO] [1675782107.006957148] [moveit.plugins.moveit_simple_controller_manager]: Added FollowJointTrajectory controller for joint_trajectory_controller
[... 24 lines cut ...]
[move_group-1] Loading 'move_group/MoveGroupCartesianPathService'...
[move_group-1] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-1] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[... cut ...]
[rviz2-2] [INFO] [1675782110.999910081] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-2] [ERROR] [1675782111.014539914] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_kinematics.ur_manipulator.kinematics_solver_timeout' has invalid type: Wrong parameter type, parameter {robot_description_kinematics.ur_manipulator.kinematics_solver_timeout} is of type {double}, setting it to {string} is not allowed.
```

</div>

fmauch, 2023-09-22:
Thanks for bringing this up again, I'll look into this ...

fmauch, 2023-09-25:
I just checked again and this works fine for me. What version are you using? 

* Built from source? If yes, what are the commit IDs of the description and the driver repo?
* From apt package? If yes, which versions of the description and driver are you running?

macmacal, 2024-02-02:
I had the same error with problem of the parameter type from `kinematics.yaml` with the binary version of `ur_moveit_config` package (2.2.10-1jammy.20240126.080424). However, I was trying to run it with `use_mock_hardware` parameter.

The logs I got were something like this:
```bash
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true use_mock_hardware:=true
[...]
[rviz2-2] [ERROR] [1706882016.433182311] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_kinematics.ur_manipulator.kinematics_solver_timeout' has invalid type: Wrong parameter type, parameter {robot_description_kinematics.ur_manipulator.kinematics_solver_timeout} is of type {double}, setting it to {string} is not allowed.
```

I find out a simple solution mentioned in the issue #723 (which comes with references from [here2022](https://github.com/ros-planning/moveit2/issues/1049#issuecomment-1047029751) and [here2017](https://github.com/ros-visualization/rviz/issues/1151#issuecomment-345687074) ): the problem (feature) lies in `std:stod` which parses input according to `LC_NUMERIC` locale parameter. Try to add `LC_NUMERIC=en_US.UTF-8` before you command:

```bash
LC_NUMERIC=en_US.UTF-8 ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true description_pacakge:=ur_description moveit_config_package:=ur_moveit_config
```

closed 2024-04-10 by fmauch