UniversalRobots/Universal_Robots_ROS2_Driver #836: ros2_control and ft_torque sensor doesn't work simultaneously

Nemantor, 2023-10-10:
### Affected ROS2 Driver version(s)

release

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in linux

### Robot SW / URSim version(s)

ign gazebo 6.15

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
ros2_control and ft_torque sensor doesn't work simultaneously
1.simulation launched with ft_torque sensor and ros2_control enabled sdf file -> no torque_force is published to a topic
2.simulation launched with ft_torque sensor enabled sdf file -> torque_force is published to a topic

## Expected Behavior
How can I get torque measurements from my gazebo simulation while having ros2_control enabled.

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2023-10-10:
I am confused. Are you using URSim or gazebo? If you are using gazebo then there is no need to actually use this driver.

urfeex, 2025-01-15:
Closing this due to missing information

closed 2025-01-15 by urfeex