UniversalRobots/Universal_Robots_ROS2_Driver #838: urscruipt_interface started with wrong IP cannot be stopped normally

fmauch, 2023-10-10:
While testing around I noticed that when I start the urscript_interface without a connected robot it cannot be stopped easily:

## Steps to reproduce
```
ros2 run ur_robot_driver urscript_interface --ros-args -p robot_ip:=1.2.3.4
```
The output will be
```
ERROR /home/ubuntu/checkout/ur_ros2_rolling/colcon_ws/src/ur_client_library/src/comm/tcp_socket.cpp 129: Failed to
connect to robot on IP 1.2.3.4. Please check that the robot is booted and reachable on 1.2.3.4. Retrying in 10 seconds
```

This will  be repeated every 10 seconds.

I could only stop the process by getting its PID and manually calling `kill -9 $PID`

## Expected behavior
Pressing CTRL-C will terminate the urscript_interface

## Actual behavior
* Pressing CTRL-C doesn't do anything
* Killing the process using SIGTERM
* Killing the process using SIGKILL works

RobertWilbrandt, 2023-10-11:
Adding a simple handler like
```cpp
rclcpp::on_shutdown([]() { rclcpp::shutdown(); });
```
seems to fix this, but results in
```
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
```

fmauch, 2023-10-12:
> `rclcpp::on_shutdown([]() { rclcpp::shutdown(); });`

This seems odd... Honestly, I guess there probably is something to be fixed in the client library. I suspect that our reconnecting mechanism isn't properly interruptible at the moment.

eholum, 2024-08-23:
Ran into this as well, and FWIW I think [this socket in the constructor](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/src/urscript_interface.cpp#L72) just needs some reasonable values for `max_num_tries ` and a `reconnection_time`, otherwise the constructor sits in [this while loop in the TCP socket](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/comm/tcp_socket.cpp#L91) forever. The constructor never completes, so the node never starts spinning.

shin4141, 2026-08-20:
Hi, I reproduced this on current `main`. The behavior has changed since the original report: startup is now bounded to 10 retries, but SIGINT is still received while the node is blocked in `PrimaryClient::start()` before the executor starts, so shutdown waits for the remaining retries and eventually exits via an uncaught `UrException`.

I tested the current UR Client Library separately and confirmed that `PrimaryClient::stop()` can safely interrupt an in-progress `start()` without changing the retry policy or URCL.

My proposed driver-local scope is:

- move the blocking `PrimaryClient::start(10, 10s)` into a joinable startup worker,
- keep the existing topic/action endpoints unavailable until startup succeeds,
- on shutdown during startup, call `stop()` and join the worker,
- convert genuine startup exhaustion into an orderly nonzero exit instead of SIGABRT,
- add a loopback-only process regression test; no robot, URSim, or hardware required.

This would not change retry timing, script/action behavior, controller behavior, or motion/safety semantics.

Would this scope be appropriate for #838?

urfeex, 2026-09-01:
~~I think, this should be resolved by https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/519 but I haven't checked since merging this.~~ The issue still persists.

@shin4141 your approach seems very reasonable, so if you want to go for it, you're welcome :-)

closed 2026-10-02 by urfeex