UniversalRobots/Universal_Robots_ROS2_Driver #839: Passthrough controllers

fmauch, 2023-10-10:
### Feature summary

We should implement a trajectory forwarding controller  similar to https://github.com/UniversalRobots/Universal_Robots_ROS_passthrough_controllers

We already have multiple possibilities to forward trajectories to the robot controller instead of interpolating it on the ROS machine and sending commands to the robot in each control cycle. [Recently](https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/151), the ur_client_library got added the possibility to do spline interpolation on the robot controller similar to the one done by ros2_control's JTC.

Interpolating on the robot has the advantage of a lower real-time requirement of the driver, since only the trajectory points are sent to the robot in the beginning. For this to be possible, we also [recently](https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/178) added the possibility to make the receive timeout on the robot depending on the actual control mode.

### Related issues
* https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/151
* https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/178
* https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/543

### Tasks
*To complete this issue involves*
- [ ] Implement the controller
- [ ] Make documentation
- [ ] Make Unit test
- [ ] Make example
- [ ] Test on real hardware

LucasLabarussiat, 2024-02-16:
Hi @fmauch,

I see this is marked as "in progress". Is there a fork somewhere to have a look at it ?
If I am able to I will gladly give a hand, as this passthrough controller is a feature that some people in my team are asking for.

fmauch, 2024-09-25:
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/944 should resolve this.

PR #944 (2024-11-11): Forward trajectory controller
The controller should be able to forward a cartesian tracjectory to the robot for interpolation.
Creates errors on compilation due to multiple definition of functions, when inheriting from the joint_trajectory_controller. But looking at the passthrough controllers from ROS1 that may not be the right controller to use anyway?
All inputs are welcome.

closes #839
PR #1183 (2024-11-28): Forward trajectory controller (backport #944)
The controller should be able to forward a cartesian tracjectory to the robot for interpolation.
Creates errors on compilation due to multiple definition of functions, when inheriting from the joint_trajectory_controller. But looking at the passthrough controllers from ROS1 that may not be the right controller to use anyway?
All inputs are welcome.

closes #839<hr>This is an automatic backport of pull request #944 done by [Mergify](https://mergify.com).
closed 2024-11-11 by urfeex