UniversalRobots/Universal_Robots_ROS2_Driver #840: Move Group: Execution of planned path fails using use_fake_hardware:=true

RobinHeitz, 2023-10-11:
### Affected ROS2 Driver version(s)

latest humble binary.

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel, Docker

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

No robot involved so far

### How is the ROS driver used.

Others

### Issue details

# Summary
Execution of a planned path in moveit via MotionPlanning-Plugin fails. I would expect that it works 'out of the box'.

## Issue details
Tested on my machine (Ubuntu 22.04) and in a Docker Dev Container (also Ubuntu 22.04), same behavior:
After installing `sudo apt install ros-humble-ur` I try to launch the move group and do some planning & execution without a real robot. So, after `ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e use_fake_hardware:=true` Rviz2 launches and the terminal tells me that I can start planning now, which indeed works. I see the (looped) animation of the planned path.

After pressing execute, it fails after 1 second. The log is provided below, but I think the critical part it this:

> [move_group-1] [INFO] [1697012885.611099782] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1697012884.611024, but latest received state has time 0.000000.
> [move_group-1] Check clock synchronization if your are running ROS across multiple machines!

I'm not running ROS across multiple machines, but I don't know why the clock is out of sync. 
I don't know whether this is the actual problem or if this is even the expected behavior.

I remember faintly, that once I tried ROS2 + Moveit2 for the first time, It worked 'out of the box'. I could plan & execute in MoveIt and afterwards execute the planned path even on a real robot.

## Steps to Reproduce
Install the binary and call `ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e use_fake_hardware:=true`. Try to execute a planned movement - it fails.

## Expected Behavior
It should not fail but execute the movement (and show the actual robot model moving along the path).

## Actual Behavior
Movement execution fails. 

### Relevant log output

```shell
[INFO] [launch]: All log files can be found below /home/ros/.ros/log/2023-10-11-08-27-54-573027-rosmachine-30392
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [30410]
[INFO] [rviz2-2]: process started with pid [30412]
[INFO] [servo_node_main-3]: process started with pid [30414]
[rviz2-2] QStandardPaths: runtime directory '/tmp' is not owned by UID 1000, but a directory permissions 0777 owned by UID 0 GID 0
[servo_node_main-3] [WARN] [1697012875.503608204] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node_main-3] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node_main-3] to the Servo composable node in the launch file
[move_group-1] [WARN] [1697012875.504649413] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1697012875.506682731] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00191749 seconds
[... 101 lines cut ...]
[move_group-1] [WARN] [1697012883.063314164] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to fetch current robot state.
[move_group-1] [INFO] [1697012883.063365324] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-1] [INFO] [1697012883.063438754] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'move_group'
[move_group-1] [INFO] [1697012883.064678313] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-1] [WARN] [1697012883.158883122] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[move_group-1] [INFO] [1697012883.161094940] [moveit_move_group_default_capabilities.move_action_capability]: Motion plan was computed successfully.
[rviz2-2] [INFO] [1697012883.161603479] [move_group_interface]: Planning request complete!
[rviz2-2] [INFO] [1697012883.161697649] [move_group_interface]: time taken to generate plan: 0.0224252 seconds
[move_group-1] [INFO] [1697012884.610698025] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-1] [INFO] [1697012884.610823214] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-1] [INFO] [1697012884.610858054] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1697012884.610877724] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1697012884.611010124] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[rviz2-2] [INFO] [1697012884.610876924] [move_group_interface]: Execute request accepted
[move_group-1] [INFO] [1697012885.611099782] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1697012884.611024, but latest received state has time 0.000000.
[move_group-1] Check clock synchronization if your are running ROS across multiple machines!
[move_group-1] [WARN] [1697012885.611193352] [moveit_ros.trajectory_execution_manager]: Failed to validate trajectory: couldn't receive full current joint state within 1s
[move_group-1] [INFO] [1697012885.611227791] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[rviz2-2] [INFO] [1697012885.611549111] [move_group_interface]: Execute request aborted
[rviz2-2] [ERROR] [1697012885.611564581] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
```

### Accept Public visibility

- [X] I agree to make this context public

RobertWilbrandt, 2023-10-11:
This is the expected behavior. Please take a look at our [MoveIt Usage documentation](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#mock-hardware): The `use_fake_hardware` flag only informs the move group which controller to use by default, as the default `scaled_joint_trajectory_controller` doesn't work for fake hardware. You still need to start the driver, e.g.:

```console
ros2 launch ur_robot_driver ur3e.launch.py robot_ip:=xxx.xxx.xxx.xxx use_fake_hardware:=true initial_joint_controller:=joint_trajectory_controller
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e use_fake_hardware:=true
```

Can you please test if this works for you?

RobinHeitz, 2023-10-12:
Thanks for your quick reply!
Also it did not worked in the beginning, it does now. Thanks a lot.

I missed to install the controllers themselves, was not easy for me (beginner) to obtain that information from the logs.
So: `sudo apt install ros-humble-ros2-controllers' is necessary!

Thanks again!

RobinHeitz, 2023-10-12:
Well, just found that its still not working on my machine (but still inside my docker container) and I don't get why.
I've updated & upgraded via `sudo apt upgrade && sudo apt update` and installed exactly the same packages on my local machine as defined in my Dockerfile. Should be the same starting point I think.

I launch these 2 commands in and outside the docker container:
```
ros2 launch ur_robot_driver ur3e.launch.py robot_ip:=xxx.xxx.xxx.xxx use_fake_hardware:=true initial_joint_controller:=joint_trajectory_controller launch_rviz:=false 
```
```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e use_fake_hardware:=true
```
After launching both commands it seems to be exactly the same as in the docker version. It states **Configured and activated joint_trajectory_controller** etc. Also, no error/ warning until then.

Here are the 2 outputs:

```
[INFO] [launch]: All log files can be found below /home/robin/.ros/log/2023-10-12-14-45-32-881295-rosmachine-82010
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [82014]
[INFO] [urscript_interface-2]: process started with pid [82016]
[INFO] [robot_state_publisher-3]: process started with pid [82018]
[INFO] [spawner-4]: process started with pid [82020]
[INFO] [spawner-5]: process started with pid [82022]
[INFO] [spawner-6]: process started with pid [82024]
[INFO] [spawner-7]: process started with pid [82026]
[INFO] [spawner-8]: process started with pid [82028]
[INFO] [spawner-9]: process started with pid [82030]
[robot_state_publisher-3] [INFO] [1697114733.571937733] [robot_state_publisher]: got segment base
[robot_state_publisher-3] [INFO] [1697114733.572032212] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1697114733.572042732] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-3] [INFO] [1697114733.572049462] [robot_state_publisher]: got segment flange
[robot_state_publisher-3] [INFO] [1697114733.572055862] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-3] [INFO] [1697114733.572061512] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-3] [INFO] [1697114733.572067152] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-3] [INFO] [1697114733.572072842] [robot_state_publisher]: got segment tool0
[robot_state_publisher-3] [INFO] [1697114733.572078512] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-3] [INFO] [1697114733.572084242] [robot_state_publisher]: got segment world
[robot_state_publisher-3] [INFO] [1697114733.572090062] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-3] [INFO] [1697114733.572095822] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-3] [INFO] [1697114733.572101512] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1697114733.590412281] [resource_manager]: Loading hardware 'ur3e' 
[ros2_control_node-1] [INFO] [1697114733.591027120] [resource_manager]: Initialize hardware 'ur3e' 
[ros2_control_node-1] [WARN] [1697114733.591050100] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-1] [WARN] [1697114733.591062760] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-1] <state_interface name="velocity"> 
[ros2_control_node-1]   <param name="initial_value">0.0</param> 
[ros2_control_node-1] </state_interface>
[ros2_control_node-1] [INFO] [1697114733.591105320] [resource_manager]: Successful initialization of hardware 'ur3e'
[ros2_control_node-1] [INFO] [1697114733.591228560] [resource_manager]: 'configure' hardware 'ur3e' 
[ros2_control_node-1] [INFO] [1697114733.591234290] [resource_manager]: Successful 'configure' of hardware 'ur3e'
[ros2_control_node-1] [INFO] [1697114733.591256190] [resource_manager]: 'activate' hardware 'ur3e' 
[ros2_control_node-1] [INFO] [1697114733.591261110] [resource_manager]: Successful 'activate' of hardware 'ur3e'
[ros2_control_node-1] [INFO] [1697114733.595237315] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1697114733.595462715] [controller_manager]: RT kernel is recommended for better performance
[ros2_control_node-1] [INFO] [1697114733.783484254] [controller_manager]: Loading controller 'forward_position_controller'
[ros2_control_node-1] [INFO] [1697114733.792970052] [controller_manager]: Loading controller 'speed_scaling_state_broadcaster'
[spawner-9] [INFO] [1697114733.807497646] [spawner_forward_position_controller]: Loaded forward_position_controller
[ros2_control_node-1] [INFO] [1697114733.810531012] [speed_scaling_state_broadcaster]: Loading UR SpeedScalingStateBroadcaster with tf_prefix: 
[ros2_control_node-1] [INFO] [1697114733.811748431] [controller_manager]: Configuring controller 'forward_position_controller'
[ros2_control_node-1] [INFO] [1697114733.812854879] [forward_position_controller]: configure successful
[spawner-7] [INFO] [1697114733.826877743] [spawner_speed_scaling_state_broadcaster]: Loaded speed_scaling_state_broadcaster
[ros2_control_node-1] [INFO] [1697114733.827744762] [controller_manager]: Configuring controller 'speed_scaling_state_broadcaster'
[... cut ...]
[move_group-1] [INFO] [1697114887.694903374] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1697114887.694967154] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[rviz2-2] [INFO] [1697114887.695280114] [move_group_interface]: Execute request aborted
[rviz2-2] [ERROR] [1697114887.695482794] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
```

Thanks!

RobinHeitz, 2023-10-23:
Well, thank you a lot for this hint!
Found out there was a sourcing to a different workspace in ~./bashrc which had an older moveit version installed from source. Seems like it op'ted for this instead of the binary install.

closed 2023-10-23 by RobinHeitz