UniversalRobots/Universal_Robots_ROS2_Driver #851: Unable to use moveit::planning_interface::MoveGroupInterface after starting "ur_moveit.launch.py"

GerardHarkemaAvans, 2023-10-19:
### Affected ROS2 Driver version(s)

Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

5.13.1

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

I am trying to create a plan group, see the code below:

    printf("My Demo CPP\n");
 
    rclcpp::init(argc, argv);
    //rclcpp::NodeOptions node_options;
    //node_options.automatically_declare_parameters_from_overrides(true);
    auto move_group_node = rclcpp::Node::make_shared("my_node", rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true));
 

    // We spin up a SingleThreadedExecutor for the current state monitor to get information
    // about the robot's state.
    rclcpp::executors::SingleThreadedExecutor executor;
    executor.add_node(move_group_node);
    std::thread([&executor]() { executor.spin(); }).detach();
 
 
    // MoveIt operates on sets of joints called "planning groups" and stores them in an object called
    // the ``JointModelGroup``. Throughout MoveIt, the terms "planning group" and "joint model group"
    // are used interchangeably.
    static const std::string PLANNING_GROUP = "ur_manipulator";

 
    // The
    // :moveit_codedir:`MoveGroupInterface<moveit_ros/planning_interface/move_group_interface/include/moveit/move_group_interface/move_group_interface.h>`
    // class can be easily set up using just the name of the planning group you would like to control and plan for.
    moveit::planning_interface::MoveGroupInterface move_group(move_group_node, PLANNING_GROUP);

Unfortunately, this code gives error messages(see log), while I see in RVIZ that the planning group exists. Who can help me with this? Thank you in advance for your answer.

### Relevant log output

```shell
[ERROR] [1697696000.025772091] [my_node]: Could not find parameter robot_description and did not receive robot_description via std_msgs::msg::String subscription within 10.000000 seconds.
Error:   Error document empty.
         at line 100 in ./urdf_parser/src/model.cpp
Failed to parse robot description using: urdf_xml_parser/URDFXMLParser
[INFO] [1697696010.038102017] [moveit_rdf_loader.rdf_loader]: Unable to parse URDF
[FATAL] [1697696010.038383347] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.
[ros2run]: Aborted
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2023-10-19:
In which context do you start your node? The usual way is to have a `robot_state_publisher` running that publishes the description on `/robot_description`. This will get started e.g. by running the launchfiles from the `ur_robot_driver`.

I hope that helps.

GerardHarkemaAvans, 2023-10-26:
Thanks!. I started the program with ros2 run. I have now created a launch file in which I load the srdf, urdt files, etc. as parameters and start the program as a node. This seems to be working fine now.

fmauch, 2023-10-27:
That's great to hear. I'll go ahead and close this issue, then.

closed 2023-10-27 by fmauch