UniversalRobots/Universal_Robots_ROS2_Driver #863: Error scaled_joint_trajectory_controller: Holding position due to state tolerance violation

AxelDegrande, 2023-10-27:
### Affected ROS2 Driver version(s)

Latest

### Used ROS distribution.

Rolling

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

UR10

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
When sending joint positions to the UR10 robot does not move. Two weeks ago it did move.

## Issue details

## Steps to Reproduce
1) I start the UR10 robot and initialize it --> add the structure for remote control in the robot program
2) Start the ur_driver
$ ros2 launch ur_robot_driver ur10.launch.py robot_ip:=192.168.1.200 launch_rviz:=true
3) Run python program that sends six joint angles to the robot (py code below)

`import rclpy
from rclpy.node import Node
from trajectory_msgs.msg import JointTrajectory, JointTrajectoryPoint

class UR10Mover(Node):

    def __init__(self):
        super().__init__('ur10_mover')
        self.publisher_ = self.create_publisher(JointTrajectory, '/scaled_joint_trajectory_controller/joint_trajectory', 10)

    def move_ur10(self):
        msg = JointTrajectory()
        msg.joint_names = ["shoulder_pan_joint", "shoulder_lift_joint", "elbow_joint", 
                           "wrist_1_joint", "wrist_2_joint", "wrist_3_joint"]   

        
        point = JointTrajectoryPoint()
        # positions: [theta1, theta2, theta3, theta4, theta5, theta6]
        point.positions = [0, -0.6109, -24435, -1.6581, 1.5708, 0]
        """
        point.positions = [-0.54, -1.4491, -2.5592, -0.6562, -0.6562, -1.6914]
        point.positions = [0.6435, -1.2161, -0.8685, 0.7594, 1.3617, -1.6916]   
        """

        point.time_from_start.sec = 16
        msg.points.append(point)

        
        self.publisher_.publish(msg)
[... 134 lines cut ...]
[ur_ros2_control_node-1] [INFO] [1698399866.220182771] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[ur_ros2_control_node-1] [INFO] [1698399866.228822392] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[spawner-11] [INFO] [1698399866.231129881] [spawner_force_torque_sensor_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[ur_ros2_control_node-1] [INFO] [1698399866.236426351] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[ur_ros2_control_node-1] [INFO] [1698399866.236509167] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[ur_ros2_control_node-1] [INFO] [1698399866.244798670] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[spawner-7] [INFO] [1698399866.246288417] [spawner_scaled_joint_trajectory_controller]: Configured and activated scaled_joint_trajectory_controller
[ur_ros2_control_node-1] [INFO] [1698399866.252206457] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[INFO] [spawner-10]: process has finished cleanly [pid 30417]
[INFO] [spawner-8]: process has finished cleanly [pid 30413]
[INFO] [spawner-12]: process has finished cleanly [pid 30422]
[INFO] [spawner-11]: process has finished cleanly [pid 30419]
[INFO] [spawner-9]: process has finished cleanly [pid 30415]
[INFO] [spawner-7]: process has finished cleanly [pid 30355]
[ur_ros2_control_node-1] [INFO] [1698400108.570537299] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1698400108.570601816] [UR_Client_Library:]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1698400108.958180040] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.
[ur_ros2_control_node-1] [INFO] [1698400108.974025114] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[ur_ros2_control_node-1] [INFO] [1698400108.974112649] [io_and_status_controller]: Configure UR gpio controller with tf_prefix: 
[ur_ros2_control_node-1] [ERROR] [1698400606.938671893] [scaled_joint_trajectory_controller]: Holding position due to state tolerance violation
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2023-10-31:
If I slightly modify your code and change the `-24435` to `-2.4435`, then your code runs fine for me. The output could probably be improved there, though.

urrsk, 2023-11-23:
Agree, we should improve the error feedback to make it easier to spot problem.
Though, nice to hear that the tolerance check worked and prevented the robot to do an unexpected fast motion.

urrsk, 2023-11-23:
@AxelDegrande did you succeed making it work and can we close this issue?

fmauch, 2024-07-15:
Going ahead and closing this due to inactivity. Feel free to comment / reopen if the issue still persists.

closed 2024-07-15 by fmauch

alin742, 2024-12-15:
Hello, I am also getting the same error with the OPs code (I fixed the wrong joint angle -24435 -> -2.4435) as well as the test_scaled_joint_controller launch file. 

for context, I am running this from a docker container with the actual robot. I am getting the warning regarding the real-time kernel although I just compiled the 6.12 kernel with the PREEMPT_RT config enabled. The ports published in the docker container are 29999-30010 and 50000-50012 as well as 9000.

### Relevant Logs
`[INFO] [launch]: All log files can be found below /root/.ros/log/2024-12-15-16-40-27-780804-eb9ebb18e450-464
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [calibration_correction-1]: process started with pid [465]
[calibration_correction-1] [WARN] [1734280827.881114559] [UR_Client_Library:]: Failed to read from stream, reconnecting in 1 seconds...
[calibration_correction-1] [INFO] [1734280829.080919440] [ur_calibration]: checksum: [2905197280 2905497409 759806402 2853233814 2852910048 2852651659 ]
[calibration_correction-1] dh_theta: [-5.7617997137327e-08 1.36820651398328 4.24324506496508 0.671742550068674 -1.23032896499854e-07 -9.43368128002685e-08 ]
[calibration_correction-1] dh_a: [-0.000124567734565739 -0.123182849936155 -0.447266677871086 4.43889902256801e-05 -7.44562913741664e-06 0 ]
[calibration_correction-1] dh_d: [0.180554786329347 -1948.05426491946 2381.78167415029 -433.553219433789 0.119884644433412 0.115597112697294 ]
[calibration_correction-1] dh_alpha: [1.57156953964481 -0.000307837167018039 -0.000819908037964153 1.57012711837425 -1.57078338179139 0 ]
[calibration_correction-1] calibration_status: 2
[calibration_correction-1] 
[calibration_correction-1] [INFO] [1734280829.181287094] [ur_calibration]: Writing calibration data to "/root/my_robot_calibration.yaml"
[calibration_correction-1] [INFO] [1734280829.181802788] [ur_calibration]: Wrote output.
[calibration_correction-1] [INFO] [1734280829.181891968] [ur_calibration]: Calibration correction done
[INFO] [calibration_correction-1]: process has finished cleanly [pid 465]
root@eb9ebb18e450:/# ros2 launch ur_robot_driver ur_control.launch.py \
  ur_type:=ur10e \
  robot_ip:=192.168.1.16 \
  kinematics_params_file:="${HOME}/my_robot_calibration.yaml"
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-12-15-16-40-51-844589-eb9ebb18e450-478
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [482]
[INFO] [controller_stopper_node-3]: process started with pid [484]
[INFO] [ur_ros2_control_node-1]: process started with pid [480]
[INFO] [urscript_interface-4]: process started with pid [486]
[INFO] [robot_state_publisher-5]: process started with pid [488]
[INFO] [rviz2-6]: process started with pid [490]
[INFO] [spawner-7]: process started with pid [492]
[INFO] [spawner-8]: process started with pid [494]
[INFO] [spawner-9]: process started with pid [496]
[dashboard_client-2] [INFO] [1734280852.066074864] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1734280852.068503247] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] [WARN] [1734280852.070475149] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1734280852.070750314] [resource_manager]: Loading hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1734280852.071782282] [resource_manager]: Initialize hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1734280852.071834131] [resource_manager]: Successful initialization of hardware 'ur10e'
[ur_ros2_control_node-1] [INFO] [1734280852.071960122] [resource_manager]: 'configure' hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1734280852.071967385] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1734280852.071975591] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1734280852.072791818] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ur_ros2_control_node-1] [INFO] [1734280852.074146102] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1734280852.074370961] [UR_Client_Library:]: Setting up RTDE communication with frequency 500.000000
[rviz2-6] qt.qpa.xcb: could not connect to display 
[rviz2-6] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-6] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-6] 
[rviz2-6] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-6] 
[robot_state_publisher-5] [INFO] [1734280852.094065703] [robot_state_publisher]: got segment base
[robot_state_publisher-5] [INFO] [1734280852.094162588] [robot_state_publisher]: got segment base_link
[... cut ...]
[ur_ros2_control_node-1] [ERROR] [1734280944.219231260] [scaled_joint_trajectory_controller]: Holding position due to state tolerance violation
[ur_ros2_control_node-1] [ERROR] [1734281315.193204023] [tolerances]: State tolerances failed for joint 2:
[ur_ros2_control_node-1] [ERROR] [1734281315.193217318] [tolerances]: Position Error: -0.200272, Position Tolerance: 0.200000
[ur_ros2_control_node-1] [ERROR] [1734281315.193223490] [scaled_joint_trajectory_controller]: Holding position due to state tolerance violation
`