UniversalRobots/Universal_Robots_ROS2_Driver #866: Multiple Arm bring up issues. Problem occurs with loading controllers

cucudasluv, 2023-10-29:
### Affected ROS2 Driver version(s)

2.4.1 maybe. 

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot, Real robot, URSim in linux

### Robot SW / URSim version(s)

1.0.5 URSIM

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

## Issue details
 I just followed the dual arm launch demo in #825. In my case I connect ur5e, ur10e to my remote PC. But when I execute the launch file below, It cannot load all of controllers. I have a experience using two arms in ROS1 and I think this launch file makes sense. 
I want to ask the way to load  controllers of dual arms in once. I saw that #825 contributor achieve to load  controllers of dual arms in once with same way. Here Is my terminal Log.

    
##launch file
`import os
 from ament_index_python.packages import get_package_share_directory
 from launch.actions import IncludeLaunchDescription
 from launch import LaunchDescription
 from launch.launch_description_sources import PythonLaunchDescriptionSource
 from launch.actions import DeclareLaunchArgument, LogInfo
 from launch.substitutions import LaunchConfiguration

 from launch.actions import GroupAction
 from launch_ros.actions import PushRosNamespace
 from launch_ros.descriptions import ParameterValue

 def generate_launch_description():

    ur_type = LaunchConfiguration('ur_type') 
    arm_0_ur_type = LaunchConfiguration('arm_0_ur_type')     
    arm_0_robot_ip = LaunchConfiguration('arm_0_robot_ip') 
    arm_0_controller_file = LaunchConfiguration('arm_0_controller_file') 
    arm_0_tf_prefix = LaunchConfiguration('arm_0_tf_prefix') 
    arm_0_script_command_port = LaunchConfiguration('arm_0_script_command_port')
    arm_0_trajectory_port = LaunchConfiguration('arm_0_trajectory_port')
    arm_0_reverse_port = LaunchConfiguration('arm_0_reverse_port')
    arm_0_script_sender_port = LaunchConfiguration('arm_0_script_sender_port')
    
    arm_1_ur_type = LaunchConfiguration('arm_1_ur_type')    
    arm_1_robot_ip = LaunchConfiguration('arm_1_robot_ip') 
    arm_1_controller_file = LaunchConfiguration('arm_1_controller_file') 
    arm_1_tf_prefix = LaunchConfiguration('arm_1_tf_prefix') 
    arm_1_script_command_port = LaunchConfiguration('arm_1_script_command_port')
    arm_1_trajectory_port = LaunchConfiguration('arm_1_trajectory_port')
    arm_1_reverse_port = LaunchConfiguration('arm_1_reverse_port')
    arm_1_script_sender_port = LaunchConfiguration('arm_1_script_sender_port')

    # # UR specific arguments
[... 329 lines cut ...]
[ERROR] [spawner-18]: process has died [pid 145755, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur5e'].
[ERROR] [spawner-8]: process has died [pid 145663, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner io_and_status_controller --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur10e'].
[spawner-21] [ERROR] [1698617754.201317504] [ur5e.spawner_force_torque_sensor_broadcaster]: Controller manager not available
[spawner-6] [ERROR] [1698617754.201833363] [ur10e.spawner_scaled_joint_trajectory_controller]: Controller manager not available
[spawner-9] [ERROR] [1698617754.216168074] [ur10e.spawner_speed_scaling_state_broadcaster]: Controller manager not available
[ERROR] [spawner-6]: process has died [pid 145659, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner scaled_joint_trajectory_controller -c /controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur10e'].
[ERROR] [spawner-21]: process has died [pid 145781, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner force_torque_sensor_broadcaster --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur5e'].
[ERROR] [spawner-9]: process has died [pid 145687, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner speed_scaling_state_broadcaster --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur10e'].
[spawner-17] [ERROR] [1698617754.519184467] [ur5e.spawner_scaled_joint_trajectory_controller]: Controller manager not available
[ERROR] [spawner-17]: process has died [pid 145738, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner scaled_joint_trajectory_controller -c /controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur5e'].
[spawner-7] [ERROR] [1698617754.815164896] [ur10e.spawner_joint_state_broadcaster]: Controller manager not available
[spawner-11] [ERROR] [1698617754.842721606] [ur10e.spawner_forward_position_controller]: Controller manager not available
[spawner-22] [ERROR] [1698617754.852284880] [ur5e.spawner_forward_position_controller]: Controller manager not available
[spawner-20] [ERROR] [1698617754.874648346] [ur5e.spawner_speed_scaling_state_broadcaster]: Controller manager not available
[spawner-19] [ERROR] [1698617754.874811932] [ur5e.spawner_io_and_status_controller]: Controller manager not available
[ERROR] [spawner-7]: process has died [pid 145661, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur10e'].
[ERROR] [spawner-11]: process has died [pid 145712, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner forward_position_controller --controller-manager /controller_manager --controller-manager-timeout 10 --inactive --ros-args -r __ns:=/ur10e'].
[ERROR] [spawner-22]: process has died [pid 145785, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner forward_position_controller --controller-manager /controller_manager --controller-manager-timeout 10 --inactive --ros-args -r __ns:=/ur5e'].
[ERROR] [spawner-19]: process has died [pid 145768, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner io_and_status_controller --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur5e'].
[ERROR] [spawner-20]: process has died [pid 145772, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner speed_scaling_state_broadcaster --controller-manager /controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur5e'].
```

### Accept Public visibility

- [X] I agree to make this context public

medvedevigorek, 2023-11-05:
I had a similar issue with my dual arm setup (both UR5e) and I narrowed it down to [this line](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/2.4.1/ur_robot_driver/launch/ur_control.launch.py#L307) - when spawning a controller the script uses `/controller_manager` but the controller manager would be running in a namespace. 

A simple fix was to use `controller_manager` without `/` at the begging as [spawner.py](https://github.com/ros-controls/ros2_control/blob/3.19.1/controller_manager/controller_manager/spawner.py#L206) will be looking for the controller manager in its namespace in this case. That required changes to `ur_control.launch.py` script though. I also tried to use `SetRemap` to avoid changing the code but with no luck.

cucudasluv, 2023-11-05:
@medvedevigorek Thank you for your comment. and I change all of **/controller_manager** into **controller_manager**. and I got the result of this, when I launch the above file (I provided). Should I modify the ur5e_controller.yaml and ur10e_controller.yaml? I just copy exactly by ur_controller.yaml for each. 
`adip@adip:~/colcon_ws/ros_ur_driver$ ros2 launch ur_robot_driver dual_bringup.launch.py 
[INFO] [launch]: All log files can be found below /home/adip/.ros/log/2023-11-05-18-56-51-553824-adip-236919
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [236926]
[INFO] [controller_stopper_node-3]: process started with pid [236928]
[INFO] [dashboard_client-13]: process started with pid [236992]
[INFO] [controller_stopper_node-14]: process started with pid [236995]
[INFO] [ur_ros2_control_node-1]: process started with pid [236924]
[INFO] [urscript_interface-4]: process started with pid [236930]
[INFO] [robot_state_publisher-5]: process started with pid [236932]
[INFO] [spawner-6]: process started with pid [236934]
[INFO] [spawner-7]: process started with pid [236936]
[INFO] [spawner-8]: process started with pid [236938]
[INFO] [spawner-9]: process started with pid [236940]
[INFO] [spawner-10]: process started with pid [236967]
[INFO] [spawner-11]: process started with pid [236985]
[INFO] [ur_ros2_control_node-12]: process started with pid [236990]
[INFO] [urscript_interface-15]: process started with pid [237003]
[INFO] [robot_state_publisher-16]: process started with pid [237018]
[INFO] [spawner-17]: process started with pid [237029]
[INFO] [spawner-18]: process started with pid [237032]
[INFO] [spawner-19]: process started with pid [237054]
[INFO] [spawner-20]: process started with pid [237057]
[INFO] [spawner-21]: process started with pid [237059]
[INFO] [spawner-22]: process started with pid [237061]
[ur_ros2_control_node-1] [WARN] [1699178212.679082058] [ur10e.controller_manager]: 'update_rate' parameter not set, using default value.
[ur_ros2_control_node-1] [INFO] [1699178212.681652190] [resource_manager]: Loading hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1699178212.692007307] [resource_manager]: Initialize hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1699178212.692087402] [resource_manager]: Successful initialization of hardware 'ur10e'
[ur_ros2_control_node-1] [INFO] [1699178212.692283176] [resource_manager]: 'configure' hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1699178212.692293754] [resource_manager]: Successful 'configure' of hardware 'ur10e'
[ur_ros2_control_node-1] [INFO] [1699178212.692320200] [resource_manager]: 'activate' hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1699178212.692329102] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1699178212.692342518] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1699178212.730797943] [UR_Client_Library:ur10e_]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ur_ros2_control_node-1] [INFO] [1699178212.794921569] [UR_Client_Library:ur10e_]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1699178212.795319229] [UR_Client_Library:ur10e_]: Setting up RTDE communication with frequency 500.000000
[robot_state_publisher-5] [INFO] [1699178212.849754286] [ur10e.robot_state_publisher]: got segment ur10e_base
[robot_state_publisher-5] [INFO] [1699178212.849872070] [ur10e.robot_state_publisher]: got segment ur10e_base_link
[robot_state_publisher-5] [INFO] [1699178212.849882224] [ur10e.robot_state_publisher]: got segment ur10e_base_link_inertia
[robot_state_publisher-5] [INFO] [1699178212.849887579] [ur10e.robot_state_publisher]: got segment ur10e_flange
[robot_state_publisher-5] [INFO] [1699178212.849893272] [ur10e.robot_state_publisher]: got segment ur10e_forearm_link
[robot_state_publisher-5] [INFO] [1699178212.849898104] [ur10e.robot_state_publisher]: got segment ur10e_ft_frame
[robot_state_publisher-5] [INFO] [1699178212.849902971] [ur10e.robot_state_publisher]: got segment ur10e_shoulder_link
[robot_state_publisher-5] [INFO] [1699178212.849907966] [ur10e.robot_state_publisher]: got segment ur10e_tool0
[robot_state_publisher-5] [INFO] [1699178212.849912729] [ur10e.robot_state_publisher]: got segment ur10e_upper_arm_link
[robot_state_publisher-5] [INFO] [1699178212.849917005] [ur10e.robot_state_publisher]: got segment ur10e_wrist_1_link
[robot_state_publisher-5] [INFO] [1699178212.849921708] [ur10e.robot_state_publisher]: got segment ur10e_wrist_2_link
[robot_state_publisher-5] [INFO] [1699178212.849926484] [ur10e.robot_state_publisher]: got segment ur10e_wrist_3_link
[robot_state_publisher-5] [INFO] [1699178212.849931703] [ur10e.robot_state_publisher]: got segment world
[ur_ros2_control_node-12] [WARN] [1699178212.734697114] [ur5e.controller_manager]: 'update_rate' parameter not set, using default value.
[ur_ros2_control_node-12] [INFO] [1699178212.735176779] [resource_manager]: Loading hardware 'ur5e' 
[ur_ros2_control_node-12] [INFO] [1699178212.737773818] [resource_manager]: Initialize hardware 'ur5e' 
[ur_ros2_control_node-12] [INFO] [1699178212.737845824] [resource_manager]: Successful initialization of hardware 'ur5e'
[... cut ...]
[ERROR] [spawner-18]: process has died [pid 237032, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner joint_state_broadcaster --controller-manager controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur5e'].
[ERROR] [spawner-21]: process has died [pid 237059, exit code 1, cmd '/home/adip/colcon_ws/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner force_torque_sensor_broadcaster --controller-manager controller_manager --controller-manager-timeout 10 --ros-args -r __ns:=/ur5e'].
`

medvedevigorek, 2023-11-05:
You need to add namespace prefix to entire `controller_manager` section of your controllers configuration file. The default one takes care of prefix in joint names but assumes that controller manager is running the global namespace. I use `/*:` as the top tag in the controllers file as controllers are the same for both arms in my case. You can do the same or have separate section for each arm prefixed with its namespace if controllers are different.

medvedevigorek, 2023-11-05:
And one more thing: from your launch file snippet you just pass `controllers_file` but `ur_control.launch.py` will be [looking](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/launch/ur_control.launch.py#L201) for that file in `runtime_config_package` which is by default set to `ur_robot_driver`. You may need to change it to your package where your controllers file is located

cucudasluv, 2023-11-09:
@medvedevigorek Thank you for you comment. following what you mentioned about '/*:' , I could solve dual arm bring up problem! I appreciate to your help and I also check that it works well integrated with Moveit2.

closed 2023-11-09 by cucudasluv

along893, 2024-02-07:
Hi! 

Could you have one for ROS2 Foxy?