UniversalRobots/Universal_Robots_ROS2_Driver #869: CI failing due to missing controller_manager

fmauch, 2023-10-30:
### Affected ROS2 Driver version(s)

Basically all PRs

### Used ROS distribution.

Iron, Rolling

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

The one from CI

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
Currently, some of our CI jobs regularly fail due to a missing connnection to `/controller_manager`

## Issue details
It seems that sometimes our controller spawners seem not to be able to contact the controller_manager. For single controllers we get 

```
 [spawner-8] [INFO] [1698650407.667882462] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
    [spawner-8] [INFO] [1698650409.677460494] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
    [spawner-8] [INFO] [1698650411.687458830] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
    [spawner-8] [INFO] [1698650413.697357365] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
    [spawner-8] [INFO] [1698650415.708340132] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
```
And the tests finally fail because the controller never comes up.

## Steps to Reproduce
Not sure how to reproduce this, as this happens in some pipeline runs, but not all.

## Expected Behavior
It should work

## Actual Behavior
e.g. https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/actions/runs/6689460108/job/18173039411

# Workaround Suggestion
none..

### Accept Public visibility

- [X] I agree to make this context public

KKSTB, 2023-10-31:
Controller spawner has a wait for service timeout of 10sec only. Could this be caused by a newly started ursim where the boot time takes more than 10sec?

fmauch, 2023-10-31:
> Controller spawner has a wait for service timeout of 10sec only. Could this be caused by a newly started ursim where the boot time takes more than 10sec?

No, we specifically [pass a timeout of 120 seconds](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/b28a8700ffe3994f5c0ddb165762db88a8957bb3/ur_robot_driver/test/test_common.py#L321) in there. But thank you for the input @KKSTB.

urfeex, 2025-01-15:
This got fixed upstream

closed 2025-01-15 by urfeex