UniversalRobots/Universal_Robots_ROS2_Driver #889: No Planning Library Loaded

kroglice, 2023-12-06:
### Affected ROS2 Driver version(s)

Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.14

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

Hello,

I am trying to use this example to control an ur3e sim and real robot, however when i follow the instructions my moveit tells me "No Planning Library Loaded" and the robot does not show in the moveit.

But if i just launch "ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.1.101 launch_rviz:=true initial_joint_controller:=joint_trajectory_controller" and i control the robot via its own software it shows on the rviz and i see it moving.

Can you help me? or give me some guidance. thank you in advance.

I am struggling with this problem.

### Accept Public visibility

- [X] I agree to make this context public

ER-ROUGUI, 2023-12-06:
the same issue

RobertWilbrandt, 2024-02-14:
Sorry for not answering earlier. If this is still relevant to you: Can you please show the exact set of commands and the full log output that lead to this error? The following sequence works fine for me:
1. `ros2 run ur_client_library start_ursim.sh -m ur3e` and start the robot (Using ursim instead of a real robot should not make a difference in this case
2. `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.56.101 headless_mode:=true`
3. `LANG=c ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e`
4. Planning through e.g. the rviz that opened with the previous command works without error.

I'm especially confused because you used the binary packets, which should automatically install `moveit_planners_ompl` on your system. Can you check if `ros-humble-moveit-planners-ompl` is correctly installed?

fmauch, 2024-07-15:
Closing this due to inactivity. If the issue still persists, please feel free to comment / reopen.

closed 2024-07-15 by fmauch