UniversalRobots/Universal_Robots_ROS2_Driver #901: Project -> subscribe and publish to robot

Daan0103, 2024-01-03:
### Feature summary

Hii :)

I'm doing a project with the UR5e and want to make a program which uses data received from the robot like actual velocities, force etc.
With this package I'm able to move the UR5e to a wanted position by using MoveIt2. But when I'm right it's not possible to use/get live data from the robot. I probably have to use ur_rtde for this?

I'm using:
Robot:
- Universal Robots UR5e

Laptop with:
- Linux 22.04
- ROS2 Humble

Is the ur_rtde (https://www.universal-robots.com/articles/ur/interface-communication/real-time-data-exchange-rtde-guide/) also working in this environment, and can I use this for my project?
Or is it possible to achieve my project goal (build a program which subscribes and publish to the UR5e --> communication between both) on another way?
Anyone with same kind of project?

RobertWilbrandt, 2024-02-14:
Hey, i'm sorry for not answering earlier. If this is still an open question to you: You are absolutely able to read out the live status data from the robot through this driver (it also uses the rtde interface after all). For joint data (position, velocity and motor current) you can take a look at the `/joint_states` topic, and the force + torque from the integrated sensor can be read from the `force_torque_sensor_broadcaster/wrench` topic. Using this directly should be easier than using the ROS driver and ur_rtde at the same time, as there are some issues with both programs claiming the same ressources on the robot otherwise. Please let us know if this works for you or if you have further questions.

fmauch, 2024-07-15:
This seems to be answered, hence closing this.

closed 2024-07-15 by fmauch