UniversalRobots/Universal_Robots_ROS2_Driver #904: ROS Humble workspace building fails after having updated it after a while

francescosemeraro, 2024-01-07:
### Affected ROS2 Driver version(s)

Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.12

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I am not able to build my unchanged ROS 2 Humble UR driver workspace. 

## Issue details
I had downloaded the driver almost a year ago and has functioned very well until now. Suddenly, I was not able to build the workspace anymore, as I was receiving a "Missing command interfaces" error, which is something related to the release itself. I saw there were some similar issues posted here around last May. I looked them up and tried to update everything, but I couldn't manage to make it work again yet.
I am using them most up-to-date version of ROS 2 Humble, and I still have the humble branch set on my laptop. When I try to build the workspace, I get the following error:

`--- stderr: ur_controllers                                                                               
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In lambda function:
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:98:9: error: ‘joints_angle_wraparound_’ was not declared in this scope
   98 |     if (joints_angle_wraparound_[index]) {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:120:55: error: ‘rt_has_pending_goal_’ was not declared in this scope
  120 |   if (current_external_msg != *new_external_msg && (*(rt_has_pending_goal_.readFromRT()) && !active_goal) == false) {
      |                                                       ^~~~~~~~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:137:3: error: ‘read_state_from_state_interfaces’ was not declared in this scope; did you mean ‘read_state_from_command_interfaces’?
  137 |   read_state_from_state_interfaces(state_current_);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   read_state_from_command_interfaces
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:140:7: error: ‘has_active_trajectory’ was not declared in this scope
  140 |   if (has_active_trajectory()) {
      |       ^~~~~~~~~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:184:35: error: ‘rt_is_holding_’ was not declared in this scope
  184 |       if (!before_last_point && *(rt_is_holding_.readFromRT()) == false && cmd_timeout_ > 0.0 &&
      |                                   ^~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:184:76: error: ‘cmd_timeout_’ was not declared in this scope
  184 |       if (!before_last_point && *(rt_is_holding_.readFromRT()) == false && cmd_timeout_ > 0.0 &&
      |                                                                            ^~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:189:62: error: invalid use of void expression
  189 |         traj_msg_external_point_ptr_.initRT(set_hold_position());
      |                                             ~~~~~~~~~~~~~~~~~^~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:200:15: error: ‘rt_is_holding_’ was not declared in this scope
  200 |             *(rt_is_holding_.readFromRT()) == false) {
      |               ^~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:207:15: error: ‘rt_is_holding_’ was not declared in this scope
  207 |             *(rt_is_holding_.readFromRT()) == false) {
      |               ^~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:270:11: error: ‘rt_has_pending_goal_’ was not declared in this scope
  270 |           rt_has_pending_goal_.writeFromNonRT(false);
      |           ^~~~~~~~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:275:64: error: invalid use of void expression
  275 |           traj_msg_external_point_ptr_.initRT(set_hold_position());
      |                                               ~~~~~~~~~~~~~~~~~^~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:284:13: error: ‘rt_has_pending_goal_’ was not declared in this scope
  284 |             rt_has_pending_goal_.writeFromNonRT(false);
      |             ^~~~~~~~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:289:49: error: ‘set_success_trajectory_point’ was not declared in this scope
  289 |             traj_msg_external_point_ptr_.initRT(set_success_trajectory_point());
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:297:13: error: ‘rt_has_pending_goal_’ was not declared in this scope
  297 |             rt_has_pending_goal_.writeFromNonRT(false);
      |             ^~~~~~~~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:303:66: error: invalid use of void expression
  303 |             traj_msg_external_point_ptr_.initRT(set_hold_position());
      |                                                 ~~~~~~~~~~~~~~~~~^~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:306:55: error: ‘rt_has_pending_goal_’ was not declared in this scope
  306 |       } else if (tolerance_violated_while_moving && *(rt_has_pending_goal_.readFromRT()) == false) {
      |                                                       ^~~~~~~~~~~~~~~~~~~~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:311:62: error: invalid use of void expression
  311 |         traj_msg_external_point_ptr_.initRT(set_hold_position());
      |                                             ~~~~~~~~~~~~~~~~~^~
/home/francesco/final_system_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:316:62: error: invalid use of void expression
  316 |         traj_msg_external_point_ptr_.initRT(set_hold_position());
      |                                             ~~~~~~~~~~~~~~~~~^~
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [26.3s, exited with code 2]
Aborted  <<< robotic_system [38.5s]                                   

Summary: 7 packages finished [57.0s]
  1 package failed: ur_controllers
  1 package aborted: robotic_system
  3 packages had stderr output: azure_kinect_ros_driver robotic_system ur_controllers
  4 packages not processed
`
I am not sure why this is happening. If I revert the driver at 2.2.6, the workspace is built properly. But, if at that point, I run the "ros2 launch ur_robot_driver ur_control.launch.py" command, I get the missing command interfaces error from ur_controllers.

Can you please help me out? Thanks!

### Accept Public visibility

- [X] I agree to make this context public

Daan0103, 2024-01-09:
Is there a .zip file from version 2.2.9 (or another way), to build version 2.2.9, humble, from source?

francescosemeraro, 2024-01-09:
Hi Daan,

are you referring to the 2.2.9 version of ROS 2 Humble or the one of the UR ROS 2 driver? 
If you are referring to the first one, then yes, I was considering reverting ROS2 Humble to the version I used to have from beginning of 2023, but I am struggling a little bit with this. Do you know how to do it? And how to install the related ROS2 libraries compatible wiht that release of ROS 2 Humble?

bmdyrdal, 2024-01-10:
> Is there a .zip file from version 2.2.9 (or another way), to build version 2.2.9, humble, from source?

Hi! You can use the version parameter in the ros2.repos file. 

```
UniversalRobots/Universal_Robots_ROS2_Driver:
        type: git
        url: git@github.com:UniversalRobots/Universal_Robots_ROS2_Driver.git
        version: 2.2.9
```

Daan0103, 2024-01-15:
Hii,
@francescosemeraro I'm reffering to the version of the ur_driver package. Why exactly you want an older ROS2 Humble version?

@bmdyrdal Thanks! I've installed the right version now. 
My problem is solved :)

francescosemeraro, 2024-01-15:
I was trying to make my code work and it was using a version of the driver from February 2023. 
Thankfully I managed to make everything work with the last version. Thanks again for the support.

By the way is it possible to permanently change the urdf of the robot and use it in the newest version of the driver for ROS 2 Humble? I located the xacro files in the share folder of ur_description package and edited it the way I would like, but the package does not seem to register the change. So, I wonder if it is not possible to edit it anymore or I am doing something wrong?

Kind regards,

Francesco

Daan0103, 2024-01-15:
Changes are only possible when you installed the ur_description from source.
When using this package, make sure you make use of this from source builded package.

Source your workspace:
`$ source ros2_ws/install/setup.bash`

This should work fine. To be sure of this you could see how the package is declared inside your package.
Can you confirm you installed the ur_description package from source inside your workspace?

You could also delete the ur_description package when you also installed it also from binary which shall mean you got the package installed multiple times which could cause problems.

Daan0103, 2024-01-16:
I had the same problem with building from source.
For me it worked to install all needed packages form source:
- ur_description
- ur_client_library
- ur_driver

For the ur_driver you will get the errors indead, so the ur_driver package I installed is the previous version.
I downloaded version 2.2.9. as a .zip file and unzipped it into the 'src' directory within my workspace.

So my 'src' includes all three packages.
When do colcon build, this gave no errors.

francescosemeraro, 2024-01-16:
I have just tried what you said ("humble" branch for Universal_Robots_ROS2_Description, "master" branch for Universal_Robots_Client_library and "2.2.9" branch for Universal_Robots_ROS2_Driver). When I build the workspace with these three within, I get the following error:
`--- stderr: ur_controllers                              
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In lambda function:
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:93:9: error: ‘normalize_joint_error_’ was not declared in this scope
   93 |     if (normalize_joint_error_[index]) {
      |         ^~~~~~~~~~~~~~~~~~~~~~
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:117:5: error: ‘traj_point_active_ptr_’ was not declared in this scope
  117 |     traj_point_active_ptr_ = &traj_external_point_ptr_;
      |     ^~~~~~~~~~~~~~~~~~~~~~
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:130:3: error: ‘read_state_from_hardware’ was not declared in this scope
  130 |   read_state_from_hardware(state_current_);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:133:7: error: ‘traj_point_active_ptr_’ was not declared in this scope
  133 |   if (traj_point_active_ptr_ && (*traj_point_active_ptr_)->has_trajectory_msg()) {
      |       ^~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [3.27s, exited with code 2]
`
Has this ever happened to you?

francescosemeraro, 2024-01-16:
I have just tried what you said ("humble" branch for Universal_Robots_ROS2_Description, "master" branch for Universal_Robots_Client_library and "2.2.9" branch for Universal_Robots_ROS2_Driver). When I build the workspace with these three within, I get the following error:

--- stderr: ur_controllers                              
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In lambda function:
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:93:9: error: ‘normalize_joint_error_’ was not declared in this scope
   93 |     if (normalize_joint_error_[index]) {
      |         ^~~~~~~~~~~~~~~~~~~~~~
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:117:5: error: ‘traj_point_active_ptr_’ was not declared in this scope
  117 |     traj_point_active_ptr_ = &traj_external_point_ptr_;
      |     ^~~~~~~~~~~~~~~~~~~~~~
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:130:3: error: ‘read_state_from_hardware’ was not declared in this scope
  130 |   read_state_from_hardware(state_current_);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:133:7: error: ‘traj_point_active_ptr_’ was not declared in this scope
  133 |   if (traj_point_active_ptr_ && (*traj_point_active_ptr_)->has_trajectory_msg()) {
      |       ^~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

Failed   <<< ur_controllers [3.27s, exited with code 2]

Has this ever happened to you?

Daan0103, 2024-01-16:
Did you do a clean build?
(delete workspace / '**build'** directory)

For me this build gave no errors.

francescosemeraro, 2024-01-16:
Yes, I did. I ran:
`rm -rf build/ install/ `
at the ros_ur_driver level. I also tried to se if building ros2_control_node from source does the trick ("humble" branch). Now, this error comes up:

/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:319:16: error: no matching function for call to ‘ur_controllers::ScaledJointTrajectoryController::publish_state(const rclcpp::Time&, trajectory_msgs::msg::JointTrajectoryPoint&, trajectory_msgs::msg::JointTrajectoryPoint&, trajectory_msgs::msg::JointTrajectoryPoint&)’
  319 |   publish_state(time, state_desired_, state_current_, state_error_);
      |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
                 from /home/francesco/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
/opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:272:8: note: candidate: ‘void joint_trajectory_controller::JointTrajectoryController::publish_state(const JointTrajectoryPoint&, const JointTrajectoryPoint&, const JointTrajectoryPoint&)’
  272 |   void publish_state(
      |        ^~~~~~~~~~~~~
/opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:272:8: note:   candidate expects 3 arguments, 4 provided
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [12.1s, exited with code 2]
Aborted  <<< controller_manager [47.3s]                                         

Summary: 12 packages finished [1min 52s]
  1 package failed: ur_controllers
  1 package aborted: controller_manager
  3 packages had stderr output: controller_manager ur_client_library ur_controllers
  5 packages not processed

VinDp, 2024-01-30:
Hi @francescosemeraro,
sorry for the late reply, do you still have this issue?

francescosemeraro, 2024-02-07:
> Hi @francescosemeraro, sorry for the late reply, do you still have this issue?

Good afternoon,

I probably still have this issue, but thankfully I managed to find a turnaround. Instead of changing the urdf, I used an AttachedCollisionObject that mimicks my end effector with good approximation.However, it would be good to change the urdf as was suggested for future works.

By the way, @VinDp , I have recently opened another issue related to the driver [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/925), could you please help me out there?

Thanks!

jssaditi, 2024-02-22:
> Hi @francescosemeraro, sorry for the late reply, do you still have this issue?

Hi @VinDp,

I am also using ROS humble and I am getting the same error regarding the publish_state function when I build. 

`--- stderr: ur_controllers                   
/home/aditi/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/aditi/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:319:16: error: no matching function for call to ‘ur_controllers::ScaledJointTrajectoryController::publish_state(const rclcpp::Time&, trajectory_msgs::msg::JointTrajectoryPoint&, trajectory_msgs::msg::JointTrajectoryPoint&, trajectory_msgs::msg::JointTrajectoryPoint&)’
  319 |   publish_state(time, state_desired_, state_current_, state_error_);
      |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/aditi/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
                 from /home/aditi/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
/opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:260:8: note: candidate: ‘void joint_trajectory_controller::JointTrajectoryController::publish_state(const JointTrajectoryPoint&, const JointTrajectoryPoint&, const JointTrajectoryPoint&)’
  260 |   void publish_state(
      |        ^~~~~~~~~~~~~
/opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:260:8: note:   candidate expects 3 arguments, 4 provided
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [5.53s, exited with code 2]
Aborted  <<< admittance_controller [6.76s]              
Aborted  <<< velocity_controllers [42.1s]   

Summary: 36 packages finished [5min 45s]
  1 package failed: ur_controllers
  2 packages aborted: admittance_controller velocity_controllers
  7 packages had stderr output: controller_manager ros2_controllers_test_nodes ros2controlcli rqt_controller_manager rqt_joint_trajectory_controller ur_client_library ur_controllers
  4 packages not processed
` 
What should I do to fix this issue?

jssaditi, 2024-02-22:
At first I was using the `src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.${ROS_DISTRO}.repos` as the input file for the `vcs import command`, but that file is currently empty...

So instead I also tried using `src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.${ROS_DISTRO}.repos` as the input file before building. But I still get the same error either way.

fmauch, 2024-07-15:
I am closing this assuming that the problem no longer persists. The mentioned API changes in the upstream library have been merged and released quite some time ago and the humble version builds fine both in CI and for the buildfarm.

Feel free to comment / reopen if the issue still persists.

closed 2024-07-15 by fmauch