UniversalRobots/Universal_Robots_ROS2_Driver #907: Ros launch ur_robot_driver ur_control.launch.py

Pecako2001, 2024-01-12:
### Affected ROS2 Driver version(s)

2.1.1

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

5.15.0

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I want to start the ros2 robot driver for a simulated robot but running the command gives error: No such file or directory

## Issue details
*Detailed description help us understand the problem. Code are welcome!*

## Steps to Reproduce
- Build the ROS2 driver for UR
- Run the docker sim 
```
 docker run --rm -it -p 5900:5900 -p 6080:6080 -v ${HOME}/.ursim/urcaps:/urcaps -v ${HOME}/.ursim/programs:/ursim/programs --name ursim universalrobots/ursim_e-series
```

- Launch the command:
```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=<IP> launch_rviz:=true
```
## Expected Behavior
rviz to start up

## Actual Behavior
Error inside terminal

### Relevant log output

```shell
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3 robot_ip:=192.168.56.1 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/engineering/.ros/log/2024-01-12-14-13-10-466769-Engineering-111040
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command failed. Command: /opt/ros/humble/bin/xacro /home/engineering/ur_ros2_driver_ws/install/ur_robot_driver/share/ur_robot_driver/urdf/ur.urdf.xacro robot_ip:=192.168.56.1 joint_limit_params:=/opt/ros/humble/share/ur_description/config/ur3/joint_limits.yaml kinematics_params:=/opt/ros/humble/share/ur_description/config/ur3/default_kinematics.yaml physical_params:=/opt/ros/humble/share/ur_description/config/ur3/physical_parameters.yaml visual_params:=/opt/ros/humble/share/ur_description/config/ur3/visual_parameters.yaml safety_limits:=true safety_pos_margin:=0.15 safety_k_position:=20 name:=ur3 script_filename:=/home/engineering/ur_ros2_driver_ws/install/ur_client_library/share/ur_client_library/resources/external_control.urscript input_recipe_filename:=/home/engineering/ur_ros2_driver_ws/install/ur_robot_driver/share/ur_robot_driver/resources/rtde_input_recipe.txt output_recipe_filename:=/home/engineering/ur_ros2_driver_ws/install/ur_robot_driver/share/ur_robot_driver/resources/rtde_output_recipe.txt tf_prefix:= use_mock_hardware:=false mock_sensor_commands:=false headless_mode:=false use_tool_communication:=false tool_parity:=0 tool_baud_rate:=115200 tool_stop_bits:=1 tool_rx_idle_chars:=1.5 tool_tx_idle_chars:=3.5 tool_device_name:=/tmp/ttyUR tool_tcp_port:=54321 tool_voltage:=0 reverse_ip:=0.0.0.0 script_command_port:=50004 reverse_port:=50001 script_sender_port:=50002 trajectory_port:=50003 
Captured stderr output: error: No such file or directory: /opt/ros/humble/share/ur_description/urdf/inc/ur_joint_control.xacro [Errno 2] No such file or directory: '/opt/ros/humble/share/ur_description/urdf/inc/ur_joint_control.xacro'
when processing file: /home/engineering/ur_ros2_driver_ws/install/ur_robot_driver/share/ur_robot_driver/urdf/ur.ros2_control.xacro
included from: /home/engineering/ur_ros2_driver_ws/install/ur_robot_driver/share/ur_robot_driver/urdf/ur.urdf.xacro
```

### Accept Public visibility

- [X] I agree to make this context public

Manu752, 2024-01-29:
I also have the same issue but  with ROS iron .

MatthijsBurgh, 2024-03-22:
How about checking whether `ur_description` is installed. So check whether the `/opt/ros/humble/share/ur_description` directory exists.
Check the output of `apt-cache policy ros-humble-ur-description`

RobertWilbrandt, 2024-03-27:
It does not look like you actually have the driver installed from binary packages, only the description. From your log:
```
/opt/ros/humble/bin/xacro /home/engineering/ur_ros2_driver_ws/install/ur_robot_driver/share/ur_robot_driver/urdf/ur.urdf.xacro [...]
```

You need to make sure that you either build both from source (see the installation instructions in the README, they show you how to install all direct source dependencies) or both from the binary packages.

fmauch, 2024-04-10:
Closing this as the correct answer seems to have been given in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/907#issuecomment-2022684807. This looks like running the launchfile from the rolling branch of the driver, which does not work with the humble version of the description.

closed 2024-04-10 by fmauch