UniversalRobots/Universal_Robots_ROS2_Driver #908: ROS 2 Humble newest package does not register changes in the urdf file of the robot

francescosemeraro, 2024-01-15:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.12

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
ROS 2 Humble newest package does not register changes in the urdf file of the robot.

## Issue details
In the previous versions of the UR driver for ROS 2 Humble it was possible to edit the urdf file of the robot by manually editing the related xacro file from the share folder of the ur_description package. 
I have recently installed the newest version of the driver for ROS 2 Humble and apparently this feature does not seem to be available anymore. I tried the same procedure, this time from the opt/ros/humble folder but the r package do not seem to register the change. So, I wonder if the packages actually still refer to the xacro files of the ur_description package. Or am I doing something wrong?

### Accept Public visibility

- [X] I agree to make this context public

VinDp, 2024-02-05:
Hi @francescosemeraro, 
if you want to change the urdf you shouldn't modify it in the share folder. The correct way to do it is to create your own description package where you have a dedicated urdf.xacro file which includes ```ur_macro.xacro``` from ur_description, like this:  
```<xacro:include filename="$(find ur_description)/urdf/ur_macro.xacro" />```.

After that, you can add more elements to the urdf  (e.g. a gripper) and set the parameters you prefer. An example to do so:
```
<xacro:ur_robot name="ur5e"
                    tf_prefix="${prefix}"
                    parent="<parent_frame>"
                    robot_ip="${robot_ip}"
                    headless_mode="true"
                    initial_positions="${dict(shoulder_pan_joint=-1.5707,shoulder_lift_joint=0.0,elbow_joint=-1.5707,wrist_1_joint=0,wrist_2_joint=0,wrist_3_joint=0)}"
                    joint_limits_parameters_file="$(find ur_description)/config/ur5e/joint_limits.yaml"
                    kinematics_parameters_file="<path_to_calibration_file>"
                    physical_parameters_file="$(find ur_description)/config/ur5e/physical_parameters.yaml"
                    visual_parameters_file="$(find ur_description)/config/ur5e/visual_parameters.yaml"
                    script_filename="$(find ur_client_library)/resources/external_control.urscript"
                    input_recipe_filename="$(find ur_robot_driver)/resources/rtde_input_recipe.txt"
                    output_recipe_filename="$(find ur_robot_driver)/resources/rtde_output_recipe.txt"
                    use_fake_hardware="${use_fake_hardware}"
      <origin xyz="0 0 0" rpy="0 0 ${pi}" /> 
    </xacro:ur_robot>
```
You can also check a more complete example [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Tutorials/blob/ros2/ur_example_dual_robot/urdf/dual_robot.xacro).

Then it will also be necessary to specify this description package and urdf when launching, by using the parameters ```description_package``` and ```description_file```, like:

```ros2 launch ur_robot_driver ur_control.launch.py ur_type:=<UR_TYPE> robot_ip:=<IP_OF_THE_ROBOT> description_package:=<NEW_PACKAGE_PATH> description_file:=<NEW_DESCRIPTION_FILE>```

or just creating a new personal launch file where the new description package and file are set as default options.

fmauch, 2024-04-10:
Closing this as a proper solution has been proposed.

closed 2024-04-10 by fmauch