UniversalRobots/Universal_Robots_ROS2_Driver #909: [scaled_joint_trajectory_controller]: Can't accept new action goals. Controller is not running.

Rudresh172, 2024-01-16:
### Affected ROS2 Driver version(s)

2.2.10

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.12.0 / 5.12.2 / 5.14.6

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
The scaled trajectory controller is inactive after starting the ur driver.
Tried it on both ur5e and ur10e.
Possibly after the new update from ros2.

## Issue details
When I launch the ur_driver, the controllers are not active. Hence when I give execute command on MoveIt2 plugin, it gives the error 
```
[ur_ros2_control_node-1] [ERROR] [1705415302.914893185] [scaled_joint_trajectory_controller]: Can't accept new action goals. Controller is not running.
```
When I run the list_controllers command, it gives the following output - 
```
ros2 control list_controllers
forward_position_controller[position_controllers/JointGroupPositionController] inactive  
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active    
io_and_status_controller[ur_controllers/GPIOController] active    
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active    
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] inactive 
```

## Steps to Reproduce
```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=<ur_type> robot_ip:=<robot_ip>
```
## Expected Behavior
All the controller to be active

## Actual Behavior
Trajectory controller is not active, causing the moveit2 commands to not execute.

### Relevant log output

```shell
not_windows@LabTop3:~$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.71.120
[INFO] [launch]: All log files can be found below /home/not_windows/.ros/log/2024-01-16-15-27-53-885308-LabTop3-17710
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [17715]
[INFO] [controller_stopper_node-3]: process started with pid [17717]
[INFO] [ur_ros2_control_node-1]: process started with pid [17713]
[... 78 lines cut ...]
[ur_ros2_control_node-1] [INFO] [1705415277.228507360] [speed_scaling_state_broadcaster]: Publisher rate set to : 100.0 Hz
[ur_ros2_control_node-1] [INFO] [1705415277.230278295] [controller_manager]: Configuring controller 'scaled_joint_trajectory_controller'
[ur_ros2_control_node-1] [INFO] [1705415277.230451988] [scaled_joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ur_ros2_control_node-1] [INFO] [1705415277.230473971] [scaled_joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ur_ros2_control_node-1] [INFO] [1705415277.230483364] [scaled_joint_trajectory_controller]: Using 'splines' interpolation method.
[ur_ros2_control_node-1] [INFO] [1705415277.231344253] [scaled_joint_trajectory_controller]: Controller state will be published at 100.00 Hz.
[ur_ros2_control_node-1] [INFO] [1705415277.232286890] [scaled_joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-8] [INFO] [1705415277.240667987] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[spawner-11] [INFO] [1705415277.244754193] [spawner_force_torque_sensor_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[spawner-9] [INFO] [1705415277.252932801] [spawner_io_and_status_controller]: Configured and activated io_and_status_controller
[spawner-10] [INFO] [1705415277.257043183] [spawner_speed_scaling_state_broadcaster]: Configured and activated speed_scaling_state_broadcaster
[spawner-7] [INFO] [1705415277.261018659] [spawner_scaled_joint_trajectory_controller]: Configured and activated scaled_joint_trajectory_controller
[INFO] [spawner-12]: process has finished cleanly [pid 17735]
[INFO] [spawner-8]: process has finished cleanly [pid 17727]
[INFO] [spawner-11]: process has finished cleanly [pid 17733]
[INFO] [spawner-9]: process has finished cleanly [pid 17729]
[INFO] [spawner-10]: process has finished cleanly [pid 17731]
[INFO] [spawner-7]: process has finished cleanly [pid 17725]
[ur_ros2_control_node-1] [INFO] [1705415302.914857595] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [ERROR] [1705415302.914893185] [scaled_joint_trajectory_controller]: Can't accept new action goals. Controller is not running.
```

### Accept Public visibility

- [X] I agree to make this context public

PhDittmann, 2024-01-24:
I had the same issue with URSim. Please try running in terminal:

```shell
ros2 service call /controller_manager/switch_controller controller_manager_msgs/srv/SwitchController '{"activate_controllers": ["scaled_joint_trajectory_controller"], "deactivate_controllers": ["joint_trajectory_controller", "forward_velocity_controller","forward_position_controller"], "strictness": 1}'
```

Rudresh172, 2024-01-26:
Hello @PhDittmann ,

Thanks for your reply. I tried the command, but it ended up giving the following error - 
`
[ur ros2_control_node-1] [WARN] [1706281297.843853243] [scaled joint_trajectory_controller]: Aborted due to state tolerance violation
`

But if I restart the external urcap and launch the ur_driver file again, it starts working. I don't know if I am doing anything incorrectly.

Rudresh172, 2024-03-12:
Hello @RobertWilbrandt,

I was performing the operations in incorrect order. When I followed the order mentioned by you, it works correctly. I was connected over WLAN, so it would also drop connections and give same error, so I got confused. Now I connect with a LAN cable and it works correctly.

closed 2024-03-12 by Rudresh172