UniversalRobots/Universal_Robots_ROS2_Driver #910: controller unconfigured

Daan0103, 2024-01-19:
### Affected ROS2 Driver version(s)

9.9.2

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

-

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

Hi, i got a problem working on the UR5e, 
I had done all safety steps and runnen the cartesian_force_controller succesfully on the UR5e. Now, when i do it again i want to deactivate the scaled_joint_trajectory_controller. When I command ros2 control list_controllers the following line prints:

....
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] unconfigured
.....

All other controller are on active or inactive.
The robot does not do anything.

### Accept Public visibility

- [X] I agree to make this context public

adamleon, 2024-01-31:
Any news on this? I'm facing the same problem. Both the scaled joint trajectory controller and the joint trajectory controller has the same problem. The forward position controller also has problems, but that's something else.

Daan0103, 2024-02-01:
For me it worked to install the Universal Robots ROS2 Driver again, and from source.
When building the workspace and try it again, it was solved.

VinDp, 2024-09-18:
Since the issue seems to have been solved, I will close it.
@adamleon if the problem still persists for you, feel free to open a new issue.

closed 2024-09-18 by VinDp