UniversalRobots/Universal_Robots_ROS2_Driver #911: Pipeline producer overflowed! <RTDE Data Pipeline>

Hytac, 2024-01-24:
### Affected ROS2 Driver version(s)

017a54c3991cb3f576d1d6dc0e7d039ef0770826

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot, URSim in docker

### Robot SW / URSim version(s)

(Baseline/Polyscope/Controller/Firmware/Platofrm) UR10e 11.3.1001/74.11.1642/41.6.15/32.3.151/0.1.1 URSim 12.3.417/74.13.232/66.7.16/38.7.85/0.1.1

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
My initial_joint_controllers file includes the following boradcaster and controllers:

```yaml
joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

io_and_status_controller:
      type: ur_controllers/GPIOController

speed_scaling_state_broadcaster:
      type: ur_controllers/SpeedScalingStateBroadcaster

force_torque_sensor_broadcaster:
      type: force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster

joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

scaled_joint_trajectory_controller:
      type: ur_controllers/ScaledJointTrajectoryController

forward_velocity_controller:
      type: velocity_controllers/JointGroupVelocityController

forward_position_controller:
      type: position_controllers/JointGroupPositionController

forward_position_controller_zstage:
      type: forward_command_controller/ForwardCommandController

joint_trajectory_controller_zstage:
      type: joint_trajectory_controller/JointTrajectoryController
```  

Last two entries are related to a prismatic axis where the UR10e is mounted. 
This file is pass to the ur_robot_driver ur_ros2_control_node as shown in here everything is configred

```bash
[ur_ros2_control_node-1] [INFO 1706012200.571395246] [controller_manager]: Loading controller 'scaled_joint_trajectory_controller' (load_controller() at ./src/controller_manager.cpp:279)
[... 26 lines cut ...]
```bash
[ur_ros2_control_node-1] [INFO 1706012225.218901302] [UR_Client_Library:ur_]: Robot requested program (() at ./src/control/script_sender.cpp:58)
[ur_ros2_control_node-1] [INFO 1706012225.218952951] [UR_Client_Library:ur_]: Sent program to robot (() at ./src/control/script_sender.cpp:71)
[ur_ros2_control_node-1] [INFO 1706012225.267773292] [UR_Client_Library:ur_]: Robot connected to reverse interface. Ready to receive control commands. (() at ./src/control/reverse_interface.cpp:219)
[ur_ros2_control_node-1] SerialComTlt::setColumnSize 0.0435531 PRISMATIC AXIS IS MOVING NOW
[ur_ros2_control_node-1] [ERROR 1706012261.287439809] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.289402578] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.291397575] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.293390850] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.295394266] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.297405407] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.299419470] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.301414504] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.303436438] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.305388572] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.307448800] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.309384373] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.311449759] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
[ur_ros2_control_node-1] [ERROR 1706012261.313388964] [UR_Client_Library:ur_]: Pipeline producer overflowed! <RTDE Data Pipeline> (() at ./include/ur_client_library/comm/pipeline.h:406)
```
While the prismatic axis is moving the errors appear in a loop. when moving the real UR the error does not appear. After the error i can keep using both controllers.

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-05-15:
This sounds like the hardware interface of your prismatic axis is slowing down the control loop while executing motions.

If the control loop isn't keeping up with the 500Hz, the robot is sending in data while it has not been processed by the hardware interface, hence the `Pipeline producer overflowed!` error.

ros2_control has a relatively new feature for running components and controllers asynchronously, that could probably help you there, but I didn't test that, yet.

urfeex, 2025-02-06:
I don't see anything that could be done inside this driver. Either use two separate control nodes or try ros2_control's async hardware feature.

@Elfits I'm not sure whether your problem is similar, as you say you use RTDE. If your problem still persists, please open a new issue specifically describing your use case and problems.

closed 2025-02-06 by urfeex