UniversalRobots/Universal_Robots_ROS2_Driver #915: Documentation on usage of gazebo simulation

urmahp, 2024-01-29:
### Feature summary

We should let people now that the robot can be simulated with the [gazebo repository](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation) or the [ignition repository](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html).  

Adding the documentation in the usage documentation of this driver would add more awareness on different ways to simulate your solution before testing it on real hardware. I therefore suggest to add a section in the usage documentation where we explain gazebo/ignition simulation and a description on how to set it up or a link to the repository with a description on how to set it up.

fmauch, 2024-02-04:
Just a note: IMHO this should not necessarily be part of this repo, but documented inside their respective repos and we should finally build our comprehensive documentation putting together all the different repos into one deployed doc.

urmahp, 2024-02-05:
I wasn't aware of that, but that sounds like a much better solution. I am just under the impression that people isn't aware of the possibility of using gazebo with UR robots. I thought  a good way of creating more awareness, was to put some information about it inside this drivers documentation, but I prefer the idea of one comprehensive documentation.

RobinHeitz, 2024-04-30:
So, what is the proper way then? I'm struggling with it.
So far, I'm using the UR10e moveit package + driver to visualize in rviz or control the real robot.

I have seen that the urdf supports something with gazebo but it seems like both mentioned packages do not give the possibility to set sim_gazebo to true. How should it be done in humble?