UniversalRobots/Universal_Robots_ROS2_Driver #917: [FEATURE NAME]

odbee, 2024-01-30:
### Feature summary

Hey, so it seems the current ur_moveit_config folder is not at all compatible with being used with  the MoveItConfigsBuilder. It would help greatly with running moveit directly through the C++ API if  one could take all the moveitconfigbuilder parameters and add them to the parameters of ones c++ API like this: `parameters=[moveit_config.to_dict()]`

### Tasks
*To complete this issue involves*
- [ ] Implement the feature
- [ ] Make documentation
- [ ] Make Unit test
- [ ] Make example
- [ ] Test on real hardware

VinDp, 2024-05-08:
Hey @odbee, 
thanks for the suggestion, we are currently working on that.
We are going to change the structure according to what is shown in this [tutorial page](https://github.com/UniversalRobots/Universal_Robots_ROS2_Tutorials/tree/main/my_robot_cell/my_robot_cell_moveit_config), do you have any opinions on that?

odbee, 2024-05-08:
Hey @VinDp thanks, for consering my Feature request, looks good! the structure looks good. I think it would be nice to have both scaled_joint_trajectory_controller and regular joint_trajectory_controller available in the "moveit_controllers.yaml", as well as different inverse kinematics  ( Trac-IK, Pic IK) and planning (pilz_industrial_motion_planner_planning_planner.yaml , ompl_planner.yaml) files,

fmauch, 2024-05-08:
Once issue I see with this is that the `robot_name` for the `robot_description` will be different for each robot model (e.g. `ur5e`. Because of that we currently use an xacro to generate the SRDF. As far as I understand the [MoveItConfigsBuilder](https://github.com/moveit/moveit2/blob/main/moveit_configs_utils/moveit_configs_utils/moveit_configs_builder.py) it doesn't support that but expects a finished srdf file.

closed 2024-06-11 by fmauch