UniversalRobots/Universal_Robots_ROS2_Driver #918: Passing "robot_calibration.yaml" not working via lauch-parameter "kinematics_config"

A13X15, 2024-01-30:
### Affected ROS2 Driver version(s)

Package: ros-humble-ur Version: 2.2.10-1jammy.20240126.081445

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

3.12.0

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Launching and passing the robot_calibration.yaml file via the launch-parameter kinematics_config does not work.

## Steps to Reproduce
The launch-command I used was
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10 robot_ip:=my_robots_ip launch_rviz:=false initial_joint_controller:=joint_trajectory_controller kinematics_config:=/path/to/my_robot_calibration.yaml`

The calibration-file was extracted via the tutorial in this repo while the robot was turned on. I'm certain that my calibration file and path to it are correct.

When launching the driver, I get multiple deprecation warnings, one of them is about passing the robot description parameter. Once the ur_ros2_control_node-1 gets to the calibration checksum, it does not output the checksum of my yaml-file, but the default checksum 'calib_17227329492635474227' from the [default UR10 kinematics file](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/rolling/config/ur10/default_kinematics.yaml). So the correct calibration file is certainly not passed. This obviously results in an error (mismatched checksums).

## Expected Behaviour
I'd expect to be able to pass the calibration file via a simple launch parameter, similar to how it was on the ROS1 version of this driver. If that's not possible anymore, I'd like to know for certain what is the proper way of passing that calibration file.

# Workaround Suggestion
I'm aware of issue #711, I decided to create a new issue because it is not clear from that issue that passing the calibration file at launch does not work. I know that there are different ways of launching with a calibration file mentioned in #711 (modify the launch.py-file directly, or creating a separate description package based on the [Universal_Robots_ROS2_Description](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description)-Repo. Both of those options seem quite complicated and are not really well documented anywhere.

I'm also aware that you want to work on documenting the usage of the calibration file (#799), I'd love to see you follow through with the documentation.

The same can be said for the huge number of deprecation warning. When you alert of something being deprecated (like here, passing the calibration file), I think there should be a distinction between "it's deprecated, but the old way still works, the new way is better/more convenient/whatever" and "it's deprecated, the old way does not work anymore, you are forced to use the new way / add a line to your launch script or set a new parameter in your launch command".
I know you want to reduce the number of deprecation warnings (#855). I do not really care about the number of warnings, I'd prefer you do some documentation for the warnings (why was that changed, what is the added functionality of the change, what will break, and how can one properly use the driver, modify his launch code and get rid of the warning). Figuring all of that out only from the deprecation-warnings is quite hard.

### Relevant log output

```shell
[ur_ros2_control_node-1] [WARN] [1706621580.699260669] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
.
.
.
[ur_ros2_control_node-1] [WARN] [1706621580.710764051] [controller_manager]: [Deprecated]: Automatic activation of all hardware components will not be supported in the future anymore. Use hardware_spawner instead.
.
.
.
[ur_ros2_control_node-1] [WARN] [1706621581.756432981] [UR_Client_Library:]: DEPRECATION NOTICE: Setting the keepalive count has been deprecated. Instead use the RobotReceiveTimeout, to set the timeout directly in the write commands. Please change your code to set the read timeout in the write commands directly. This keepalive count will overwrite the timeout passed to the write functions.
[ur_ros2_control_node-1] [WARN] [1706621581.756467176] [UR_Client_Library:]: DEPRECATION NOTICE: Setting the keepalive count has been deprecated. Instead you should set the timeout directly in the write commands. Please change your code to set the read timeout in the write commands directly. This keepalive count will overwrite the timeout passed to the write functions.
[ur_ros2_control_node-1] [INFO] [1706621581.756476774] [URPositionHardwareInterface]: Calibration checksum: 'calib_17227329492635474227'.
[ur_ros2_control_node-1] [ERROR] [1706621582.762050289] [URPositionHardwareInterface]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_calibration/README.md] for details.
.
.
.
[ur_ros2_control_node-1] [WARN] [1706621583.169140952] [joint_trajectory_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-05-08:
Thank you for reporting this. We are currently restructuring things a bit (for rolling) where this will be covered.

VinDp, 2024-05-22:
Hey @A13X15, 

I tried to reproduce your issue and I noticed that for Humble, differently from Rolling, it's not possible to pass the robot calibration as launch argument. To add that I opened a pull request [(#1001)](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/1001) which introduces the `kinematics_params_file` argument, like in Rolling.

In this way, changing your command to:

`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10 robot_ip:=my_robots_ip launch_rviz:=false initial_joint_controller:=joint_trajectory_controller kinematics_params_file:=/path/to/my_robot_calibration.yaml` 

should do the trick, I tested it and it correctly uses the calibration file that I indicate.

Regarding documentation, we are currently working on the one for Humble, but I can point you to the updated documentation for Rolling, where you can check:
- [calibration docs](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_calibration/doc/index.rst)
- the [robot cell tutorial](https://github.com/UniversalRobots/Universal_Robots_ROS2_Tutorials/tree/main/my_robot_cell) which describes how to create a custom description package.

For the deprecation warnings, we are currently tackling those in order to make them disappear from Rolling and on.

VinDp, 2024-07-03:
Since the PR regarding the calibration file has been merged and the deprecation warnings have been removed from Rolling and Jazzy, I will close this issue. 
Feel free to reopen if still relevant.

closed 2024-07-03 by VinDp

A13X15, 2024-07-12:
Just updated the driver and passing the calibration file works. Thanks a lot for fixing!

cpsiff, 2024-12-13:
I'm still running into this issue on Jazzy using the latest version of the driver (2.4.13). But it seems like this was fixed a few releases ago. Am I missing something?

output of `apt list --installed | grep ur-robot`:
```
ros-jazzy-ur-robot-driver/noble,now 2.4.13-1noble.20241118.194418 amd64 [installed,automatic]
```

cpsiff, 2024-12-13:
I discovered my problem - I didn't realize that the parameter was changed from `kinematics_config` to `kinematics_params_file`. All is working now :+1: Leaving the comment in case someone else runs into the same problem