UniversalRobots/Universal_Robots_ROS2_Driver #924: Adding tf_prefix and prefix

Rudresh172, 2024-02-05:
### Affected ROS2 Driver version(s)

2.4.2

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.12.0 / 5.12.2 / 5.14.6

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
How to add prefix to control the ur5e through MoveIt2

## Issue details
I am assembling ur5e on a mobile base. So I need to change the prefixes in order to use it. I have added the tf_prefix in the robot driver launch file -
```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=robot_ip tf_prefix:=ur_
```
I am able to see the tf_prefix in the tf_tree
But when I run the moveit launch file -
```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true prefix:=ur_
```
I get the following error -
```
[move_group-1] [ERROR] [1707128337.136662308] [moveit_robot_model.robot_model]: Joint 'ur_shoulder_pan_joint' not found in model 'ur'
```

How do I add the prefixes so that the moveit file runs smoothly. 
Also, I saw in the ur_control launch file that the joint names in the configuration also need to be updated. Which yaml file needs to be updated in this case?
The ur_controllers file already uses a tf_prefix, how do I assign the tf_prefix there?

## Steps to Reproduce
```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=robot_ip tf_prefix:=ur_
```
```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true prefix:=ur_
```

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Relevant log output

```shell
[move_group-1] [ERROR] [1707136533.955008910] [moveit_robot_model.robot_model]: Joint 'ur_shoulder_pan_joint' not found in model 'ur'
[move_group-1] [ERROR] [1707136533.955013884] [moveit_robot_model.robot_model]: Joint 'ur_elbow_joint' not found in model 'ur'
[move_group-1] [ERROR] [1707136533.955015650] [moveit_robot_model.robot_model]: Joint 'ur_wrist_1_joint' not found in model 'ur'
[move_group-1] [ERROR] [1707136533.955017289] [moveit_robot_model.robot_model]: Joint 'ur_shoulder_lift_joint' not found in model 'ur'
[move_group-1] [ERROR] [1707136533.955018948] [moveit_robot_model.robot_model]: Joint 'ur_wrist_2_joint' not found in model 'ur'
[move_group-1] [ERROR] [1707136533.955020586] [moveit_robot_model.robot_model]: Joint 'ur_wrist_3_joint' not found in model 'ur'
```

### Accept Public visibility

- [X] I agree to make this context public

Rudresh172, 2024-02-13:
Hey @fmauch,

I just saw that the ur_moveit.launch.py file needs to have 'tf_prefix' instead of 'prefix' while loading the [urdf](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/31993abdd5a22284adde08378668457c484e86eb/ur_moveit_config/launch/ur_moveit.launch.py#L117C13-L117C24).

Now I am able to loading the model in the MoveIt Rviz window, but there are errors with the move group name. Here is the detailed log file - 
```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true prefix:=ur_
[INFO] [launch]: All log files can be found below /home/not_windows/.ros/log/2024-02-13-15-01-16-518239-LabTop3-23728
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [23735]
[INFO] [rviz2-2]: process started with pid [23737]
[INFO] [servo_node_main-3]: process started with pid [23739]
[servo_node_main-3] [WARN] [1707832877.261940233] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node_main-3] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node_main-3] to the Servo composable node in the launch file
[move_group-1] [WARN] [1707832877.265416769] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[servo_node_main-3] [INFO] [1707832877.266593881] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00250802 seconds
[servo_node_main-3] [INFO] [1707832877.266622238] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[servo_node_main-3] [INFO] [1707832877.266628897] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [INFO] [1707832877.267614946] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00204439 seconds
[move_group-1] [INFO] [1707832877.267637117] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[move_group-1] [INFO] [1707832877.267642805] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[servo_node_main-3] [WARN] [1707832877.275285098] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[move_group-1] [WARN] [1707832877.278531714] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[servo_node_main-3] [INFO] [1707832877.297968972] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/joint_states'
[servo_node_main-3] [INFO] [1707832877.299760182] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[servo_node_main-3] [INFO] [1707832877.299770213] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[servo_node_main-3] [INFO] [1707832877.300206367] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[servo_node_main-3] [INFO] [1707832877.300692758] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on '/servo_node/publish_planning_scene'
[servo_node_main-3] [ERROR] [1707832877.301857138] [moveit_robot_model.robot_model]: Group 'ur_manipulator' not found in model 'ur'
[servo_node_main-3] [ERROR] [1707832877.301870332] [moveit_servo.servo_calcs]: Invalid move group name: `ur_manipulator`
[servo_node_main-3] terminate called after throwing an instance of 'std::runtime_error'
[servo_node_main-3]   what():  Invalid move group name
[servo_node_main-3] Stack trace (most recent call last):
[servo_node_main-3] #14   Object "", at 0xffffffffffffffff, in 
[servo_node_main-3] #13   Object "/opt/ros/humble/lib/moveit_servo/servo_node_main", at 0x555799a99ca4, in 
[move_group-1] [INFO] [1707832877.311031193] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1707832877.311134441] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-1] [INFO] [1707832877.311377689] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-1] [INFO] [1707832877.311622305] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1707832877.311634405] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1707832877.312076162] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1707832877.312086905] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1707832877.312275122] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1707832877.312470476] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1707832877.312594560] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1707832877.312600936] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[... cut ...]
[rviz2-2] [INFO] [1707832880.894890624] [move_group_interface]: Ready to take commands for planning group ur_ur_manipulator.
[rviz2-2] [INFO] [1707832880.895175331] [moveit_ros_visualization.motion_planning_frame]: group ur_ur_manipulator
[rviz2-2] [INFO] [1707832880.895190987] [moveit_ros_visualization.motion_planning_frame]: group ur_ur_manipulator
[rviz2-2] [INFO] [1707832880.927225976] [interactive_marker_display_94670730909520]: Service response received for initialization

```

I have to mount the UR5e on top of a mobile base, hence I need to change the link names.

Thanks!

RobertWilbrandt, 2024-03-14:
Are you talking about [this](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/blob/ros2/ur_simulation_gazebo/urdf/ur_gazebo.urdf.xacro#L47) `world` reference? It does indeed seem like `ur_simulation_gazebo` incorrectly specifies as `tf_prefix` parameter in the [urdf](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/blob/ros2/ur_simulation_gazebo/urdf/ur_gazebo.urdf.xacro#L15) but passes a `prefix` in the [launch file](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/blob/ros2/ur_simulation_gazebo/launch/ur_sim_control.launch.py#L92). Can you verify if this is the problem you are seeing?

Rudresh172, 2024-03-15:
Yeah that was one of the problems. This same thing also occurs in the [ur_moveit.launch.py](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/31993abdd5a22284adde08378668457c484e86eb/ur_moveit_config/launch/ur_moveit.launch.py#L117C13-L117C24) file.

Regarding the world reference, if I remove the world link and make the mobile base link as the parent, the ur_moveit.launch.py terminal keeps throwing this error and I can't see the interactive marker -
```
[rviz2-2] [INFO] [1710538940.941615619] [interactive_marker_display_98530526167552]: Sending request for interactive markers
[rviz2-2] [INFO] [1710538940.989779617] [interactive_marker_display_98530526167552]: Service response received for initialization
[rviz2-2] [INFO] [1710538941.026142870] [interactive_marker_display_98530526167552]: Sending request for interactive markers
[rviz2-2] [INFO] [1710538941.075549261] [interactive_marker_display_98530526167552]: Service response received for initialization
```
[image]

The message only stops when I do Ctrl+C.

If I insert the world link, but keep the mobile base link as the parent, the interactive marker shows up, but it is very buggy.
[Screencast from 03-15-2024 10:37:46 PM.webm](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/assets/93994789/734dbdda-c267-4e22-8281-8c3e418b64f2)

Rudresh172, 2024-03-18:
The [srdf](https://github.com/Rudresh172/mobile_manipulator/blob/3a554de37b1948a06008cc6806604abd925aa3d9/Universal_Robots_ROS2_Driver/ur_moveit_config/srdf/mir_100.srdf#L13) has the correct joints. The only difference I made between the screenshot and the video in the previous post is the addition of world link. One possible explanation I can think of is that since the world frame is absent and moveit rviz window has world as the global fixed frame (it crashes if I change the global fixed frame), it fails to display the interactive marker.

The second problem: I have updated the collision matrix from the setup assistant GUI. It shows the red warning if I try to collide the arm with the base (verifying the collision matrix setup). I have kept the default values for sampling density and min. collisions. Do I need to change anything there?

RobertWilbrandt, 2024-03-19:
Using a link that does not exist as rviz fixed frame will definitely lead to problems. Does rviz give you any additional output when it crashes?

You should not have to change any settings in the collision updater. Is the behavior still the same after updating the srdf?

Rudresh172, 2024-03-21:
It gives this log when I change the fixed frame, Rviz freezes and then crashes in 5-10 seconds - 
```
[rviz2-2] [INFO] [1711007735.350631896] [interactive_marker_display_99721940551904]: Target frame is now base_link
[ERROR] [rviz2-2]: process has died [pid 439854, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/rudresh/Downloads/relffok_mir_30.1/install/ur_moveit_config/share/ur_moveit_config/rviz/view_robot.rviz --ros-args -r __node:=rviz2_moveit --params-file /tmp/launch_params_8urzsb5j --params-file /tmp/launch_params_arzrorba --params-file /tmp/launch_params_iepfirjx --params-file /home/rudresh/Downloads/relffok_mir_30.1/install/ur_moveit_config/share/ur_moveit_config/config/kinematics.yaml --params-file /tmp/launch_params_q0jjgt0y'].
```

The behaviour is still the same. I had updated the collision matrix while setting up MoveIt itself. I retried it with a fresh setup assistant, but the same output.

RobertWilbrandt, 2024-03-21:
At this point its getting kind of hard for me to follow - Can you verify that your other config files (`joint_limits.yaml`, `kinematics.yaml`) also reflect the prefix? From my understanding the tf chain is now `world` -> [...] -> platform `base_link` -> `ur_base_link` -> [...] and your SRDF group contains the chain `ur_base_link` to `ur_tool0`. You updated the SRDF allowed collision matrix with moveit setup assistant. Is that correct?

Rudresh172, 2024-03-21:
1. Yes, I have added prefixes to the relevant files, such that now I am able to control the joint with the joint value pane of the motion planning panel - 
[image]

2. The tf-chain is - 
[image]

3. My [srdf](https://github.com/Rudresh172/mobile_manipulator/blob/3a554de37b1948a06008cc6806604abd925aa3d9/Universal_Robots_ROS2_Driver/ur_moveit_config/srdf/mir_100.srdf#L13) group contains the chain ```ur_base_link``` to ```ur_tools0```
4. I have updated my collision matrix with moveit setup assistant

You can also checkout the [GitHub](https://github.com/Rudresh172/mobile_manipulator) repo for clarifications. 

Additionally, I also cloned the [gazebo simulation](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/tree/humble) repo and made the same changes, but the interactive marker is working fine there.

So I am really confused why the interactive marker isn't working with the mobile base.

RobertWilbrandt, 2024-04-10:
Sorry for the late response. At this point i am no longer sure what else to try, seeing that the tf tree looks correct for you and the SRDF also works this probably has some reason outside of this driver. We'll try fixing the handling of tf prefixes in all driver components. Please let us know if you find out what the problem in your setup is.