UniversalRobots/Universal_Robots_ROS2_Driver #932: [ur_robot_driver] ur_ros2_control_node crashes when loading robot_description from a topic

mrjogo, 2024-02-14:
### Affected ROS2 Driver version(s)

2.2.10-1jammy.20240126.080209

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.13.0

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Passing the `robot_description` to `ur_ros2_control_node` as a topic instead of a parameter causes it to crash

## Issue details
To have only one source of truth / DRY, the `robot_description` should be read off of `/robot_description` published by `robot_state_publisher`, not passed as a param. Doing so, however, causes a crash. Could be related to https://github.com/ros-controls/ros2_control/issues/1262.

## Steps to Reproduce
Here is a minimal launchfile that reproduces the failure:

```python
# reproduce.launch.py
# ros2 launch my_pkg reproduce.launch.py use_topic:=true
from launch_ros.actions import Node
from launch_ros.parameter_descriptions import ParameterFile
from launch_ros.substitutions import FindPackageShare

from launch import LaunchDescription
from launch.actions import SetLaunchConfiguration, DeclareLaunchArgument
from launch.substitutions import (
    LaunchConfiguration,
    Command,
    FindExecutable,
    PathJoinSubstitution,
)
from launch.conditions import IfCondition, UnlessCondition

def generate_launch_description():
    # Vars setup
    set_tf_prefix = SetLaunchConfiguration("tf_prefix", "")
    use_topic = LaunchConfiguration("use_topic")
    declare_use_topic = DeclareLaunchArgument(
        "use_topic",
        default_value="true",
        description="Use topic to pass robot_description to ur_ros2_control_node",
    )

    # Robot description
    script_filename = PathJoinSubstitution(
        [
            FindPackageShare("ur_client_library"),
[... 176 lines cut ...]
[ur_ros2_control_node-2] [INFO] [1707954295.126391763] [resource_manager]: Loading hardware 'ur' 
[ur_ros2_control_node-2] [INFO] [1707954295.128072168] [resource_manager]: Initialize hardware 'ur' 
[ur_ros2_control_node-2] [INFO] [1707954295.128122733] [resource_manager]: Successful initialization of hardware 'ur'
[ur_ros2_control_node-2] [WARN] [1707954295.128277301] [controller_manager]: [Deprecated]: Automatic activation of all hardware components will not be supported in the future anymore. Use hardware_spawner instead.
[ur_ros2_control_node-2] [INFO] [1707954295.128300251] [resource_manager]: 'configure' hardware 'ur' 
[ur_ros2_control_node-2] [INFO] [1707954295.128307874] [resource_manager]: Successful 'configure' of hardware 'ur'
[ur_ros2_control_node-2] [INFO] [1707954295.128331640] [resource_manager]: 'activate' hardware 'ur' 
[ur_ros2_control_node-2] [INFO] [1707954295.128343106] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-2] [INFO] [1707954295.128354026] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-2] [WARN] [1707954295.128956698] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ur_ros2_control_node-2] [INFO] [1707954295.575642183] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-2] [INFO] [1707954295.576014970] [UR_Client_Library:]: Setting up RTDE communication with frequency 500.000000
[ur_ros2_control_node-2] [WARN] [1707954296.600732736] [UR_Client_Library:]: DEPRECATION NOTICE: Setting the keepalive count has been deprecated. Instead use the RobotReceiveTimeout, to set the timeout directly in the write commands. Please change your code to set the read timeout in the write commands directly. This keepalive count will overwrite the timeout passed to the write functions.
[ur_ros2_control_node-2] [WARN] [1707954296.600770473] [UR_Client_Library:]: DEPRECATION NOTICE: Setting the keepalive count has been deprecated. Instead you should set the timeout directly in the write commands. Please change your code to set the read timeout in the write commands directly. This keepalive count will overwrite the timeout passed to the write functions.
[ur_ros2_control_node-2] [INFO] [1707954296.600791785] [URPositionHardwareInterface]: Calibration checksum: 'calib_12788084448423163542'.
[ur_ros2_control_node-2] [ERROR] [1707954297.694331192] [URPositionHardwareInterface]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_calibration/README.md] for details.
[ur_ros2_control_node-2] [INFO] [1707954297.694428555] [URPositionHardwareInterface]: System successfully started!
[ur_ros2_control_node-2] [INFO] [1707954297.694459343] [resource_manager]: Successful 'activate' of hardware 'ur'
[ur_ros2_control_node-2] [WARN] [1707954297.699331089] [controller_manager]: Could not enable FIFO RT scheduling policy
[ur_ros2_control_node-2] [WARN] [1707954297.699511046] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-03-05:
That should hopefully be fixed by https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/936. @mrjogo could you please check that?

mrjogo, 2024-03-05:
That seems to fix it! Thanks.

closed 2024-03-05 by mrjogo