UniversalRobots/Universal_Robots_ROS2_Driver #935: Execution stuck or timeout when running ur_moveit_config

karimsiala, 2024-02-22:
### Affected ROS2 Driver version(s)

2.2.9-1focal.20231213.212503 arm64 

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel, Other Linux system

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

3.15.8.106339

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
While setting up an Nvidia Jetson Orin NX board with a Universal Robots UR3 using the `Universal_Robots_ROS2_Driver`, I encountered an issue during trajectory execution through RViz with MoveIt2.

## Issue details
After successfully configuring the Nvidia board with the UR3 and ensuring connectivity (I can ping the robot, and execute scripts that pops up a window in the Polyscope interface), I face an issue when attempting to execute trajectories through RViz. Specifically, upon launching the default MoveIt config launch file and trying to execute a trajectory, it either stuck in a planning loop or I receive an error the controller is taking too long to execute the trajectory.

## Steps to Reproduce
1. Installed Universal_Robots_ROS2_Driver from binaries
2.  Launch the driver: `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3 robot_ip:=192.168.10.166 launch_rviz:=false kinematics_config:=~/my_robot_calibration.yaml`
3. Launch the included MoveIt config: `ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3 launch_rviz:=true`
4. Try to plan and execute a path

## Expected Behavior
I expected the robot to execute the planned trajectory smoothly without any errors, as the setup was completed according to the provided documentation.

## Actual Behavior
The trajectory execution either takes forever and stucks in a loop or fails with an error message from MoveIt, indicating for example  that `the controller is taking too long to execute the trajectory, specifically exceeding an expected upper bound of 5.029743 seconds`. This prevents any trajectory execution through RViz, despite successful communication tests.

[image]

Here is the log of an instance where it was stuck in a loop

Thank you in advance and I hope I'm note creating a duplicate because I made sure to search everywhere for a solution.

### Relevant log output

```shell
[INFO] [launch]: All log files can be found below /home/csia-jetson/.ros/log/2024-02-22-11-01-17-262718-csiajetson-desktop-6881
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [6889]
[INFO] [rviz2-2]: process started with pid [6891]
[INFO] [servo_node_main-3]: process started with pid [6893]
[servo_node_main-3] [WARN] [1708596081.300036882] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node_main-3] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node_main-3] to the Servo composable node in the launch file
[servo_node_main-3] [INFO] [1708596081.310716255] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00432986 seconds
[servo_node_main-3] [INFO] [1708596081.310778336] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[servo_node_main-3] [INFO] [1708596081.310796033] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [WARN] [1708596081.320804223] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1708596081.326334425] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00484945 seconds
[... 88 lines cut ...]
[rviz2-2] [INFO] [1708596086.509007432] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'ur_manipulator' in namespace ''
[rviz2-2] [INFO] [1708596086.531162195] [interactive_marker_display_187651643367712]: Sending request for interactive markers
[rviz2-2] [INFO] [1708596086.536440200] [move_group_interface]: Ready to take commands for planning group ur_manipulator.
[rviz2-2] [INFO] [1708596086.566530466] [interactive_marker_display_187651643367712]: Service response received for initialization
[move_group-1] [INFO] [1708596163.524992454] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[rviz2-2] [INFO] [1708596163.525309709] [move_group_interface]: Plan and Execute request accepted
[move_group-1] [INFO] [1708596163.525376847] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-1] [INFO] [1708596163.528392209] [moveit_move_group_default_capabilities.move_action_capability]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[move_group-1] [INFO] [1708596163.528755001] [moveit_ros.plan_execution]: Planning attempt 1 of at most 1
[move_group-1] [INFO] [1708596163.528803738] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'move_group'
[move_group-1] [INFO] [1708596163.530938281] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-1] [WARN] [1708596163.595062918] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[move_group-1] [INFO] [1708596163.601556917] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1708596163.601627382] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1708596163.601813178] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-1] [INFO] [1708596163.608204486] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-1] [INFO] [1708596163.608319977] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1708596163.608363658] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1708596163.608763827] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to scaled_joint_trajectory_controller
[rviz2-2] [INFO] [1708596208.813028302] [move_group_interface]: MoveGroup action client/server ready
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-03-05:
Well, did you start the external control program on the robot? Do you see the `Robot ready to receive control commands` output in the driver's shell?

Also, when executing a trajectory, the driver should print about that event, so that log output might be relevant as well, to see what's going on.

karimsiala, 2024-03-27:
I was unable to further troubleshoot the issue due to a sudden change in the project hardware. Thank you so much for your assistance.
I will close the issue.

closed 2024-03-27 by karimsiala