UniversalRobots/Universal_Robots_ROS2_Driver #941: UR5 reverse interface keeps dropping in WSL2

AGuthmann, 2024-03-05:
### Affected ROS2 Driver version(s)

2.4.3-1jammy.20240228.072527

### Used ROS distribution.

Rolling

### Which combination of platform is the ROS driver running on.

Windows using WSL2

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR CB3 robot, Real robot

### Robot SW / URSim version(s)

3.15.8

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I'm currently trying to setup the driver with a real UR5 Robot and after launching the external control program on the Teach Panel the reverse interface keeps loosing connection and reconnecting.

## Issue details
Platform used: WSL2 (Ubuntu 22.04.4) on Windows 10 (22H2)
Laptop and Robot are directly connected via ethernet cable, can ping Robot from Windows as well as from WSL2.
I'm launching the driver with custom urdf-file and `ur_type:=ur5 robot_ip:=192.168.1.10 launch_rviz:=false reverse_ip:=192.168.1.254 `.
The Driver starts fine until here, when i start the external control program on the TP the driver shows `robot requested program` , , `sent program to robot` and `Robot connected to reverse interface. Ready to receive control commands.` immediately followed by `Connection to reverse interface dropped.`, which pops up every second.

I found this [Issue](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/507) for the first version of the driver and the follow up for the ros2 driver #534, and tried to add `keep_alive_count` to the driver call, but the connection keeps dropping.

## Steps to Reproduce
1. Install ros-rolling-desktop, ros-rolling-ur and dependencies
2. Install URCap (external-control-1.0.5) on UR5 and create program for external control
3. Connect laptop with robot
4. Start driver
5. Start external-control program

## Expected Behavior
Robot connects to reverse interface and waits for movement commands.

## Actual Behavior
Robot connects to reverse interface and immediately looses connection, then reconnects only to loose connection again and so on.

### Relevant log output

```shell
[ur_ros2_control_node-1] [INFO] [1709625804.751190441] [UR_Client_Library:]: Robot requested program                                                                                                                                         [ur_ros2_control_node-1] [INFO] [1709625804.751516175] [UR_Client_Library:]: Sent program to robot                                                                                                                                           [ur_ros2_control_node-1] [INFO] [1709625804.950769462] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.                                                                                        [ur_ros2_control_node-1] [INFO] [1709625807.084426649] [UR_Client_Library:]: Connection to reverse interface dropped.                                                                                                                        [ur_ros2_control_node-1] [INFO] [1709625807.087728689] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.                                                                                        [ur_ros2_control_node-1] [INFO] [1709625809.212327378] [UR_Client_Library:]: Connection to reverse interface dropped.                                                                                                                        [ur_ros2_control_node-1] [INFO] [1709625809.219461721] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.                                                                                        [ur_ros2_control_node-1] [INFO] [1709625811.348872465] [UR_Client_Library:]: Connection to reverse interface dropped.                                                                                                                        [ur_ros2_control_node-1] [INFO] [1709625811.351471736] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.                                                                                        [ur_ros2_control_node-1] [INFO] [1709625812.605244396] [UR_Client_Library:]: Connection to reverse interface dropped.
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-03-05:
As written in other places: We aren't that familiar with how network traffic is handled on WSL2. As we are currently relying on  rather strict timing constraints we encourage people to use a native Linux with a low-latency kernel.

So, I'm afraid I cannot give much more help at the time being.

Jshulgach, 2024-04-25:
@AGuthmann this was a problem I experienced myself as well using the foxy distribution on a Jetson nano running Ubuntu 20.04. The fix for me was to uninstall the binary packages, build from source, go into the `hardware_interfaces.cpp` file and insert a new line to the initialization function: `ur_driver_->setKeepaliveCount(500);` right before `ur_driver_->startRTDECommunication();`.

For the foxy distibution the startup function was `URPositionHardwareInterface::start()`. 

I'm running WSL2 on a Windows 11 PC and have set up the drivers from source with a similar hard-coded modification, until I find an easy way to modify the system parameter. 500ms could be significantly longer than needed, so you can adjust accordingly.