UniversalRobots/Universal_Robots_ROS2_Driver #947: Issue name

MatthijsBurgh, 2024-03-18:
### Affected ROS2 Driver version(s)

2.4.3

### Used ROS distribution.

Rolling

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

5.15

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
The current version of [`ur_control.launch.py`](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/6aa6d57a8bf3309dea1c3b7bd03f63b2d7846c07/ur_robot_driver/launch/ur_control.launch.py) doesn't allow for a custom description file. As you can only specify a file inside the `ur_robot_driver` package.
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/6aa6d57a8bf3309dea1c3b7bd03f63b2d7846c07/ur_robot_driver/launch/ur_control.launch.py#L128

## Issue details
Of course you could fork the repo and add a custom description file. Of course, you shouldn't. I wanted to suggest a change to use the `description_package` argument instead of the hardcoded `ur_robot_driver`, though, the default of the `description_package` is `ur_description`. So there is a conflict. So I leave it up to you, whether you want to introduce another argument or come with another solution. But I think it very useful if the [`ur_control.launch.py`](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/6aa6d57a8bf3309dea1c3b7bd03f63b2d7846c07/ur_robot_driver/launch/ur_control.launch.py) can be used with a custom description file from a custom package.

## Steps to Reproduce
Not relevant

## Expected Behavior
Being able to use the [`ur_control.launch.py`](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/6aa6d57a8bf3309dea1c3b7bd03f63b2d7846c07/ur_robot_driver/launch/ur_control.launch.py) with my own custom description file located in my own description package.

## Actual Behavior
Only being able to use a description file from `ur_robot_driver`.

# Workaround Suggestion
Besides just copying (parts of) the [`ur_control.launch.py`](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/6aa6d57a8bf3309dea1c3b7bd03f63b2d7846c07/ur_robot_driver/launch/ur_control.launch.py) to my own package, there is no workaround.

### Accept Public visibility

- [X] I agree to make this context public

RobertWilbrandt, 2024-03-18:
Thank you for the issue, this was introduced in #877 and seems like an oversight to me. I'll create a PR to fix this.

MatthijsBurgh, 2024-03-18:
I have now implemented the following change for myself. Let me know you want to a PR with this solution.

Change the default value of the `description_file` to `PathJoinSubstitution([FindPackageShare("ur_robot_driver"), "urdf", "ur.urdf.xacro"]),`
and just replace 
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/6aa6d57a8bf3309dea1c3b7bd03f63b2d7846c07/ur_robot_driver/launch/ur_control.launch.py#L128
with `description_file,`.

But I could also see, you will move all the urdf/xacro files to `ur_description`. Then you can use `description_package`. But than you can't use default xacro, with custom calibration file.

So let me know what you want.

liamcarlos, 2024-03-19:
Hey @RobertWilbrandt there's a similar issue but with the moveit_config_package variable in ur_moveit.launch.py where this variable is used to set the package for the srdf, rviz, kinematics, and other configurations.  It would probably be best to have separate variables for some of these that would most likely have different values in a custom project.  

That could also be said for the description_package variable in the same file, since it dictates the package for the urdf and yaml parameters.

RobertWilbrandt, 2024-03-20:
Oh yeah it makes sense that we had to take a urdf from a package different from `description_package` at that point, as we wanted to move all specific driver / gazebo / ignition parts out of the description package. Maybe we have to do some restructuring here, e.g. have some new `kinematics_package` argument to load the yamls from and use `description_package` for the urdf again. Do you have any thoughts on this @fmauch @VinDp ?

liamcarlos, 2024-03-22:
@RobertWilbrandt Yeah just another heads up, the Gazebo ROS2 simulation package would also have to be updated.  It also uses those packages, so changing them would break that package as well..

RobertWilbrandt, 2024-03-27:
Thanks, yeah i think we have to go through all ur packages and see how this is handled after the change.

fmauch, 2024-04-24:
We're currently in the processing of updating things, see #977 and https://github.com/ros-controls/ros2_control/pull/1501

closed 2024-05-06 by fmauch