UniversalRobots/Universal_Robots_ROS2_Driver #952: Compile error: str_cat not defined

Muslinmin, 2024-03-21:
### Affected ROS2 Driver version(s)

https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/actions/runs/4782447849

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

Polyscope 3.7

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary

What I wanted to do was connect my ROS2 in my pc (in linux) to the URCAP external control on the teach pendant of the UR10. After launching the ur_robot_driver in the command prompt (logs below), I pressed the play button on the external control of the teach pendant. I got a compile error as shown in the photo attached here: 
[image]

## Steps to Reproduce

I installed the latest binary build of the ur_robot_driver, which is listed on at this page: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/humble.

The URCAP external program was installed from this repository, versioned 1.0.5.

Did sudo apt update and sudo apt upgrade.

Configured the URCAP external program on the teach pendant and set the proper IP addresses as instructed by the documentation here: https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/install_urcap_cb3.html

Launched the ur_robot_driver with this command:  `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10 robot_ip:=192.168.1.102 launch_rviz:=false initial_joint_controller:=joint_trajectory_controller.`

I tried with the default scaled_joint_trajector_controller as well.

I pressed the play button, and the compile error pops up at every instance I pressed the button. The log prints:

```
[ur_ros2_control_node-1] [INFO] [1710762390.849514083] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1710762390.849674378] [UR_Client_Library:]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1710762412.302452697] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1710762412.302595572] [UR_Client_Library:]: Sent program to robot
```

Any help is appreciated. Thank you

### Relevant log output

```shell
lusmse@lusmse:~$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10 robot_ip:=192.168.1.102 launch_rviz:=false initial_joint_controller:=joint_trajectory_controller
[INFO] [launch]: All log files can be found below /home/lusmse/.ros/log/2024-03-18-19-45-08-073239-lusmse-24649
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [24654]
[INFO] [controller_stopper_node-3]: process started with pid [24656]
[INFO] [ur_ros2_control_node-1]: process started with pid [24652]
[INFO] [urscript_interface-4]: process started with pid [24658]
[... 98 lines cut ...]
[ur_ros2_control_node-1] [INFO] [1710762386.791123098] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1710762386.791260142] [UR_Client_Library:]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1710762390.849514083] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1710762390.849674378] [UR_Client_Library:]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1710762412.302452697] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1710762412.302595572] [UR_Client_Library:]: Sent program to robot
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[controller_stopper_node-3] [INFO] [1710762523.212251707] [rclcpp]: signal_handler(signum=2)
[dashboard_client-2] [INFO] [1710762523.212251701] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-5] [INFO] [1710762523.212236818] [rclcpp]: signal_handler(signum=2)
[ur_ros2_control_node-1] [INFO] [1710762523.212269797] [rclcpp]: signal_handler(signum=2)
[urscript_interface-4] [INFO] [1710762523.212273509] [rclcpp]: signal_handler(signum=2)
[ur_ros2_control_node-1] [INFO] [1710762523.272208998] [URPositionHardwareInterface]: Stopping ...please wait...
[INFO] [urscript_interface-4]: process has finished cleanly [pid 24658]
[INFO] [controller_stopper_node-3]: process has finished cleanly [pid 24656]
[INFO] [robot_state_publisher-5]: process has finished cleanly [pid 24660]
[INFO] [dashboard_client-2]: process has finished cleanly [pid 24654]
[ur_ros2_control_node-1] [INFO] [1710762525.164283181] [URPositionHardwareInterface]: System successfully stopped!
[INFO] [ur_ros2_control_node-1]: process has finished cleanly [pid 24652]
lusmse@lusmse:~$
```

### Accept Public visibility

- [X] I agree to make this context public

RobertWilbrandt, 2024-03-27:
The polyscope version you are using is not supported by this driver, we require at least [version 3.14.3](https://github.com/UniversalRobots/Universal_Robots_Client_Library?tab=readme-ov-file#requirements). In particular, `str_cat` was only introduced in [version 3.8.0](https://www.universal-robots.com/articles/ur/release-notes/release-note-software-version-38xx/). We suggest you update the software on your robot.

fmauch, 2024-07-10:
Closing this as the given answer seems to be appropriate.

closed 2024-07-10 by fmauch