UniversalRobots/Universal_Robots_ROS2_Driver #954: Aborted during moving

yinguoxiangyi, 2024-03-25:
### Affected ROS2 Driver version(s)

2.4.3-1jammy.20240228.070912

### Used ROS distribution.

Rolling

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.13.0

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Aborted suddenly during moving.

## Issue details
I am using rolling version of ur-robot-driver.  After a period of running, it always aborted during moving, then the robot (ur 5e) stops.

```
$ apt show ros-rolling-ur-robot-driver
Package: ros-rolling-ur-robot-driver
Version: 2.4.3-1jammy.20240228.070912
Priority: optional
Section: misc
Maintainer: Denis Stogl <denis@stoglrobotics.de>
Installed-Size: 2,701 kB
Depends: libc6 (>= 2.34), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libdw1 (>= 0.126), libgcc-s1 (>= 3.3.1), libstdc++6 (>= 11), ros-rolling-backward-ros, ros-rolling-controller-manager, ros-rolling-controller-manager-msgs, ros-rolling-force-torque-sensor-broadcaster, ros-rolling-geometry-msgs, ros-rolling-hardware-interface, ros-rolling-joint-state-broadcaster, ros-rolling-joint-state-publisher, ros-rolling-joint-trajectory-controller, ros-rolling-launch, ros-rolling-launch-ros, ros-rolling-pluginlib, ros-rolling-position-controllers, ros-rolling-rclcpp, ros-rolling-rclcpp-lifecycle, ros-rolling-rclpy, ros-rolling-robot-state-publisher, ros-rolling-ros2-controllers-test-nodes, ros-rolling-rviz2, ros-rolling-std-msgs, ros-rolling-std-srvs, ros-rolling-tf2-geometry-msgs, ros-rolling-ur-client-library, ros-rolling-ur-controllers, ros-rolling-ur-dashboard-msgs, ros-rolling-ur-description, ros-rolling-ur-msgs, ros-rolling-urdf, ros-rolling-velocity-controllers, ros-rolling-xacro, socat, ros-rolling-ros-workspace
Download-Size: 595 kB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: The new driver for Universal Robots UR3, UR5 and UR10 robots with CB3 controllers and the e-series.

```

```
### after a period of running ###
[ur_ros2_control_node-1] [INFO] [1711334254.582643196] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [INFO] [1711334254.582673027] [scaled_joint_trajectory_controller]: Accepted new action goal
[ur_ros2_control_node-1] [INFO] [1711334255.861731211] [scaled_joint_trajectory_controller]: Goal reached, success!
[ur_ros2_control_node-1] [INFO] [1711334255.889982073] [io_and_status_controller]: Setting digital output '0' to state: '0'.
[ur_ros2_control_node-1] [INFO] [1711334258.594450840] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [INFO] [1711334258.594476390] [scaled_joint_trajectory_controller]: Accepted new action goal
[ur_ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ur_ros2_control_node-1]   what():  Taking data from action server but nothing is ready
[ur_ros2_control_node-1] Stack trace (most recent call last):
[ur_ros2_control_node-1] #15   Object "", at 0xffffffffffffffff, in 
[ur_ros2_control_node-1] #14   Object "/opt/ros/rolling/lib/ur_robot_driver/ur_ros2_control_node", at 0x59389878ec84, in _start
[ur_ros2_control_node-1] #13   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x761ee4a29e3f]
[ur_ros2_control_node-1] #12   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x761ee4a29d8f]
[ur_ros2_control_node-1] #11   Object "/opt/ros/rolling/lib/ur_robot_driver/ur_ros2_control_node", at 0x59389878e9e8, in main
[ur_ros2_control_node-1] #10   Object "/opt/ros/rolling/lib/librclcpp.so", at 0x761ee5313474, in rclcpp::executors::MultiThreadedExecutor::spin()
[ur_ros2_control_node-1] #9    Object "/opt/ros/rolling/lib/librclcpp.so", at 0x761ee5313011, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[ur_ros2_control_node-1] #8    Object "/opt/ros/rolling/lib/librclcpp.so", at 0x761ee5309eee, in rclcpp::Executor::get_next_ready_executable_from_map(rclcpp::AnyExecutable&, std::map<std::weak_ptr<rclcpp::CallbackGroup>, std::weak_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::owner_less<std::weak_ptr<rclcpp::CallbackGroup> >, std::allocator<std::pair<std::weak_ptr<rclcpp::CallbackGroup> const, std::weak_ptr<rclcpp::node_interfaces::NodeBaseInterface> > > > const&)
[ur_ros2_control_node-1] #7    Object "/opt/ros/rolling/lib/librclcpp_action.so", at 0x761eb87a7785, in 
[ur_ros2_control_node-1] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x761ee4eae4d7, in __cxa_throw
[ur_ros2_control_node-1] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x761ee4eae276, in std::terminate()
[ur_ros2_control_node-1] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x761ee4eae20b, in 
[ur_ros2_control_node-1] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x761ee4ea2b9d, in 
[ur_ros2_control_node-1] #2    Source "./stdlib/abort.c", line 79, in abort [0x761ee4a287f2]
[ur_ros2_control_node-1] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x761ee4a42475]
[ur_ros2_control_node-1] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ur_ros2_control_node-1]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ur_ros2_control_node-1]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x761ee4a969fc]
[ur_ros2_control_node-1] Aborted (Signal sent by tkill() 207969 1000)
[ERROR] [ur_ros2_control_node-1]: process has died [pid 207969, exit code -6, cmd '/opt/ros/rolling/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_4tos6rap --params-file /home/robot/work/bin-picking-ws/install/picking_robot_driver/share/picking_robot_driver/config/ur5e_update_rate.yaml --params-file /home/robot/work/bin-picking-ws/install/picking_robot_driver/share/picking_robot_driver/config/picking_controllers.yaml'].

```

## Steps to Reproduce
N/A

## Expected Behavior
not abort

## Actual Behavior
abort

# Workaround Suggestion
not found

### Accept Public visibility

- [X] I agree to make this context public

firesurfer, 2024-03-26:
@yinguoxiangyi I guess this is a known issue with ros2 actions while running with a multithreaded executor. We have the same issue occasionally (actually it is really rare) with our setup. 
See https://github.com/ros2/rclcpp/issues/2242 for more information

yinguoxiangyi, 2024-04-01:
In my situation, it happens about 5 times one day. Maybe let it go for now. If any update or solution, please inform me. Thanks in advance.

firesurfer, 2024-04-02:
Just subscribe to the corresponding https://github.com/ros2/rclcpp/pull/2250 
I just stumbled upon your issue here because I run into this issue myself sometimes....

RobertWilbrandt, 2024-04-02:
For reference: #380. We used to have a long-term test running the driver continuously and regularly ran into this error. This was resolved with PR 2250.

fmauch, 2024-04-24:
Closing this as a duplicate of #380 

@yinguoxiangyi to stay up-to-date please subscribe either to #380 or https://github.com/ros2/rclcpp/pull/2250

closed 2024-04-24 by fmauch