UniversalRobots/Universal_Robots_ROS2_Driver #955: I can't change velocity of robot when scaled joint trajectory control is actived.

ozgur-kurt, 2024-04-02:
### Affected ROS2 Driver version(s)

Humble UR

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

UR E-series robot, Real robot, URSim in docker

### Robot SW / URSim version(s)

5.15.2

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I am running ROS2 Humble UR10e in RVIZ and Moveit2 by validating URSim Virtual Controller Docker and my robot is moving so slowly when scaled_joint_trajectory_controller is used. I could not find to change correct parameters in terms of increasing velocity of robot. joint_limits.yaml did not work for me. I need to control my robot by using Moveit2 Servo and Pilz Motion Planner(PTP,LIN) in terms of checking singularity, collision and joint limits. I thought that it can be good solution if I switch my robot from scaled_joint_trajectory_controller to joint_trajectory_controller but I failed. If I can find a detailed tutorial between scaled_joint_trajectory_controller and forward_position_controller controllers it would be very good for me.

## Issue details
Everything seems to work but so slowly. I need to operate my robot in max actual speed.

### Accept Public visibility

- [X] I agree to make this context public

ozgur-kurt, 2024-04-17:
Thanks a lot Robert. When I send goals to the controllers that this script:

```
ros2 topic pub --once /urscript_interface/script_command std_msgs/msg/String '{data:
"def my_prog():
  textmsg(\"motion begin\")
  movej([2.3, -1.48, -1.55, -1.22, 1.23, -0.92], a=2.0, v=0.90, r=0)
  movel(p[0.188, 0.450, 0.589, 2.477, 2.470, 0.677], a=1.2, v=0.25, r=0)
  textmsg(\"motion finished\")
end"}'
```
UR robot moves without scaled speed that it worked but this script executed only UR controller. I need to control and move real robot by using RVIZ and Moveit. I am not sure which controller should I use that among joint_trajectory_controller, scaled_joint_trajectory_controller, forward_position_controller. I am only sure that scaled JTC is not useful for me. What do you suggest in terms of using moveit features such as avoidance of singularity, collision, joint limit ?

fmauch, 2024-04-18:
As Robert wrote: The motion speed is not a result from the controller, but from the controller's input. Please also see https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/966#issuecomment-2057771003