UniversalRobots/Universal_Robots_ROS2_Driver #956: Controlling Tool Output while using a newer version of the Robotiq Gripper URCap is not possible

firesurfer, 2024-04-02:
### Affected ROS2 Driver version(s)

Prob. all

### Used ROS distribution.

Iron

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.14.6.123463 

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

We have a UR16e with a Robotiq 2f-85 connected to the wrist. 
We currently use the Robotiq Gripper URCap on the arm in order to control the gripper. The URCap opens a TCP Socket which allows to remotely control the gripper as well as controller the gripper with the Teachpanel, what is quite convenient. 

But when upgrading the gripper URCap from  UCG-1.8.13.22852 to anything higher we can not select in the Installation tab -> ToolIO -> Controlled by: User, as the URCap will then refuse the interact with the gripper.

I contacted the Robotiq support and UR support. Basically the answer was. Yes this is intended (but they didn't tell me why). BUT: When selecting: ToolIO -> Controlled by:  Robotiq Gripper UrCap it is still possible to control the IO via the script command: `set_tool_digital_out` (Confirmed by the UR support. I didn' test it myself to be honest). 

When taking a look at the [external control](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/resources/external_control.urscript) urscript I can not find this command being used to set the tool io. (But we definitely can via the ROS2 Driver). I guess the IOs are set via a different control method.

So my request is: Support setting the tool IO via the urscript in order to workaround with issues in combination with the Robotiq Gripper UrCap. Perhaps you have a better contact to UR and can figure out why Robotiq was forced to include this check...

### Accept Public visibility

- [X] I agree to make this context public

RobertWilbrandt, 2024-04-02:
I don't quite understand how this relates to our ROS 2 driver. We use the [RTDE](https://www.universal-robots.com/articles/ur/interface-communication/real-time-data-exchange-rtde-guide/) protocol to read the robot state, move the arm and set i/os. This did not change.

What exactly do you want to achieve? Why do you need to be able to directly set tool i/os if the robotiq urcap is used?

RobertWilbrandt, 2024-04-08:
I see, that makes sense.

We are aware that using RTDE for IOs can lead to problems when integrating with additional equipment (also e.g. if you try to control the UR IOs from a PLC while controlling the joints through ROS). Enabling IO control through a different interface is on our roadmap, but we don't have a timeline for that yet.

@urrsk do you maybe have some more information on this specific conflict with robotiq?

urrsk, 2024-04-11:
@firesurfer you have a good point and we are aware about it. In general, we are missing a good way to integrate URCaps with the use of the ROS drivers.
On the short term, regarding the digital and analog outputs, we have talked about expending the ```script_commands``` thread in the [Universal_Robots_Client_Library/resources/external_control.urscript#L496](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/80aee077fe1b49879513e88e1d531df8372b7441/resources/external_control.urscript#L496) to also control the outputs as an alternative to use RTDE.

firesurfer, 2024-04-11:
@urrsk 
Could you perhaps elaborate why it should make a difference to control an IO via RTDE or via a urscript? In both cases one would need to accept that one can interfere with the settings from an installed URcap (or it has to be blocked). Unfortunately I did't get any useful answer from the UR support why this specific behavior has been changed (That the robotiq urcap may only control the gripper if the IOs are controlled by the urcap) .

EbbeFuglsang, 2024-04-18:
@firesurfer Our current way of managing the Tool Connector have it limitations. The intend with selecting one owner is that they are in control. In the past we have seen cases with other tool manufactures tools have been disturbed by the presence of Robotiqs URCap, even when they were selected as owner of the tool connector. That is what Robotiq have fixed here.
Is it an option to use the UCG-1.8.13.22852 version of the URCap?

firesurfer, 2024-04-21:
@EbbeFuglsang  Thanks for the info. 

> Is it an option to use the UCG-1.8.13.22852 version of the URCap?

Thats what we are currently doing. But given that there were some necessary updates in the past I would like being able to run the newest version of the URCap to avoid bad surprises in the future. 
(E.g. at some point the combination of the URCap and a newer Polyscope version lead to the gripper shutting down during startup because of an over current condition)